moveit/moveit #1944: CI based on *commit* instead of *branch*?

v4hn, 2020-03-09:
Playing around with https://github.com/ros-planning/moveit/pull/1899 ,
it looks like the following inconsistency can appear with our current CI system:

- Create a new pull-request with a broken commit / failing test
- While CI is pending, push another commit that fixes the problem
- CI will be triggered twice for both commits (I would expect that much), but will use the latest version of the pull-request branch in each case
- Both commits will be marked as succeeded in the pull-request

This is rather frustrating with the workflow we envision for bug-fixes:
- write a test that fails
- demonstrates the failure in CI
- write a fix for it
- push the fix and demonstrate it resolves the problem

Is it possible to have moveit_ci run on the concrete commit that triggered the run instead of the pull-request branch?

I don't really know who to highlight here aside from @rhaschke, please feel free to forward it to anyone else involved with the CI setup.

rhaschke, 2020-03-09:
I'm not sure what exactly you are complaining about. The last two Travis tests are
[failing](https://travis-ci.org/ros-planning/moveit/builds/659780686?utm_source=github_status&utm_medium=notification) and [passing](https://travis-ci.org/ros-planning/moveit/builds/659823577?utm_source=github_status&utm_medium=notification) as expected. If at all, this is a Travis issue, because Travis handles checking out the to-be-tested source repo.

v4hn, 2020-03-09:
On Mon, Mar 09, 2020 at 01:04:35AM -0700, Robert Haschke wrote:
> I'm not sure what exactly you are complaining about. The last two Travis tests are [...]

Yes, after I force-pushed the branch ~5 times.
The last time, I pushed the first commit, waited for CI to fail, and only then pushed the second commit.
But this requires a possibly hours-long delay in between.
People tend to simply forget to push branches often enough, as is...