moveit/moveit #1945: Rviz Plugin: "Object Status" field in "Scene Objects" tab does not wrap lines

felixvd, 2020-03-09:
### Description

When selecting a Scene Object, the text that is displayed in the Object Status field can become very long, causing the window to resize.

Fixing this issue would probably involve changing a Qt parameter in the Rviz GUI. I am not familiar with it, but to someone who is this might be as trivial a change as changing a textbox setting from "Fixed" to "Wrapping", if it exists.

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Source build (master)

### Steps to reproduce
Publish a `CollisionObject` with a lot of subframes and select it in Rviz.

### Expected behaviour
The lines should fill the text box without enlarging it significantly.

### Actual behaviour
The text box can increase in size to fill the entire screen. There is no immediately visible resizing option, either.

rhaschke, 2020-03-09:
Could you please provide a corresponding CollisionObject YAML and a screenshot of the undesired behavior?

felixvd, 2020-03-13:
My bad - this is introduced in #1866 in [this commit](https://github.com/ros-planning/moveit/pull/1866/commits/39d23a1935573a229072aeab3f2187ed33c51a99). But the question remains how to fix it. These are screenshots from the subframes tutorial:

1) [The original state](https://user-images.githubusercontent.com/4535737/76624484-becfed00-6578-11ea-826d-f71ac664458f.png)
2) [The widget after selecting "cylinder", which causes the widget to resize to accommodate the text](https://user-images.githubusercontent.com/4535737/76624464-b37cc180-6578-11ea-8056-8696c16b0b6f.png)  
3) [The widget after selecting the other object, with the widget remaining at its original size](https://user-images.githubusercontent.com/4535737/76624465-b4155800-6578-11ea-9b9c-fcc89a454b3d.png)

You can imagine how the widget would resize with additional subframes on the object.

v4hn, 2020-03-13:
The magic property you are looking for is probably [wordWrap](https://doc.qt.io/qt-5/qlabel.html#wordWrap-prop).

Aside from this, I would propose to move the object status text to the below `Scene Geometry`, in order to avoid repositioning of the GUI elements in between.

felixvd, 2020-03-23:
It is, thanks! I moved it text down, made it selectable and increased the contrast. Will update in the PR.

closed 2020-03-23 by felixvd