moveit/moveit #1954: Make Pointcloud Octomap Updater more general

Dale-Koenig, 2020-03-16:
**Is your feature request related to a problem? Please describe.**
Opening this to foster initial discussion on this idea and collect more suggestions.

Pointcloud octomap updater at the moment seems to be appropriate only for a very specific use case.  That is, where

- The camera location changes
- It is necessary to maintain a permanent updated map of the environment
- Speed/high resolution is not particularly important, and it is NOT necessary to ensure a most recent octomap

It is my opinion that this (especially the last) makes it pretty much unusable for serious applications in its current state.

**Describe the solution you'd like**
I think to make it useful, it should be refactored such that:

- Slower operations (ray tracing, filtering the robot) can be disabled with parameters in the sensor yaml file when they are not needed.  This can change the processing time by an order of magnitude (I was frequently seeing 0.5s with ray tracing, 0.05s without)
- When disallowing ray tracing, the octomap can be generated from scratch from each pointcloud.  I understand that this can almost certainly be done without using MoveIt at all, but it would be convenient to have all the functionality in one place.
- Refactored such that an inner component can be used to synchronously process the pointcloud and add it to a planning scene/psm when desired.  Currently, especially with the large processing time from ray tracing, processing can finish later than expected resulting in plans using an old octomap. Bonus if this could be used without needing the locks/mutexes.
- Perhaps improved interfaces for filtering by shapes, intended for filtering objects to pick rather than just parts of the robot.

These are features that I currently have in my hacked version, but would take a decent amount of work still to integrate back into the upstream version.

v4hn, 2020-03-16:
> The camera location changes

Isn't this the more general use-case? I believe you might refer to a required `waitForTransform` call?
If the camera is properly registered as a static transform in your tf2 system w.r.t. your  planning frame, it should not wait at all.

> It is necessary to maintain a permanent updated map of the environment

Not entirely true, that's why there is a service to clear the octomap.
In some setups, I previously implemented a selective forwarding of point cloud messages, so that the octomap is only updated when the external system relies on it.

> Speed/high resolution is not particularly important, and it is NOT necessary to ensure a most recent octomap

That is indeed a severe bottleneck in the current system. There is probably potential to improve computation time, e.g., utilizing [GPU-Voxels](https://www.gpu-voxels.org/).
I'm not aware of anyone working along these lines at the moment though.

> When disallowing ray tracing, the octomap can be generated from scratch from each pointcloud. I understand that this can almost certainly be done without using MoveIt at all, but it would be convenient to have all the functionality in one place.

Fair enough, this sounds like a nice parameter to have in the yaml.
Call it `incremental=false`, `snapshot=true`, or something like that and add the feature in a pull-request. I'd like to use it. :)
The obvious downside, that needs to be documented, is that you loose all collision checking with objects currently hidden behind other things, e.g., a robot arm.

> Perhaps improved interfaces for filtering by shapes, intended for filtering objects to pick rather than just parts of the robot.

I believe all `CollisionObject`s in the world are added here? That of course includes all "objects to pick".

> Refactored such that an inner component can be used to synchronously process the pointcloud and add it to a planning scene/psm when desired. Currently, especially with the large processing time from [...]

There is indeed no syncronous way to feed in a point cloud at the moment.
It seems like a simple matter to add a `MoveGroupCapability` that would open a ROS service for this?

> These are features that I currently have in my hacked version, but would take a decent amount of work still to integrate back into the upstream version.

Its always nice if you can refer to a git repository for references like "my hacked version". ;)

Your improvements would be very much welcome upstream!

Dale-Koenig, 2020-03-17:
>Isn't this the more general use-case? I believe you might refer to a required waitForTransform call?
If the camera is properly registered as a static transform in your tf2 system w.r.t. your planning frame, it should not wait at all.

I was thinking of the ray tracing/incremental updates as being for when the camera is attached to the 6th robot link and is taking images from multiple angles.  I have no problems with the tf2 usage.

>I believe all CollisionObjects in the world are added here? That of course includes all "objects to pick".

I suppose so.  My use case is basically a simple "Take pointcloud, turn into octomap, apply to planning scene" without going through PSM so I think that just didn't go well with the current transform cache strategy.

It does seem like adding a service capability and an option to skip ray tracing would take it from it's current state to "good enough for most people using move group".  It might take some thinking for me to figure out how to refactor to preserve the current use but also facilitate low level usage though.

v4hn, 2020-03-17:
> I was thinking of the ray tracing/incremental updates as being for when the camera is attached to the 6th robot link and is taking images from multiple angles. I have no problems with the tf2 usage.

raytracing is relevant for all kind of dynamic motions, also with a static camera. This includes the arm (or robot) moving around as well as user intervention.

> My use case is basically a simple "Take pointcloud, turn into octomap, apply to planning scene" without going through PSM so I think that just didn't go well with the current transform cache strategy.

If you circumvent the perception stack altogether, this will probably not take objects into account, yes.

> It does seem like adding a service capability and an option to skip ray tracing would take it from it's current state to "good enough for most people using move group". It might take some thinking [...]

Looking forward to the pull-request! :+1:

simonschmeisser, 2020-03-19:
We have been using a trimmed down version of the PointCloudOctomapUpdater for a couple of years now. As the updater is plugin based, I suggest having a different plugin rather than parameters? Might keep the code more comprehensible. I might get around to opening a PR ...
(we basically cut out the recognized objects in our object recognition already and then always only use the most recent shot without any filtering or raytracing)

Getting notified when the octomap changed (or a way to update synchronously) would be great indeed. We basically reset the octomap and then listen to all GEOMETRY_CHANGED "signals" of PlanningSceneMonitor and check for a non-zero octomap ... not very clean but works ... 

Another downside of the current filter is that (afaik) it assumes convex meshes. Doesn't work for bin-picking that well ...

Dale-Koenig, 2020-03-23:
[Allowing incremental updates](https://github.com/Dale-Koenig/moveit/commits/improve_pc_octomaps) is easy enough to do.  This also refactors a bit to make it not too difficult (although still a bit unnatural) to bypass the perception pipeline, as demonstrated in the tests.  I may make a PR just from this after fixing it up a bit more.

I'm having doubts about synchronous updates using a service though - there would likely have to be separate services for depth map updating and pointcloud updating, and I don't think the move group capability can know for sure which type it is when it is constructed.  I guess the service request could contain both a depth map and a pointcloud, and only the relevant one would be used?  Otherwise, I am not sure how this could be done in a way that adapts to the plugin system.

It also requires passing a pointcloud over messaging an extra time in a situation where performance is probably desired (due to synchronous updates).

v4hn, 2020-03-23:
> I'm having doubts about synchronous updates using a service though - there would likely have to be separate services for depth map updating and pointcloud updating

Yes, but where is the problem if these are fully independent services?

> It also requires passing a pointcloud over messaging an extra time in a situation where performance is probably desired (due to synchronous updates).

You could also add a service to signal the pipeline to read a single message (preferably with a timestamp newer than...) and wait for it to be processed before returning.
This somewhat complicates control flow though.

Dale-Koenig, 2020-03-24:
>Yes, but where is the problem if these are fully independent services?

I suppose it's okay, just less pretty and could get really messy if there were more plugins.  I'd be happier about it if it was easier to start (or not start) the service with the knowledge of which type of updater was launched.

v4hn, 2020-03-24:
On Mon, Mar 23, 2020 at 05:19:39PM -0700, Dale Koenig wrote:
> I'd be happier about it if it was easier to start (or not start) the service with the knowledge of which type of updater was launched.

Having one `MoveGroupCapability` that spawns the service/service type based on the configured type of octomap updater should be perfectly feasible.

Dale-Koenig, 2020-03-25:
> On Mon, Mar 23, 2020 at 05:19:39PM -0700, Dale Koenig wrote: I'd be happier about it if it was easier to start (or not start) the service with the knowledge of which type of updater was launched. [...]

It's possible, but the plugin system makes this pretty annoying since as far as I can tell you basically have to:

 - Read the parameter server to see which types of plugins were loaded
 - Implement public interfaces to allow accessing all the way to the updaters from the PSM, going through the OccupancyMapMonitor.
 - Dynamic pointer cast the updater (which means including plugin headers in the main code)

Allowing the PointcloudOctomapUpdater to advertise a service directly in its private namespace would make everything a lot easier.

v4hn, 2020-03-25:
> It's possible, but the plugin system makes this pretty annoying

True.

> Allowing the PointcloudOctomapUpdater to advertise a service directly in its private namespace would make everything a lot easier.

Makes sense at first glance, though it should be configurable with the plugin parameters then.
@simonschmeisser proposed to keep such features in separate plugins altogether, but I'm not convinced.
The current plugins provide a functionality almost nobody really uses for one reason or another.
So addressing use-cases that are actually *used* by users seems a good idea as long as the tutorials are updated properly.

simonschmeisser, 2020-05-05:
What do you think about an alternative approach where the time of the last update (ideally the stamp from the point cloud /depth image) is accessible somehow?

Currently there is `bool PlanningSceneMonitor::waitForCurrentRobotState(const ros::Time& t, double wait_time = 1.)` or `bool planning_scene_monitor::CurrentStateMonitor::haveCompleteState(const ros::Duration& age)` as well as `ros::Time planning_scene_monitor::CurrentStateMonitor::getCurrentStateTime()` but no similar mechanism for `OccupancyMapMonitor`.

If we had those values at hand, one could add a callback to `PlanningSceneMonitor` and react once you get `UPDATE_GEOMETRY` and a timestamp newer than when you captured.

Still sounds a bit complicated but would save one intermediate stop (and copy) on the way from camera to octomap.

Dale-Koenig, 2020-05-05:
In my use case the pointcloud must be processed at a very specific time, after performing various perception algorithms to identify/retrieve some objects from the pointcloud (as in the perception pipeline tutorial) and after adding those objects to the planning scene.  So I think it is necessary to have at least some method of actively controlling exactly when the pointcloud is processed.