moveit/moveit #1956: How to run cpp file after catkin build

Hubert51, 2020-03-16:
I want to run CPP test file in  `moveit/moveit_core/collision_detection_bullet/test`. But I did not know how to do that after I did `catkin build`. Would you like to provide some help?

Thank you!

rhaschke, 2020-03-17:
Read the docs: http://wiki.ros.org/ROS/Tutorials/UnderstandingNodes#Using_rosrun

closed 2020-03-17 by rhaschke

Hubert51, 2020-03-20:
Would you like to tell me which package `collision_detection_bullet` belongs to? I can not find `collision_detection_bullet` package. In `moveit_core` package I can not find this node. 

By the way, the moveit_core does not need to run in the ros. Can I just compile the `collision_detection_bullet` folder and run it?

Thank you!