moveit/moveit #1961: moveit_setup_assistant:  "moveit_controller_manager" launch file arg plumbing problems

zultron, 2020-03-20:
### Description

The `moveit_setup_assistant` generates e.g. a `myrobot_moveit_config` package with launch files using a `moveit_controller_manager` arg, plumbed as follows; for this example, the robot is named `myrobot`, and uses real hardware (not `fake_execution`):

- `move_group.launch`
  - Includes `trajectory_execution.launch.xml`
  - Passes `myrobot_moveit_config` arg, set to `myrobot`
- `trajectory_execution.launch.xml`
  - Sets `myrobot_moveit_config` arg, defaults to `myrobot`, set to `myrobot` by `move_group.launch`
  - Includes `myrobot_moveit_controller_manager.launch.xml` but does NOT set `myrobot_moveit_config`
- `myrobot_moveit_controller_manager.launch.xml`
  - Sets `myrobot_moveit_config` arg, defaults to `moveit_simple_controller_manager/MoveItSimpleControllerManager`

To summarize:  In this three-level hierarchy of launch file inclusions, the `myrobot_moveit_config` arg is plumbed through the first two levels with the value `myrobot`, and is NOT plumbed through to the third level, where it is set by default to `moveit_simple_controller_manager/MoveItSimpleControllerManager`, a value with entirely different meaning.

This means that an out-of-the-box (or out-of-the-assistant) `myrobot_moveit_config` package can't be configured to use `moveit_ros_control_interface::MoveItControllerManager` (which was the default a couple of years ago) without directly editing the `myrobot_moveit_controller_manager.launch.xml` file; that is, even though the file is set up to enable specifying a custom controller manager via a launch file arg, the arg isn't plumbed through from top-level launch files.

### Your environment
* ROS Distro: Kinetic
* OS Version: Debian Stretch
* Source build, 0.9.18-1 release

### Steps to reproduce

This problem will be visible in any MoveIt config package generated by the assistant, such as the tutorial `franka_robot` example.

### Expected behaviour

There should be a way to pass the controller manager class through top-level launch file args.

zultron, 2020-03-20:
[Oops, egregious copy-n-paste error in description now fixed!]

One way to fix it would be to rename the `moveit_controller_manager` arg to `moveit_controller_manager_class` in `myrobot_moveit_controller_manager.launch.xml`, and plumb that arg through from the top-level launch files with default set to `moveit_simple_controller_manager/MoveItSimpleControllerManager` (the current default).

Would that be expected to break a lot of configurations?  I'd guess no, since I doubt anyone includes that bottom-level launch file directly from outside the assistant-generated package.  I'd submit a PR if I had confirmation.

v4hn, 2020-03-20:
As you are surely aware, given the details in your report, `trajectory_execution.launch.xml`
includes `$(arg moveit_controller_manager)_moveit_controller_manager.launch.xml`.
The idea of this approach was to have maintainers write additional configuration launch files, e.g., `ros_control_moveit_controller_manager.launch.xml` if they want to support multiple plugins.
Keep in mind these plugins could also need additional parameters, which you cannot specify through a single parameter.
Originally, the setup assistant was never meant to generate a package nobody would ever edit and adding another file there can be done without even affecting later runs of the assistant to update the package.

Now, the `moveit_controller_manager` argument in the default `*_moveit_controller_manager.launch.xml` is indeed rather ridiculous with this workflow.

> This means that an out-of-the-box (or out-of-the-assistant) `myrobot_moveit_config` package can't be configured to use `moveit_ros_control_interface::MoveItControllerManager` (which was the default [...]

Did Dave ever make it a default? It never made sense as a default anyway, because a lot of supported robots out there do/did not support the `ros_control` API.

Either way you are wrong in that it is very much possible to configure the alternative controller manager. If you include your `move_group.launch` from your own launch file, you can always override the parameter afterwards:
```
<launch>
  <include file=$(find myrobot_moveit_config)/launch/move_group.launch)/>
  <param name="move_group/moveit_controller_manager" value="my_custom_plugin" />
</launch>
```
I'm not saying this is the best approach, but it should work even if you dislike the approach to add other launch files in `*_moveit_config`.

> One way to fix it would be to rename the moveit_controller_manager arg to moveit_controller_manager_class in myrobot_moveit_controller_manager.launch.xml, and plumb that arg through from the [...]

This adds one more mechanism to the already complex launch file hierarchy.
The trouble with all these files is that we cannot easily reduce complexity again because it all depends on the users running the setup assistant again and understanding what's going on.

I'm not convinced this additional parameter in all these launch files is justified, but maybe other maintainers disagree?

zultron, 2020-03-20:
@v4hn You've correctly divined my purpose:  generate a `_moveit_config` package and not touch it again.  We want to be able to re-run the assistant to apply updates without it clobbering our changes.

At any rate, I'm happy to configure the `move_group/moveit_controller_manager` param in our launch files, external to the `moveit_config` package.  As far as I'm concerned, this issue may be closed, but I'll leave it up to maintainers in case the question about the `moveit_controller_manager` argument in the default `*_moveit_controller_manager.launch.xml` needs to be addressed.

v4hn, 2020-03-20:
>  We want to be able to re-run the assistant to apply updates without it clobbering our changes.

Believe me, I do understand the motivation. Please notice that the first mechanism I pointed out, adding a new launch file and including it by a standard launch file parameter, does not clobber the potential updates through the setup assistant either.
The assistant does not know about the new file!

Closing this as no-fix.

closed 2020-03-20 by v4hn