moveit/moveit #1962: moveit_setup_assistant generates incorrect joint_position_controller configuration

zultron, 2020-03-20:
### Description

In `moveit_setup_assistant`, "ROS Control" tab, adding a controller type `position_controllers/JointPositionController` should allow adding only a single joint, and add it to the [`joint` param][1] under the controller namespace.  Instead, it adds it to a list in the `joints` param.  (That would be suitable for a `position_controllers/JointGroupPositionController` type, if the assistant supported it.)

Also, it adds PID gains, which are inapplicable to position controllers.

### Your environment
* ROS Distro: Kinetic
* OS Version: Debian Stretch
* Source build, 0.9.18-1 release

### Steps to reproduce

In the setup assistant:
- Click the "ROS Control" tab
- Click "Add Controller"
- Controller Name "joint_1_position_controller"
- Controller Type "position_controllers/JointPositionController"
- Click "Add Individual Joints"
- Click "joint_1"; click ">" to add it
- Click "Save"
- Generate the `moveit_config` package

### Expected behaviour

`ros_controllers.yaml` should contain a top-level key with only `type` and `joint` second-level keys:

```
joint_1_position_controller:
  type: position_controllers/JointPositionController
  joint: joint_1
```

### Actual behaviour

`ros_controllers.yaml` contains a top-level key with incorrect `joints` and inapplicable `gains` parameters:

```
joint_1_position_controller:
  type: position_controllers/JointPositionController
  joints:
    joint_1
  gains:
    joint_1:
      p: 100
      d: 1
      i: 1
      i_clamp: 1
```

### Backtrace or Console output

```
[INFO] [/controller_spawner] [rosout]: Loading controller: joint_1_position_controller
[ERROR]: No joint given (namespace: /joint_1_position_controller)
[ERROR]: Failed to initialize the controller
[ERROR]: Initializing controller 'joint_1_position_controller' failed
[ERROR] [/controller_spawner] [rosout]: Failed to load joint_1_position_controller
```

[1]: https://github.com/ros-controls/ros_controllers/blob/melodic-devel/position_controllers/include/position_controllers/joint_position_controller.h#L53

v4hn, 2020-03-20:
Thank you for the report. A patch for this is very much welcome.

https://github.com/ros-planning/moveit/issues/1937 is somewhat related, as the "ROS Control" part of this tab was merged premature and is not well tested or thought out...

Avdbergnmf, 2021-02-11:
Ran into the same bug today in ROS Melodic!