moveit/moveit #1966: Should `get_current_state_bounded` be available in MoveGroupCommander, and why is it bounded?

felixvd, 2020-03-23:
I wanted to use `get_current_state` in Python and came across this. Can someone do a quick sanity check?

[This function](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/move_group_interface/src/wrap_python_move_group.cpp#L310) `getCurrentStateBoundedPython` was introduced in the wrapper 4 years ago, but isn't used in `move_group.py`. I think this is an oversight.

Also, why is it bounded? Since there is no equivalent in the C++ MoveGroupInterface, I am a little confused.

If there's no objections I would submit a PR to connect this function in `move_group.py`, and maybe make it `get_current_state` instead of `get_current_state_bounded`, when I get around to it.

felixvd, 2020-03-23:
I tried to implement this, but when I call `py_bindings_tools::serializeMsg`, for some reason receive a string instead of a `moveit_msgs.msg.RobotState` object. Anyone have an idea what the problem might be? The very small changes are [here](https://github.com/felixvd/moveit/commit/ab4ecdda8579f9309c5075df9a6725429d36f28c).

```
Traceback (most recent call last):
  File "/home/panda/catkin_ws/src/osx_cgm_experiments/scripts/test_cgm_trajectory.py", line 158, in <module>
    velocity_scaling_factor=0.05, acceleration_scaling_factor=0.05)
  File "/home/panda/catkin_ws/src/moveit/moveit_commander/src/moveit_commander/move_group.py", line 602, in retime_trajectory
    ser_ref_state_in = conversions.msg_to_string(ref_state_in)
  File "/home/panda/catkin_ws/src/moveit/moveit_commander/src/moveit_commander/conversions.
```

I get that error when I try to call the `retime_trajectory` function, for which the `get_current_state` seems to be a prerequisite:

```
[...]
trajectory = 
robot_state = move_group.get_current_state()
trajectory = move_group.retime_trajectory(robot_state, trajectory, velocity_scaling_factor=0.05, acceleration_scaling_factor=0.05)
```

PeterMitrano, 2020-10-07:
I actually found this because I _want_ to use get_current_state_bounded, and saw it wasn't fully implemented. Did you ever finish this?

rhaschke, 2020-10-08:
The Python API by far isn't complete. If you are missing something, please file a corresponding PR.

commit 73416fb9 (2020-10-14): Python interface improvements. Fix #1966, add enforceBounds (#2356)

* implement get state and get state bounded

* add enforceBounds