moveit/moveit #1969: Integrating OMPL's Cartesian planning capabilities

JeroenDM, 2020-03-24:
This is a Google Summer of Code project idea proposed by @mamoll  I would love to get some extra feedback on from the community.

**Is your feature request related to a problem? Please describe.**
OMPL has specific planners for [constrained planning](http://ompl.kavrakilab.org/constrainedPlanning.html). These are not exposed by the current planning plugin as far as I know.

**Describe the solution you'd like**
Exposing the planners in the existing [OMPL planning plugin](https://github.com/ros-planning/moveit/tree/master/moveit_planners/ompl).

**Describe alternatives you've considered**
There exist other planners that solve similar problems, [Descartes](https://github.com/PickNikRobotics/descartes_capability) and [TrajOpt](https://github.com/ros-planning/moveit/tree/master/moveit_planners/trajopt). But as OMPL is already integrated with MoveIt it seems a good idea to keep improving this interface.

Also, the OMPL planners could be better suited for different types of problems than Descartes. The latter mainly focusses on end-effector paths with little or no tolerance. While the OMPL constrained planners could be better suited if there's a lot more freedom.

In addition, the OMPL planners use a different approach than TrajOpt, so comparing them would be interesting.

**Additional context**
I would like to integrate these capabilities in the context of Google Summer of Code 2020. See [discourse](https://discourse.ros.org/t/gsoc-2020-cartesian-planning-some-trajopt/13045/10) for more context. Other related links: [https://github.com/ros-planning/moveit/issues/467](https://github.com/ros-planning/moveit/issues/467), [https://github.com/ros-planning/moveit/pull/1593](https://github.com/ros-planning/moveit/pull/1593), [https://github.com/ros-planning/moveit/issues/797](https://github.com/ros-planning/moveit/issues/797), 
[https://github.com/PickNikRobotics/descartes_capability](https://github.com/PickNikRobotics/descartes_capability).

(PS: sorry for the spam for people who are getting a little bit too many notifications from me, since I'm asking feedback on different ideas :) )

v4hn, 2020-03-24:
This is a very good idea, that's why it's a GSoC project :)

This should fully replace [our broken pose_model_state_space](https://github.com/ros-planning/moveit/tree/master/moveit_planners/ompl/ompl_interface/src/parameterization/work_space).

JeroenDM, 2020-03-24:
Thank you for the specific pointer.

And, indeed, I forgot to mention it is one of the [future projects](https://moveit.ros.org/documentation/contributing/future_projects/) listed on the MoveIt website.

felixvd, 2020-03-25:
I'm looking forward to seeing this happen, and maybe even a few other improvements if you're quick (e.g. multiple planners per `move_group`).