moveit/moveit #1973: no CI checks for melodic-devel pull-requests?

v4hn, 2020-03-24:
Why does Travis not run `moveit_ci` for pull-requests against `melodic-devel`?

https://github.com/ros-planning/moveit/pull/1971#issuecomment-603364181

@davetcoleman @henningkayser @rhaschke ... Forward as required.

rhaschke, 2020-03-24:
On our github repository somebody disabled all required checks, which I fixed meanwhile. However, Travis is not triggered for another reason, which I don't understand yet.
@davetcoleman, did you guys at Picknik changed the configuration recently?

rhaschke, 2020-03-24:
The moveit repo doesn't even show up on my Travis account!?!

rhaschke, 2020-03-24:
github oauth services seem to be deprecated and we should switch to github apps:
https://github.com/ros-planning/moveit/settings/installations

rhaschke, 2020-03-25:
Seems to work again (see #1975 and #1972), but I didn't explicitly change anything...

davetcoleman, 2020-03-26:
The moveit repo disappeared from my travis account also :-/

gavanderhoorn, 2020-03-26:
@rhaschke wrote:
> github oauth services seem to be deprecated and we should switch to github apps:
> https://github.com/ros-planning/moveit/settings/installations

I believe you are correct.

The non-app integration has been deprecated for quite some time now.

This needs to be fixed at the organisation level.

rhaschke, 2020-03-26:
Seems to be still an issue, e.g. in https://github.com/ros-planning/moveit/pull/1964.

rhaschke, 2020-03-26:
What is really strange, that it seemed to work for master, but not for melodic.

davetcoleman, 2020-03-26:
I'm setting up Travis CI as a Github App now

rhaschke, 2020-03-27:
The branch protection still uses the [old method](https://travis-ci.community/t/known-issue-travis-ci-reports-expected-waiting-for-status-to-be-reported-on-the-github-status-api-but-the-status-never-arrives/1154):
[image]
But there is no other option!

Do we need to [migrate our repositories from travis-ci.org to travis-ci.com](https://docs.travis-ci.com/user/migrate/open-source-repository-migration) [here](https://travis-ci.com/organizations/ros-planning/migrate)? There, it is stated that this is a beta feature.

@gavanderhoorn, can you provide a hint, please?

gavanderhoorn, 2020-03-27:
I don't recall whether migration is required, but it would not hurt, as `travis-ci.org` is going away, iirc.

rhaschke, 2020-03-27:
Can you verify whether you migrated, i.e. do travis links bring you to travis-ci.**org** or travis-ci.**com**? I puzzled, why we don't see the GitHub App checks in branch settings.

gavanderhoorn, 2020-03-27:
I have migrated everything. What I meant was that I don't remember whether migration is required to get the app to work.

> I puzzled, why we don't see the GitHub App checks in branch settings.

Afaik it only shows up when a build has been triggered at least once. And for that to happen, it needs to be configured correctly.

But again: I don't know whether this is what is causing the problems with CI here at `ros-planning/moveit`.

davetcoleman, 2020-03-27:
We're switched to an app rather than the depreciated service (for Travis, not Docker yet):

[image]

I didn't know travis-ci.org is going away, does that mean the free accounts are going away also? PickNik currently pays for a .com business account for our internal projects, not sure if that is relevant or helpful.

gavanderhoorn, 2020-03-27:
Did you also configure the travis app and allow it access to the relevant repositories?

rhaschke, 2020-03-27:
Where do you configure it? Travis can see all the repositories and moveit is correctly configured I think:
[image]

gavanderhoorn, 2020-03-27:
Seems Travis has been having some trouble: [traviscistatus.com](https://www.traviscistatus.com/) (search for "March 26").

gavanderhoorn, 2020-03-27:
And that's a screenshot of `travis-ci.org`. I'm not an expert here, but if you go to https://github.com/organizations/ros-planning/settings/installations and click on *Configure* for `Travis CI` in that list, it should how you the list of repositories it's allowed to access.

Alternatively, go to `travis-ci.com`, login with your GH account and check the settings for `ros-planning/moveit` (or `ros-planning`, as it lists repositories by organisation).

rhaschke, 2020-03-28:
The MoveIt repos are definitely not yet migrated to travis-ci.com.
Looking at the mentioned [settings page](https://github.com/ros-planning/moveit/settings/installations), I cannot see a _Configure_ button. Obviously, I am missing permissions to configure those.
@davetcoleman, do you mind to grant me permissions for this as well:
https://github.com/organizations/ros-planning/settings/apps ?

davetcoleman, 2020-06-02:
Permission granted.

Can we close this issue? I assume we've figured out CI by now for the new approach?

rhaschke, 2020-06-02:
Sorry, @davetcoleman, but I still don't have permissions to configure github apps. Nothing changed yet, we are still running on the flaky (and deprecated) travis.org.

[image]

davetcoleman, 2020-06-02:
I did grant you the requested permission, but I'm not sure what else might be wrong:
[image]

rhaschke, 2020-06-02:
No idea.

davetcoleman, 2020-06-02:
I'm reading through [Beta - Migrating repositories to travis-ci.com](https://docs.travis-ci.com/user/migrate/open-source-repository-migration)

The issue is on this step:
> Select the repositories you’d like to migrate and click “Migrate selected repositories”, You’ll be asked for a final confirmation - please remember that you’ll need to update protected branches (See: Do I need to make any other changes?).

It says:
[image]

Other TODOs:
- [ ] “Protected Branches” feature in GitHub to require a passing Travis CI build before merging a Pull Request: make sure to edit your required status checks to now use Travis CI - Pull Request and/or Travis CI - Branch. See: Required status checks at GitHub.

v4hn, 2020-06-02:
Let's wait for the response of Travis Support then. Thanks for looking into it @davetcoleman

davetcoleman, 2020-06-02:
Just emailed support.

davetcoleman, 2020-06-02:
I got a response, they fixed something, I completed the migration. Not sure if our CI still works, looking into that now...

rhaschke, 2020-06-02:
Looks like migration was successful: https://travis-ci.com/github/ros-planning

davetcoleman, 2020-06-03:
Going to close this!

closed 2020-06-03 by davetcoleman