moveit/moveit #1974: PlanningScene::getFrameTransform on subframe on attached collision object is not wrt world

tylerjw, 2020-03-24:
### Description

This PR (https://github.com/ros-planning/moveit/pull/1757) demonstrates the issue.  

If you call getFrameTransform on a subframe of an attached collision object you get a relative transform instead of one wrt world like you do with collision objects that are not attached.  This is inconsistent with what `getFrameTransform` normally returns and the returned type `Eigen::Affine3d` has no way of expressing what frame the transform is with respect to.

v4hn, 2020-03-24:
Thank you for pointing this out.

Could you contribute a patch applying the corresponding link transform before returning the result?

tylerjw, 2020-03-24:
@v4hn where would be the right place to implement this?  Would it be in the `getFrameTransform` method?

tylerjw, 2020-03-25:
The problem stems from `AttachedBody::getSubframeTransform` returns a transform that is relative to the link it is attached to.  Is there a good way in there to get a the transform to the world there?  Or should it be done it `RobotState::getFrameInfo` where it gets called?

tylerjw, 2020-03-25:
I poked around trying to figure out where to make this change and haven't found any good options yet.  Maybe AttachedBody should be modified to store both relative and world transforms.  Thoughts?  @felixvd

felixvd, 2020-03-25:
Sorry, I looked at this yesterday too, but had no time to post. It's the `robot_state` that should return the world transform of the subframe, so after the `robot_state` retrieves the `AttachedBody`'s subframe transform ([here](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/src/robot_state.cpp#L1055) in line 1055), we should apply the transform of the parent link (`global_link_transforms_[robot_link->getLinkIndex()]`).

Changing the section like this should work, but the syntax and multiplication order may be incorrect:

```
[...]
const auto& transform = body.second->getSubframeTransform(frame_id, &frame_found);
if (frame_found)
{
  robot_link = body.second->getAttachedLink();
  transform = transform * global_link_transforms_[robot_link->getLinkIndex()];
[...]
```

tylerjw, 2020-03-25:
I tried something similar and it didn't work because they would create a temporary and then return a reference to that temporary variable.  The issue is because these are Eigen objects they can't be copy constructed so we have to store them somewhere and pass a reference to them.  With the current interface that returns a reference we need to find somewhere to store this value we are returning.  I'm thinking maybe AttachedBody should be the place to store them.

tylerjw, 2020-03-25:
From reading the code I think I need to modify `AttachedBody` for this.  I think the way I'm going to try to do is add a `global_subframe_poses_` member that is updated by `computeTransform` which updates `global_collision_body_transforms_` with a new transform for the new parent transform.  It will use the first transform in `global_collision_body_transforms_` and apply it to each of `subframe_poses_` to `global_subframe_poses_`.  I will then rename `getSubframeTransform` to `getGlobalSubframeTransform` as it will now use the global_subframe_poses_`.  I've attached some images from sourcetrail that helped me understand this.

Note that I can't just update `subframe_poses_` to be global transforms because I need the transform from the parent for each subframe to create the global ones.  Also there are outside methods that depend on the state of `subframe_poses_` through `getSubframeTransforms`.

[image]
[image]
[image]
[image]

tylerjw, 2020-03-25:
I updated my test PR with my proposed changes.  While they fix the transform to be global they now break something about how path planning works.  Somehow the logic of path planning depends on the subframe being a relative pose to the link it is attached to.  Here are my changes: https://github.com/ros-planning/moveit/pull/1757/commits/384563b1cd8f16b47c89629ce97a2ef2d6f631c0

@felixvd do you know what code is depending on the pose of subframes being relative?

felixvd, 2020-03-26:
It's almost certainly the `ResolveConstraintFrames` PlanningAdapter, but I can't check it myself at the moment

rhaschke, 2020-03-26:
> From reading the code I think I need to modify AttachedBody for this. I think the way I'm going to try to do is add a global_subframe_poses_ member that is updated by computeTransform which updates global_collision_body_transforms_ with a new transform for the new parent transform.

While caching global frames of shapes is important for (frequent) collision checking, I don't think we need to cache the global frames of virtual sub frames. Those are only seldomly used, namely to assemble planning constraints.
I will propose an alternative solution following the lines of https://github.com/ros-planning/moveit/issues/1974#issuecomment-603646342 (and indeed requiring an API change to return a copy instead of a reference).
I will also fix the `ResolveConstraintFrames` adapter.

rhaschke, 2020-03-26:
Hm. Changing API from returning a const-reference to a copy is not trivial as well, as dozens of other functions rely on the existing API...

tylerjw, 2020-03-26:
@rhaschke is there not a an issue created when you return a Eigen object, causing it to be copy constructed that makes it out of memory alignment?

tylerjw, 2020-03-26:
> While caching global frames of shapes is important for (frequent) collision checking, I don't think we need to cache the global frames of virtual sub frames. Those are only seldomly used, namely to assemble planning constraints.

Are you concerned about the memory cost of storing all of them?  And/or the time cost of updating them?  One way to reduce the cost but keep the same API would be to have a `Eigen::Isometry3d last_requested_global_subframe` that is used when one is requested?

rhaschke, 2020-03-26:
> Is there not an issue created when you return an Eigen object, causing it to be copy constructed that makes it out of memory alignment?

No, construction of Eigen objects takes care of aligned memory allocation.

> Are you concerned about the memory cost of storing all of them? And/or the time cost of updating them?

I am concerned about code complexity and update costs. Virtual frames are not frequently accessed. Thus updating them regularly in `computeTransform()` would waste resources. In my point of view, this is the natural location to update them.

> One way to reduce the cost but keep the same API would be to have an Eigen::Isometry3d last_requested_global_subframe that is used when one is requested?

Yes, but this solution will not be thread-safe. No option in my point of view.

v4hn, 2020-03-26:
> No, construction of Eigen objects takes care of aligned memory allocation.

I was also under the impression this was not valid, but I believe the respective documentation in Eigen changed at some point: http://eigen.tuxfamily.org/dox/group__TopicPassingByValue.html

rhaschke, 2020-03-26:
In the very bottom of the referenced page, it is stated:
  _Note that on the other hand, there is no problem with functions that return objects by value._

v4hn, 2020-03-26:
That's exactly why I linked the page :sunglasses: 

Whether it would be a good idea to change the interfaces to return by value is a different question I'm not a fan of it and I don't see a problem with @tylerjw's proposal to add a global transform field to the `AttachedBody` structure.
If you have an object attached, you are usually *very much* interested in its pose, otherwise you would not usually attach it.

rhaschke, 2020-03-26:
> If you have an object attached, you are usually very much interested in its pose, otherwise you would not usually attach it.

I was not questioning this. However, I don't think we need to update the global transform of the subframes each and every time the robot moves, because these subframe poses are only _rarely_ (I didn't say _never_) accessed.

v4hn, 2020-03-26:
> I don't think we need to update the global transform of the subframes each and every time the robot moves, because these subframe poses are only rarely (I didn't say never) accessed.

I agree that subframes should not be updated by [`updateCollisionBodyTransforms`](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/include/moveit/robot_state/robot_state.h#L1280), but surely we want to update them by [`updateLinkTransforms`](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/include/moveit/robot_state/robot_state.h#L1284).
The whole point of the latter method is to allow arbitrary frame lookup in constant time afterwards.

tylerjw, 2020-03-26:
@rhaschke @v4hn what are your thoughts on changing the API to return a temporary transform instead of a reference to one?  

How much user code would this actually break?

tylerjw, 2020-03-26:
If the solution provided by @rhaschke is approved in my PR close this issue.

tylerjw, 2020-04-20:
This was fixed in #1757, thank you for your help @rhaschke

closed 2020-04-20 by tylerjw