moveit/moveit #1976: Enable test_abi for melodic, kinetic releases?

davetcoleman, 2020-03-26:
Does anyone disagree with turning on ABI checking in Bloom for future releases of MoveIt 1 (not master)?

See [Discourse post](https://discourse.ros.org/t/enabling-the-abi-checker-for-your-next-ros2-f-releases/12811) from @thomas-moulard

The change appears extremely easy:
https://github.com/ros/rosdistro/pull/23281/files

In the future we should do this for MoveIt 2, also.

v4hn, 2020-03-26:
This is a really, really nice feature that was pointed out to me today too.

For our current release system with MoveIt, it does not make sense to activate it though.
We *always* break ABI *everywhere*. We change the SONAME with every release, in order to allow for ABI breakage between versions.

Some people might hate us for doing this, but the alternative would be
- a lot of unpaid overhead to explicitly mark ABI changes in every single sub-library by bumping the SONAME there
- or breaking ABI without marking it. The new buildfarm check would find the break (and rebuild all downstream dependencies, but *user*code, not maintained on the buildfarm, would silently misbehave or segfault.

The alternative policy encouraged by the new check, to *avoid* ABI breakage within a release altogether, is simply no option for MoveIt where we regularly have to break ABI for safety or correctness.

For geometric_shapes and srdfdom, on the other hand, they are very nice to have.

rhaschke, 2020-03-26:
To add to Michael's comments: I'm not sure how fast/slow the check on the build farm would be.
In the past, on Travis, I experienced run times of >1,5 hours (due to the monolithic repo).
For smaller repos, I support this feature as well.

thomas-moulard, 2020-03-26:
Just a precision here: the flag controls the report generation, it's not actually blocking anything AFAIK. @j-rivero, could you confirm please?

Here is the report this generates: http://build.ros2.org/job/Epr__rcl__ubuntu_bionic_amd64/API_5fABI_20report/

You could use it, for instance, to generate better release notes (between X and Y, beware we broke A, B, and C).

v4hn, 2020-03-26:
> Just a precision here: the flag controls the report generation, it's not actually blocking anything AFAIK.

I do hope so. Actually, it *should* be blocked unless the library's SONAME changed (which is the correct way to handle ABI breakage, if it has to happen). But very few packages/maintainers in ROS seem to know about that and version their libraries, so it's hard to force people to do that.

But it does not matter for MoveIt unless we change our attitude to ABI-breakage again.
The only thing the test would tell us is that ABI is broken *everywhere* *on every release* and that is on purpose because we bump all SONAMEs. Thus, there is currently no point in using this for MoveIt.

All that aside, the style of "better" release notes you propose is only marginally better than not announcing it at all! Everyone compiling downstream dependencies for bigger systems rebuilds their code bases anyway after a sync because someone might *not* add their breakage to the notes. Anyone who experiences weird segfaults will try to rebuild before anything else.

The neat upside of the automated ABI check is that, in theory, the buildfarm would not have to recompile downstream dependencies if the ABI is not violated in a new release.
I don't think anyone integrated that yet though.
Aside from forcing users to do ABI changes properly, the buildfarm maintainers could also actually list all packages that broke ABI on every sync to the public repository!
That information could, in theory, be used by downstream maintainers to decide what to rebuild.

davetcoleman, 2020-06-02:
Sounds like the consensus is that its not worth the effort / noise. This isn't an actionable issue anymore, closing.

closed 2020-06-02 by davetcoleman

j-rivero, 2020-06-05:
> Just a precision here: the flag controls the report generation, it's not actually blocking anything AFAIK. @j-rivero, could you confirm please?

I don't have any news that we are blocking anything based in the results of ABI checker. All 'damage' it can do is to [turn your builds unstable](https://github.com/ros-infrastructure/ros_buildfarm/blob/master/scripts/devel/build_and_install.py#L118-L127) if it detects problems.