moveit/moveit #1988: Stack smashing detected when following 'Getting Started', with Debug mode

roversch, 2020-03-27:
### Description

I follow the 'Getting Started' instructions (https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html). In there, you need to build moveit and related packages from source. I carry out everything as mentioned, except that I configure with `-DCMAKE_BUILD_TYPE=Debug`. With build type `Release`, the tutorial works as expected.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch? master for moveit_tutorials, melodic for panda_moveit_config

### Steps to reproduce
```
cd ~/ws_moveit
catkin config --extend /opt/ros/${ROS_DISTRO} --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin build
source ~/ws_moveit/devel/setup.bash
roslaunch panda_moveit_config demo.launch
```
From a second shell:
```
roslaunch moveit_tutorials move_group_interface_tutorial.launch 
```

### Expected behaviour
The RViz window comes up, and `move_grou_interface_tutorial` prompts to 'click next'.

### Actual behaviour
In shell 2:
Process hangs at
```
[ INFO] [1585318303.782749026]: Loading robot model 'panda'...
```

In shell 1:
```
[ INFO] [1585318260.841025484]: Listening to '/planning_scene_world' for planning scene world geometry
*** stack smashing detected ***: <unknown> terminated
[move_group-6] process has died [pid 20730, exit code -6, cmd /home/robin/ABB_FutureLabs_Projects/powder2product/ws_moveit/devel/lib/moveit_ros_move_group/move_group --debug /joint_states:=/joint_states_desired __name:=move_group __log:=/home/robin/.ros/log/c84a05ca-7034-11ea-847c-e4a47194286b/move_group-6.log].
log file: /home/robin/.ros/log/c84a05ca-7034-11ea-847c-e4a47194286b/move_group-6*.log
[ INFO] [1585318261.495665111]: Stereo is NOT SUPPORTED
[ INFO] [1585318261.495772293]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1585318261.669318931]: Loading robot model 'panda'...
[ INFO] [1585318262.028533395]: Loading robot model 'panda'...
[ INFO] [1585318262.132940196]: Starting planning scene monitor
[ INFO] [1585318262.139427307]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1585318262.150580937]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ INFO] [1585318267.169468441]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
```

roversch, 2020-03-27:
It is related to #588, but I did try re-installing `libpcl-dev`, to no avail.

rhaschke, 2020-03-27:
Which MoveIt version to you use? The referenced master branch of moveit_tutorials requires the master branch of MoveIt as well. Otherwise go with melodic-devel for both.

roversch, 2020-03-27:
Yes, `master` branch of `moveit` as well.

roversch, 2020-03-27:
I can also offer a backtrace : 
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff536f801 in __GI_abort () at abort.c:79
#2  0x00007ffff53b8897 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff54e5988 "*** %s ***: %s terminated\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff5463cd1 in __GI___fortify_fail_abort (need_backtrace=need_backtrace@entry=false, msg=msg@entry=0x7ffff54e5966 "stack smashing detected")
    at fortify_fail.c:33
#4  0x00007ffff5463c92 in __stack_chk_fail () at stack_chk_fail.c:29
#5  0x00007ffff196df6e in Eigen::DenseBase<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<double, double>, Eigen::Matrix<double, 3, 1, 0, 3, 1> const, Eigen::Matrix<double, 3, 1, 0, 3, 1> const> const> >::redux<Eigen::internal::scalar_sum_op<double, double> > (
    this=0x7fffffffc260, func=...) at /usr/local/include/eigen3/Eigen/src/Core/Redux.h:419
#6  0x00007ffff196759a in Eigen::DenseBase<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<double, double>, Eigen::Matrix<double, 3, 1, 0, 3, 1> const, Eigen::Matrix<double, 3, 1, 0, 3, 1> const> const> >::sum (this=0x7fffffffc260)
    at /usr/local/include/eigen3/Eigen/src/Core/Redux.h:455
#7  0x00007ffff1962618 in Eigen::MatrixBase<Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<double, double>, Eigen::Matrix<double, 3, 1, 0, 3, 1> const, Eigen::Matrix<double, 3, 1, 0, 3, 1> const> >::squaredNorm (this=0x7fffffffc340) at /usr/local/include/eigen3/Eigen/src/Core/Dot.h:98
[ INFO] [1585318797.655534089]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
#8  0x00007ffff19bf5df in shapes::Mesh::mergeVertices (this=0x555555e15c20, threshold=0.0001)
    at /home/<>/ws_moveit/src/geometric_shapes/src/shapes.cpp:535
#9  0x00007ffff77f3c62 in planning_scene_monitor::PlanningSceneMonitor::excludeRobotLinksFromOctree (this=0x55555584db50)
    at /home/<>/ws_moveit/src/moveit/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:688
#10 0x00007ffff77f865a in planning_scene_monitor::PlanningSceneMonitor::startWorldGeometryMonitor (this=0x55555584db50, collision_objects_topic="collision_object", 
---Type <return> to continue, or q <return> to quit---
    planning_scene_world_topic="planning_scene_world", load_octomap_monitor=true)
    at /home/<>/ws_moveit/src/moveit/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:1076
#11 0x00005555555ad328 in main (argc=2, argv=0x7fffffffcac8)
    at /home/<>/ws_moveit/src/moveit/moveit_ros/move_group/src/move_group.cpp:212
```

rhaschke, 2020-03-27:
From the backtrace, I guess an Eigen an alignment error. Could you have a look at https://github.com/ros-planning/moveit/issues/1385 and validate your optimization options, e.g. not using `-march=native` optimization?

roversch, 2020-03-27:
I can confirm, I added `-v` to the `g++` flags, looks like `mtune=generic` and `march=x86-64`.

roversch, 2020-04-01:
I'm not afraid to debug this, but I will need your guidance. Where do I start?

roversch, 2020-04-01:
I found the problem. I was linking to a built-from-source Eigen library.

Now, the stack smashing is not there.

closed 2020-04-01 by roversch

rhaschke, 2020-04-01:
What do you mean by this? Eigen is a header-only library.

roversch, 2020-04-02:
Right, I meant it's including the wrong Eigen header libraries, installed
in /usr/local/include. The right ones seem to be the ones in /usr/include

rhaschke, 2020-04-02:
Yes, it is particularly important that all pieces of software use the very same headers.

gavanderhoorn, 2020-04-02:
Indeed: [ethz-asl/eigen_catkin/wiki/Eigen-Memory-Issues](https://github.com/ethz-asl/eigen_catkin/wiki/Eigen-Memory-Issues).

haozhang666, 2021-12-11:
Hello, I met the same mistake as you, my version is MOVEIt under Noetic

May I ask whether the version of Eigen is wrong? How do you solve it? Thank you