moveit/moveit #1990: Releases?

v4hn, 2020-03-27:
According to [December's meeting notes](https://discourse.ros.org/t/moveit-maintainer-meeting-dec-5th/11501/3), @JafarAbdi was supposed to take over the release process and we wanted to aim for more regular releases.

Now we get [bug reports about features merged in July 2019 not being available](https://github.com/ros-planning/moveit/issues/1989).
Additionally, I believe there are dozens of minor patches in master that should be backported before release.

@davetcoleman @rhaschke How do we proceed here?

rhaschke, 2020-03-27:
Please identify a list of potential candidates for back porting, back port them on a separate PR branch, test. Then I can trigger a release. I don't have the resources to do this alone. These days my inbox fills faster than I can work on it...

v4hn, 2020-03-27:
> Please identify a list of potential candidates for back porting, back port them on a separate PR branch, test. Then I can trigger a release

I know you are busy @rhaschke and I was actually hoping for support from one of the other maintainers (maybe one of those who actually get paid for working with the code :sunglasses: ).

It is simply not an acceptable state to not have a single release in over 9 months - especially with all the minor changes we did in the meanwhile.

jschleicher, 2020-03-31:
Cross-reference: see also https://github.com/ros-planning/moveit/issues/1662#issuecomment-547982618

> @JafarAbdi was supposed to take over the release process and we wanted to aim for more regular releases.

tylerjw, 2020-04-07:
Thank you @v4hn for getting the last 9 months of patches merged.  Did we kick off a release or are there still bugs we want to fix before?

rhaschke, 2020-04-07:
Today, we agreed that I will trigger a new Melodic release as soon as the next sync is out (@clalancette, did you already schedule it?)
In this fashion, we will have maximal time in ros-testing. We also have a few small bugs reported, with fixups pending, e.g. #2014.

clalancette, 2020-04-08:
> (@clalancette, did you already schedule it?)

Not yet; the buildfarm is in pretty rough shape: http://repositories.ros.org/status_page/ros_melodic_default.html?q=REGRESSION.  I'll probably schedule one in a week or two.

davetcoleman, 2020-04-22:
> Cross-reference: see also #1662 (comment)

@jschleicher for context on that offered commitment from PickNik, the following post from @v4hn argued @JafarAbdi was not qualified to do the task https://github.com/ros-planning/moveit/issues/1662#issuecomment-547994611

> This requires maintainer-level insight into the framework.

Thus it has remained undone for so lone

jschleicher, 2020-04-22:
@davetcoleman Thanks for the clarification.

v4hn, 2020-04-23:
I don't see me mention any names in this post :smile_cat: 
The maintainer meeting (without my participation) decided things two months later,
so I considered this settled until I noticed nothing happened.

Anyway, I guess we will talk about the release process later today in the meeting and it all boils down to "do we want to backport things to melodic?" and "can we release `master` for our user base?"

rhaschke, 2020-04-26:
Melodic release triggered: https://github.com/ros/rosdistro/pull/24607.
Let's hope, we don't see too many issues. I will not have time to fixed them on short notice...

v4hn, 2020-04-27:
Thank you for your work @rhaschke , esp. also further refinements for the `geometric_shapes` requests!

It's nice to see releases with actual progress. :)

rhaschke, 2020-04-27:
Our migration of `occupancy_map_monitor` (#1533) causes some issues on debian updates:
It looks like the new `occupancy_map_monitor` package is installed before the old one (`ros_perception`) is updated, causing a file conflict:
```
Unpacking ros-melodic-moveit-ros-occupancy-map-monitor (1.0.3-1bionic.20200427.184432) ...
1107dpkg: error processing archive /tmp/apt-dpkg-install-PwfXlk/07-ros-melodic-moveit-ros-occupancy-map-monitor_1.0.3-1bionic.20200427.184432_amd64.deb (--unpack):
1108 trying to overwrite '/opt/ros/melodic/include/moveit/occupancy_map_monitor/occupancy_map.h', which is also in package ros-melodic-moveit-ros-perception 1.0.2-1bionic.20200406.134557
```
Thus a question to our debian package experts (@JafarAbdi?):
Can we impose a specific update order in the .deb update process?
If so, how can we convey this information via the release repository?

jspricke, 2020-04-28:
You can probably add a `<conflict version_lt="1.0.3~">moveit_ros_perception</conflict>` in your package.xml

rhaschke, 2020-04-29:
@jspricke, thanks for this hint. What is the meaning of `~` in the version specifier?
Can you point me to some documentation on this?

jspricke, 2020-04-29:
It means anything before that version, see: https://www.debian.org/doc/debian-policy/ch-controlfields.html#s-f-version

rhaschke, 2020-04-29:
Thanks a lot, @jspricke. I triggered a new release: https://github.com/ros/rosdistro/pull/24637

rhaschke, 2020-04-30:
Looks like the new release rolls out without problems. :tada:

gavanderhoorn, 2020-04-30:
You probably already realised, but I believe the syntax @jspricke suggested is Debian/Ubuntu only.

There is quite some work underway to setup a CentOS/RHEL extension to the buildfarm, so this may not work in the future.

Not a problem right now, but something to keep in mind.

jspricke, 2020-04-30:
Yeah, actually the tilde should not be necessary, which would make it work for CentOS/RHEL as well, right?

davetcoleman, 2020-05-06:
Thanks for making the Melodic release @rhaschke! Can this issue be closed?

rhaschke, 2020-05-06:
We usually keep the issue open until the actual sync happened.

rhaschke, 2020-05-28:
I just noticed an issue with the generated ikfast plugins. @clalancette, can you please hold back the next sync a few more days (I know, the release is pending for a while already)?
I will try to look into this asap.

clalancette, 2020-05-28:
> I just noticed an issue with the generated ikfast plugins. @clalancette, can you please hold back the next sync a few more days (I know, the release is pending for a while already)?

Sorry, I didn't see this until right now, and I did the sync this morning.  Go ahead and make the fixes as needed, and once we merge the rosdistro PR we can think about doing a quick turnaround sync for melodic.

closed 2020-09-03 by v4hn