moveit/moveit #1994: build moveit_ros_planning_interface, it shows can't find numpy/numpyconfig.h in ~/catkin_ws/install/include/eigenpy

qintianhaohao, 2020-03-28:
### Description

i try build moveit from source, but it shows need `eigenpy` when build moveit_ros_planning_interface. 
firstly i try install eigenpy by: `sudo apt install robotpkg-py27-eigenpy`, but it don't make sence.

so i build eigenpy from source according to @lucarinelli 's method and succeed:
<https://github.com/ros-planning/moveit/issues/1850>

but when i build `moveit_ros_planning_interface` again, it shows can't find numpy/numpyconfig.h in ~/catkin_ws/install/include/eigenpy.

so i try find numpyconfig.h use `locate numpyconfig.h`, and i got this:
`/usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/_numpyconfig.h`
`/usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/numpyconfig.h`

then i copy folder `/usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy` to `/home/tianbot/catkin_ws/install/include`, and rebuild moveit, finally succeed.

so why we need build eigenpy from source before build moveit?

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Source and moveit master
* python 2.7

rhaschke, 2020-03-29:
Just follow the official [install instructions from source](https://moveit.ros.org/install/source), which means to use rosdep to install all MoveIt dependencies, including eigenpy, from debian packages.
This will install eigenpy from this catkinized package: `ros-melodic-eigenpy`.
Note that the master branch builds on both, ROS Kinetic and Melodic:
https://travis-ci.org/github/ros-planning/moveit/builds/667773878

One more side note: Currently, the [eigenpy package is broken](https://github.com/ros-planning/moveit/issues/1996) in ROS Melodic, when installing it from the [ros-testing repos](http://wiki.ros.org/action/show/TestingRepository?action=show&redirect=ShadowRepository).

closed 2020-03-29 by rhaschke

huangjiancong1, 2020-06-15:
> ### Description
> i try build moveit from source, but it shows need `eigenpy` when build moveit_ros_planning_interface.
> firstly i try install eigenpy by: `sudo apt install robotpkg-py27-eigenpy`, but [...]

I  also miss this issue when follow (https://moveit.ros.org/install/source/) to compiler the moveIt. Have you fixed it?

huangjiancong1, 2020-06-15:
> ### Description
> i try build moveit from source, but it shows need `eigenpy` when build moveit_ros_planning_interface.
> firstly i try install eigenpy by: `sudo apt install robotpkg-py27-eigenpy`, but [...]

You need to follow the installation about eigenpy (https://github.com/stack-of-tasks/eigenpy) to install it first.

rhaschke, 2020-06-15:
Just install `ros-melodic-eigenpy`! This should be automatically installed via rosdep when following the _Getting started_ instructions from the MoveIt tutorials.