moveit/moveit #1996: master failing CI on clang

tylerjw, 2020-03-29:
### Description

Someone snuck a bug past CI (or an external dependency got us somehow).  Here is a link to a CI run that demonstrates it (This CI run is from my PR that adds an integration test for move_group, my test is not failing in this run):

https://travis-ci.org/github/ros-planning/moveit/jobs/668233574?utm_medium=notification&utm_source=github_status

These tests are failing:
```
build/moveit_commander/test_results/moveit_commander/rostest-test_python_moveit_commander.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/moveit_commander/test_results/moveit_commander/rostest-test_python_moveit_commander_ns.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/moveit_commander/test_results/moveit_commander/rostest-test_python_time_parameterization.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/moveit_core/test_results/moveit_core/gtest-test_bullet_continuous_collision_checking.xml: 8 tests, 0 errors, 0 failures, 2 skipped
build/moveit_ros_planning_interface/test_results/moveit_ros_planning_interface/rostest-test_python_move_group.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/moveit_ros_planning_interface/test_results/moveit_ros_planning_interface/rostest-test_python_move_group_ns.xml: 2 tests, 0 errors, 2 failures, 0 skipped
build/moveit_ros_planning_interface/test_results/moveit_ros_planning_interface/rostest-test_robot_state_update.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/moveit_ros_planning_interface/test_results/moveit_ros_planning_interface/rosunit-cleanup_tests.xml: 2 tests, 0 errors, 1 failures, 0 skipped
Summary: 481 tests, 0 errors, 8 failures, 2 skipped
```

rhaschke, 2020-03-29:
These failing tests are related to an update of eigenpy that is broken: https://github.com/ros/rosdistro/pull/24254#issuecomment-605435738

tylerjw, 2020-04-07:
This seems to be working now.

closed 2020-04-07 by tylerjw