moveit/moveit #2006: Codecov needs login

gleichdick, 2020-04-01:
### Description

Login to Codecov is required to analyze the codecov report generated by the pipeline.
Otherwise, HTTP 403 occurs.

### Steps to reproduce

Open a private browser window and click on any link in one of the latest [codecov reports](https://github.com/ros-planning/moveit/pull/1983#issuecomment-607150995).

gleichdick, 2020-04-02:
Tested it again, seems to work again

gleichdick, 2020-04-02:
Nope, broken again :thinking: 
[image]

gavanderhoorn, 2020-04-02:
It's always been like this for me, at least here in `ros-planning`.

tylerjw, 2020-04-07:
This seems like an issue with how the codecov website is trying to get you to sign up using your github more than an issue with how we are using it.  Would you suggest an alternative to codecov.io that doesn't gate viewing details of the reports?  Atleast codecov.io is free for open source projects like ours and makes nicely formatted comments on PRs that you can view without signing in to their site.

gleichdick, 2020-04-08:
Ah, thanks, I didn't know that codecov does that in general.

closed 2020-04-08 by gleichdick