moveit/moveit #2009: Using getGlobalLinkTransform() for a RobotState object get wrong results.

lijiashushu, 2020-04-03:
I try to get the result of the getGlobalLinkTransform(). But I find that for a state with same joint angles, I get different results.  It seems that the values of some positions of the returned matrix are wrong.  I don't know why this happens. I creat three RobotState instances, set the same angles and ask for the pose of the same link, but I got the results like this.
```
robot_state::RobotState open_state(_robot_model_ptr);
    robot_state::RobotState start_state(_robot_model_ptr);
    robot_state::RobotStatePtr start_state_ptr(new robot_state::RobotState(_robot_model_ptr));
    open_state.setJointGroupPositions("both_arms", _startAngles);
    start_state.setJointGroupPositions("both_arms", _startAngles);
    start_state_ptr->setJointGroupPositions("both_arms", _startAngles);

    open_state.update();
    ros::Duration(0.5).sleep();
    start_state.update();
    ros::Duration(0.5).sleep();
    start_state_ptr->update();
    ros::Duration(0.5).sleep();

    std::vector<double> tmp1;
    std::vector<double> tmp2;
    std::vector<double> tmp3;
    open_state.copyJointGroupPositions(_both_group, tmp1);
    start_state.copyJointGroupPositions(_both_group, tmp2);
    start_state_ptr->copyJointGroupPositions(_both_group, tmp3);
    for(int i=0; i<tmp1.size(); i++){
        std::cout<<tmp1[i];
    }
    std::cout<<std::endl;
    for(int i=0; i<tmp2.size(); i++){
        std::cout<<tmp2[i];
    }
    std::cout<<std::endl;
    for(int i=0; i<tmp3.size(); i++){
        std::cout<<tmp3[i];
    }
    std::cout<<std::endl;

    std::cout<<"test1 "<<open_state.dirty()<<" "<<(open_state.getGlobalLinkTransform("right_gripper").translation()).eval().transpose()<<std::endl;
    std::cout<<"test2 "<<start_state.dirty()<<" "<<(start_state.getGlobalLinkTransform("right_gripper").translation()).eval().transpose()<<std::endl;
    std::cout<<"test3 "<<start_state_ptr->dirty()<<" "<<(start_state_ptr->getGlobalLinkTransform("right_gripper").translation()).eval().transpose()<<std::endl;

![1](https://user-images.githubusercontent.com/34027307/78352564-1b07b900-75db-11ea-9a7b-cc05914c9285.png)
![2](https://user-images.githubusercontent.com/34027307/78352571-222ec700-75db-11ea-98f3-7d93749db0ef.png)
![3](https://user-images.githubusercontent.com/34027307/78352579-24912100-75db-11ea-95b6-c550c770a70a.png)

```

lijiashushu, 2020-04-03:
[image]
[image]
[image]

rhaschke, 2020-04-03:
Did you compile MoveIt from source? If not, did you use the same Eigen version as MoveIt? Which optimization options did you use?
There are known issues with Eigen and too strong optimization: https://github.com/ros-planning/moveit/issues/1902

lijiashushu, 2020-04-03:
I did compliled MoveIt from source in kinetic-devel.  And  I didn't use any optimiztion options. The founded Eigen version in my enviorment is 3.2.92

v4hn, 2020-04-03:
> I did compile MoveIt from source in kinetic-devel. And I didn't use any optimiztion options.
> The founded Eigen version in my enviorment is 3.2.92

That probably means you compiled the `kinetic-devel` branch?
Could you try to setup MoveIt's `master` branch and verify you can reproduce your problem with the latest version?
It should compile on Ubuntu 16.04/kinetic if you compile the current versions of its direct dependencies (moveit_msgs, [ompl](https://moveit.ros.org/install/source/dependencies/), srdfdom, geometric_shapes) together in the same workspace.

lijiashushu, 2020-04-03:
I get the compile error about eigen3. 

In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /opt/ros/kinetic/include/eigen_stl_containers/eigen_stl_vector_container.h:40,
                 from /opt/ros/kinetic/include/eigen_stl_containers/eigen_stl_containers.h:40,
                 from /home/lijiashushu/ws_moveit/src/geometric_shapes/include/geometric_shapes/bodies.h:45,
                 from /home/lijiashushu/ws_moveit/src/moveit/moveit_core/collision_distance_field/include/moveit/collision_distance_field/collision_distance_field_types.h:47,
                 from /home/lijiashushu/ws_moveit/src/moveit/moveit_core/collision_distance_field/src/collision_distance_field_types.cpp:37:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/lijiashushu/ws_moveit/src/moveit/moveit_core/collision_distance_field/src/collision_distance_field_types.cpp:55:22:   required from here
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:330:5: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
moveit/moveit_core/collision_distance_field/CMakeFiles/moveit_collision_distance_field.dir/build.make:62: recipe for target 'moveit/moveit_core/collision_distance_field/CMakeFiles/moveit_collision_distance_field.dir/src/collision_distance_field_types.cpp.o' failed
make[2]: *** [moveit/moveit_core/collision_distance_field/CMakeFiles/moveit_collision_distance_field.dir/src/collision_distance_field_types.cpp.o] Error 1
CMakeFiles/Makefile2:12916: recipe for target 'moveit/moveit_core/collision_distance_field/CMakeFiles/moveit_collision_distance_field.dir/all' failed
make[1]: *** [moveit/moveit_core/collision_distance_field/CMakeFiles/moveit_collision_distance_field.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

lijiashushu, 2020-04-04:
I found the problems that  I ignored before. I created a RobotState instance from the RobotModel, but at first I didn't notice that I have to call setToDefaultValues().  After calling it, the problem is fixed.

rhaschke, 2020-04-05:
Nice to hear. Closing the issue now.

closed 2020-04-05 by rhaschke

jih189, 2022-11-07:
I tried setToDefaultValues. It does not work for me.