moveit/moveit #2015: Noetic Ninjemys Release

tylerjw, 2020-04-07:
Do we have a good place to track what work we want to do before Noetic other than an issue here?  If so, close this.

This issue should track three things:
* Bug fixes that should be given priority to be finished before release.
* Maintenance tasks (new branches, tutorial and ros website updates).
* New features we want to polish up and create release notes about.
* Testing done on ROS Noetic (including travis) on Ubuntu 20.04 release candidate.

Please comment with issue links and I'll update the main description here to have a checklist of tasks needed to cut the Noetic release.

Noetic Release Taks:
- [ ] Populate this list with bugs that need squashed
- [ ] noetic_devel branches for moveit, moveit_tutorials, moveit_msgs
- [ ] rosdistro updates
- [ ] travis tests
- [ ] Write release notes
- [ ] Dog-fooding on on 20.04 rc
- [ ] Kick of release build
- [ ] Discord announcement

Here is the migration page to help with getting up to date with the latest changes and what we might need to change in MoveIt to be compatible: http://wiki.ros.org/noetic/Migration

rhaschke, 2020-04-07:
One major point for Noetic migration is the transition to Python3. @v4hn and his group have already done a lot of work into this direction. However, I'm not sure about the current state.
Another major point is to have all the upstream packages released beforehand. OSRF is gently pushing maintainers to do so... [Here](http://repositories.ros.org/status_page/blocked_releases_noetic.html?q=id%3D%22moveit%22&s=6&r=1) is the list of blocking packages.

v4hn, 2020-04-24:
On Tue, Apr 07, 2020 at 03:29:35PM -0700, Robert Haschke wrote:
> One major point for Noetic migration is the transition to Python3.
> @v4hn and his group have already done a lot of work into this direction.
> However, I'm not sure about the current state.

I sent small patches for python2/3 compatibility in the past, but only for problems I actually encounter.
And I do not use the python APIs a lot.

I believe there are quite a few undetected issues still lingering around.

CI support for noetic and an extended python test suite would help a lot.

davetcoleman, 2020-05-06:
Thanks @tylerjw for creating this issue! 
I've just [updated the homepage](https://github.com/ros-planning/moveit.ros.org/pull/438) with a target release date of June 19th, which is roughly 1 month after the ROS distro comes out. Aspirational maybe...

I think a good next step is to ask people to volunteer various tasks, and we put their user names next to the checkbox items above. I'm happy to take on branching the moveit_tutorials, and can branch everything else also if needed. I've done this in the past.

>  Populate this list with bugs that need squashed

This seems idealistic and not a necessary blocker for getting Noetic out on time. Focusing on required things like Python3 is enough scope already, no?

tylerjw, 2020-05-06:
> This seems idealistic and not a necessary blocker for getting Noetic out on time. Focusing on required things like Python3 is enough scope already, no?

The list doesn't have to be extensive.  The intent of that bullet to make sure we scope out the bugs that we will fix before the release to avoid us slipping on the release because we want to wait for just-one-more-thing.  My goal was to figure out exactly what the work involved was so it was easier to defend against scope creep.  Fixing Python3 compatibility is a feature we'll need for this and if we agree that is the scope of fixes I'll mark this one as complete.

I'll put you down for the branching and take the responsibility of CI infrastructure changes to run in 20.04.  This should find all the areas where we have existing testing of python code that isn't python3 compatible yet.

tylerjw, 2020-05-26:
> Finally, we need to find a new clang-format configuration as clang-format-3.9 is not available anymore in Focal and newer versions reformat virtually everything using the current config.

Should we make this the point where we apply google style clang-format to moveit.  That would make keeping future changes in sync with ROS2 development nicer as ROS2 could start using ament_clang_foramt which has this .clang-format file: https://github.com/ament/ament_lint/blob/master/ament_clang_format/ament_clang_format/configuration/.clang-format

rhaschke, 2020-05-26:
I think we should keep the formatting changes as minimal as possible. If ament_lint supports clang-format, I guess/hope that the can handle custom `.clang-format` files as well!
I already made the clang-format-10 transition for rviz: https://github.com/ros-visualization/rviz/commit/ebfbb1f5e3281f969966dc9c305b5afc444cb79c

tylerjw, 2020-05-26:
Oh, that doesn't seem so bad.  I had assumed based on your statement that it would be a huge change.  If you think the same .clang-format you used for rviz will work we should apply it here and switch to clang-format-10 on master.

rhaschke, 2020-05-26:
We need to try the clang-format config here and check whether we can further tweak it to reduce reformatting.

davetcoleman, 2020-06-02:
When should we start forking the branches and tutorials? We are currently scheduled for the [feature freeze on 1 July 2020](https://moveit.ros.org/), but we probably need the branches created before then to include various changes? 

Ideally we just work out of master until that point, IMHO. We might have to use pragmas to keep support for Melodic.

To make this manageable, we should probably drop support for Kinetic on ``master`` now.

mikeferguson, 2020-06-02:
I've created a PR for python3-pyassimp key to rosdistro: https://github.com/ros/rosdistro/pull/25263

mikeferguson, 2020-06-02:
This should take care of the rosdep keys needed for Noetic/Python3 as well as the orocos_kdl rosdep key change (it's a system dependency now): https://github.com/ros-planning/moveit/pull/2131

davetcoleman, 2020-06-02:
I've emailed Ruben and created this Github issue: https://github.com/orocos/orocos_kinematics_dynamics/issues/252

mikeferguson, 2020-06-02:
> I've emailed Ruben and created this Github issue: [orocos/orocos_kinematics_dynamics#252](https://github.com/orocos/orocos_kinematics_dynamics/issues/252)

@davetcoleman Orocos_kdl is now a system dependency - it will not be released as a ROS package in noetic, so the rosdep key has changed and the blocked releases page lies (but the fix we need is in #2131).

I just finished successfully building moveit on Noetic - will post shortly how it actually runs

mikeferguson, 2020-06-02:
Blocked releases page is a little out of date (since orocos_kdl is a system dependency now, and we have added some dependencies since the last melodic release):
 * [ ] ompl
 * [ ] srdfdom
 * [ ] rosparam_shortcuts
 * [ ] ros_pytest

mikeferguson, 2020-06-02:
So far I've tested the following (in a source install directly on 20.04):

 * running moveit_setup_assistant (both from an existing config and a new config from URDF) - OK
 * running the demo.launch generated - this doesn't work - get the following error (but it might be my config - maybe something isn't setup right?):
   ```
   load_parameters: unable to set parameters (last param was [/move_group/planning_scene_monitor/publish_transforms_updates=True]): cannot marshal None unless allow_none is enabled
   ```
 * running move_group.launch, starting RVIZ, and using a MotionPlanning plugin to generate some movements - OK.

davetcoleman, 2020-06-02:
Regarding orocos, its actually in the Noetic Migration Guide:
http://wiki.ros.org/noetic/Migration#Use_Orocos_KDL_and_BFL_rosdep_keys

davetcoleman, 2020-06-02:
@rhaschke your comment in the top description is outdated I believe. Correct?

> There are no official Noetic docker images available yet, but I manually prepare one: moveit/ros:noetic.

I'm using ``ros:noetic-ros-base``

Also, I started working on the Docker PR because I wanted to test Noetic without upgrading / partitioning my system, but then I noticed you said in this issue that you had already started working on it. My bad. I don't see a PR from you, but I hope we can synchronize our work...

JafarAbdi, 2020-06-02:
> Blocked releases page is a little out of date (since orocos_kdl is a system dependency now, and we have added some dependencies since the last melodic release):
> 
> * [ ]  ompl
> * [ ]  srdfdom
> * [ ]  rosparam_shortcuts
> * [ ]  ros_pytest
> * [ ]  joystick_drivers repo (need spacenav for jogarm)
> * [ ]  joy_teleop from https://github.com/ros-teleop/teleop_tools (for jogarm)
> 
> (also, if you aren't on shadow-fixed, geometric_shapes needs to be built from source)

Just released rosparam_shortcuts https://github.com/ros/rosdistro/pull/25273

rhaschke, 2020-06-02:
Noetic docker images became available just recently: https://discourse.ros.org/t/ros-noetic-ninjemys-release/14262/3

davetcoleman, 2020-06-02:
Using the [experimental docker build](https://github.com/ros-planning/moveit/pull/2132), rosdep is failing for all the following dependencies:

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
mouse_teleop: No definition of [python-numpy] for OS version [focal]
franka_example_controllers: Cannot locate rosdep definition for [libfranka]
franka_hw: Cannot locate rosdep definition for [libfranka]
franka_control: Cannot locate rosdep definition for [libfranka]
moveit_commander: No definition of [python-pyassimp] for OS version [focal]
franka_visualization: Cannot locate rosdep definition for [libfranka]
franka_gripper: Cannot locate rosdep definition for [libfranka]
moveit_core: Cannot locate rosdep definition for [orocos_kdl]
moveit_ros_planning_interface: No definition of [python] for OS version [focal]
moveit_kinematics: Cannot locate rosdep definition for [orocos_kdl]
```

@mikeferguson did you work on these already?

mikeferguson, 2020-06-02:
#2131 addresses the orocos_kdl, python-pyassimp ones (although also depends on rosdep being merged which is what's holding up 2131).

I did not address python-numpy for mouse_teleop (I just noticde I catkin_ignored that package). Also didn't do anything with franka

mikeferguson, 2020-06-02:
mouse_teleop is addressed in https://github.com/ros-teleop/teleop_tools/pull/52

mikeferguson, 2020-06-02:
Finally ran full tests for moveit repo, here's the output:
```
test_results/moveit_commander/rostest-test_python_moveit_commander.xml: 1 tests, 0 errors, 1 failures, 0 skipped
test_results/moveit_commander/rostest-test_python_moveit_commander_ns.xml: 1 tests, 0 errors, 1 failures, 0 skipped
test_results/moveit_commander/rostest-test_python_time_parameterization.xml: 1 tests, 0 errors, 1 failures, 0 skipped
test_results/moveit_core/gtest-test_bullet_continuous_collision_checking.xml: 8 tests, 0 errors, 0 failures, 2 skipped
test_results/moveit_ros_perception/gtest-mesh_filter_test.xml: 48 tests, 0 errors, 2 failures, 0 skipped
test_results/moveit_ros_planning_interface/MISSING-rostest-test_move_group_interface_cpp_test.xml: 1 tests, 0 errors, 1 failures, 0 skipped
test_results/moveit_ros_planning_interface/MISSING-rostest-test_subframes_test.xml: 1 tests, 0 errors, 1 failures, 0 skipped
test_results/moveit_ros_planning_interface/rostest-test_python_move_group.xml: 1 tests, 0 errors, 1 failures, 0 skipped
test_results/moveit_ros_planning_interface/rostest-test_python_move_group_ns.xml: 2 tests, 0 errors, 2 failures, 0 skipped
test_results/moveit_ros_planning_interface/rostest-test_robot_state_update.xml: 1 tests, 0 errors, 1 failures, 0 skipped
test_results/moveit_ros_planning_interface/rosunit-cleanup_tests.xml: 2 tests, 0 errors, 1 failures, 0 skipped
Summary: 521 tests, 0 errors, 12 failures, 2 skipped
```
Mostly python, gonna dig into a few.

mikeferguson, 2020-06-02:
It looks like all the moveit_commander and moveit_ros_planning_interface tests are failing because the python bindings are missing a symbol. The import of:

    from moveit_ros_planning_interface import _moveit_move_group_interface

causes a missing symbol for void tf2::fromMsg<geometry_msgs::Quaternion_<std::allocator<void> >, Eigen::Quaternion<double, 0> >(geometry_msgs::Quaternion_<std::allocator<void> > const&, Eigen::Quaternion<double, 0>&)

rhaschke, 2020-06-02:
Looks like `#include <tf2_eigen/tf2_eigen.h>` is missing somewhere.

mikeferguson, 2020-06-02:
Mesh filter bug looks more significant:
```
[ RUN      ] float_test/MeshFilterTestFloat.float/7
/home/ubr/noetic/moveit/test/src/moveit/moveit_ros/perception/mesh_filter/test/mesh_filter_test.cpp:234: Failure
The difference between filtered_depth[idx] and gt_depth[idx] is 3.599998950958252, which exceeds 1e-6, where
filtered_depth[idx] evaluates to 3.599998950958252,
gt_depth[idx] evaluates to 0, and
1e-6 evaluates to 9.9999999999999995e-07.
```

rhaschke, 2020-06-02:
The mesh filter bug is an old friend, only occurring with libMesa... (see #374, partially fixed in #1661, #2044). Now the deviation looks more severe. Maybe a new issue with libMesa?

davetcoleman, 2020-06-03:
I just did a summary of Noetic Docker failures, in this PR:
https://github.com/ros-planning/moveit/pull/2132#issuecomment-638250671

I emailed my contact at Franka to push on that side.

gleichdick, 2020-06-03:
> It looks like all the moveit_commander and moveit_ros_planning_interface tests are failing because the python bindings are missing a symbol. The import of:
> 
> ```
> from moveit_ros_planning_interface import _moveit_move_group_interface
> ```
> 
> causes a missing symbol for void tf2::fromMsg<geometry_msgs::Quaternion_<std::allocator >, Eigen::Quaternion<double, 0> >(geometry_msgs::Quaternion_<std::allocator > const&, Eigen::Quaternion<double, 0>&)

Have a look at #1785 and ros/geometry2#430.

mamoll, 2020-06-04:
ompl release PR: https://github.com/ros/rosdistro/pull/25302

henningkayser, 2020-06-26:
> We need to try the clang-format config here and check whether we can further tweak it to reduce reformatting.

Picked this up in #2169

davetcoleman, 2020-08-27:
Geometry 2 released: https://github.com/ros/rosdistro/pull/26309

rhaschke, 2020-09-07:
Finally, the release PR is out: https://github.com/ros/rosdistro/pull/26484

closed 2020-09-07 by rhaschke

RicoJia, 2020-10-23:
>     * `load_parameters: unable to set parameters (last param was [/move_group/planning_scene_monitor/publish_transforms_updates=True]): cannot marshal None unless allow_none is enabled`

I got the same error too. I got rid of it by adding a home pose in the setup_assistant. Was that the same issue for you? @mikeferguson Is it mandatory to add a home pose in the Noetic distro?