moveit/moveit #2017: Adding Path Constraints to a MotionPlanRequest in an Action Server causes: [ERROR] mutex lock failed in pthread_mutex_lock: Invalid argument Aborted (core dumped)

DHeroPlasma, 2020-04-08:
### Description

Code is very similar to the MotionPlanningAPI-tutorial [see: https://ros-planning.github.io/moveit_tutorials/doc/motion_planning_api/motion_planning_api_tutorial.html]
Everything works perfectly fine until adding path constraints, which casues a core dump in multiple applications on my machine:

### Your environment
* ROS Distro: Melodic 1.14.3
* OS Version: Linux Mint Cinnamon 19.03 (based on Ubuntu 18.04 Xenial)
* Binary install, version: 1.0.2

### Steps to reproduce
Add a geometry_msgs::QuaternionStamped just as described in the tutorial (link in the description) to your application case to your MotionPlanRequest object as a path-constraint attribute. Further details regarding mine will be provided below. I embedded the code in the callback function of an action server and built an object-oriented environment which performs some initialization routines for my Bin-Picking robot, but the issue also appears when I'm running similar code in a simple ROS-node without my robot-specific class.

### Expected behaviour
A valid path should be calculated after adding several such kinematic contraints (regarding the goal, the trajectory, the path ...) to my MotionPlanRequest which is then being executed by the move_group object. By adding a path constraint, I am asking my endeffector to stay level because I am grabbing boxes from a rack.

### Actual behaviour
Everything works fine until adding path constraints. I encountered this issue in different applications (action for ptp-movement of the endeffector, action for cp-movement of the endeffector, ...)

### My Code (executeCB of the action sever):
```
// Initializing variables and objects:
std::string group_name = "gripper";
Strategic_Base strategy(group_name);
planning_interface::MotionPlanRequest req;
planning_interface::MotionPlanResponse res;
moveit_msgs::MotionPlanResponse response;
req.group_name = group_name;

// This is where I am loading specific robot info throug a usecase-specific class (initializing everything locally doesn't change the output)
moveit::planning_interface::MoveGroupInterface* move_group = strategy.getMoveGroup();
robot_model::RobotModelPtr robot_model = strategy.getRobotModel();

// Setting and checking parameters:
move_group->setMaxVelocityScalingFactor(0.05);
feedback_.isProcessing = true;
as_.publishFeedback(feedback_);

// Refering the robot's environment from different topics through Moveit! messages:
planning_scene::PlanningScenePtr planning_scene(new planning_scene::PlanningScene(robot_model));

// Loading a valid planner from the parameter sever:
boost::scoped_ptr<pluginlib::ClassLoader<planning_interface::PlannerManager>> planner_plugin_loader;
planning_interface::PlannerManagerPtr planner_instance = strategy.loadPlanner("ompl_interface/OMPLPlanner", nh_);

// Mapping:
double goal_coordinates[7] = {goal->initialX, goal->initialY, goal->initialZ, goal->initOrientX, goal->initOrientY, goal->initOrientZ, goal->initOrientW};
geometry_msgs::PoseStamped target_pose = strategy.movePTP(goal_coordinates, "base_link");
std::vector<double> tolerance_pose(3, 0.01);
std::vector<double> tolerance_angle(3, 0.01);
moveit_msgs::Constraints pose_goal = kinematic_constraints::constructGoalConstraints("link_grip", target_pose, tolerance_pose, tolerance_angle);
req.goal_constraints.push_back(pose_goal);

// Asking the endeffector to stay level and specifying the planning volume - Everything is working fine when I'm leaving this section out:
geometry_msgs::QuaternionStamped quaternion;
quaternion.header.frame_id = "base_link";
quaternion.quaternion.w = 1.0;
req.path_constraints = kinematic_constraints::constructGoalConstraints("link_grip", quaternion);
req.workspace_parameters.min_corner.x = req.workspace_parameters.min_corner.y = req.workspace_parameters.min_corner.z = -2.0;
req.workspace_parameters.max_corner.x = req.workspace_parameters.max_corner.y = req.workspace_parameters.max_corner.z = 2.0;
```
Calculating a valid path: This is where my program crashes
```
planning_interface::PlanningContextPtr context = planner_instance->getPlanningContext(planning_scene, req, res.error_code_);
```
Further code is not being executed and can therefore be ignored. All I am doing afterwards is checking whether a valid path has been found and passing it's trajectory to a Motion Plan for my move_group.

### ERROR message as it is output by the console without using a debugger:
```
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
[... 295 lines cut ...]
==9962==    by 0xA0D6857: moveit::planning_interface::getSharedTF() (in /opt/ros/melodic/lib/libmoveit_common_planning_interface_objects.so.1.0.2)
==9962==    by 0x51B5579: moveit::planning_interface::MoveGroupInterface::MoveGroupInterface(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<tf2_ros::Buffer> const&, ros::WallDuration const&) (in /opt/ros/melodic/lib/libmoveit_move_group_interface.so.1.0.2)
==9962==    by 0x4EC45EC: Strategic_Base::Strategic_Base(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (Strategic_Base.cpp:16)
==9962==    by 0x19E2CD: cpAction::executeCB(boost::shared_ptr<manipulator::CPGoal_<std::allocator<void> > const> const&) (cp_server.cpp:28)
==9962==    by 0x1C2DF8: boost::_mfi::mf1<void, cpAction, boost::shared_ptr<manipulator::CPGoal_<std::allocator<void> > const> const&>::operator()(cpAction*, boost::shared_ptr<manipulator::CPGoal_<std::allocator<void> > const> const&) const (mem_fn_template.hpp:165)
==9962==    by 0x1BF931: void boost::_bi::list2<boost::_bi::value<cpAction*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, cpAction, boost::shared_ptr<manipulator::CPGoal_<std::allocator<void> > const> const&>, boost::_bi::rrlist1<boost::shared_ptr<manipulator::CPGoal_<std::allocator<void> > const> const&> >(boost::_bi::type<void>, boost::_mfi::mf1<void, cpAction, boost::shared_ptr<manipulator::CPGoal_<std::allocator<void> > const> const&>&, boost::_bi::rrlist1<boost::shared_ptr<manipulator::CPGoal_<std::allocator<void> > const> const&>&, int) (bind.hpp:319)
==9962== 
==9962== LEAK SUMMARY:
==9962==    definitely lost: 114 bytes in 1 blocks
==9962==    indirectly lost: 0 bytes in 0 blocks
==9962==      possibly lost: 2,544 bytes in 8 blocks
==9962==    still reachable: 105,332,146 bytes in 15,560 blocks
==9962==                       of which reachable via heuristic:
==9962==                         stdstring          : 119 bytes in 1 blocks
==9962==         suppressed: 0 bytes in 0 blocks
==9962== Reachable blocks (those to which a pointer was found) are not shown.
==9962== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==9962== 
==9962== For counts of detected and suppressed errors, rerun with: -v
==9962== Use --track-origins=yes to see where uninitialised values come from
==9962== ERROR SUMMARY: 35 errors from 10 contexts (suppressed: 0 from 0)
Aborted (core dumped)
```

Any help is greatly appreciated as I have been unable to solve this issue for days. If further information is required, I will gladly work for and provide additional output or information.

v4hn, 2020-04-08:
There have been issues with this in the past too. It's important to point out that adding the orientation path constraint ends up using a totally different planning model environment which is not used very often.

Could you compile OMPL & MoveIt (preferably the `master` branch, but you could try `melodic-devel` as well) on your machine and see whether you can reproduce the problem there?
Consider a (very slowly running) debug build for more information in gdb.

DHeroPlasma, 2020-04-11:
Thanks for your hints and help so far. I will do as suggested and provide the results asap

DHeroPlasma, 2020-04-16:
So, I have built Moveit! from Source (latest stable version, Melodic), as well as OMPL, following the instructions from:
https://moveit.ros.org/install/source/dependencies/

Furthermore, I've built the trac_ik solver from source, changing the kinematics-solver back to kdl does not change the outcome though:
https://bitbucket.org/traclabs/trac_ik/src/master/

Here's what I noticed when running the exact same code:

- Sometimes, for example when I am adding an orientation-constraint, the program hangs itself up right before executing the "getPlanningScene" command (that's where it used to crash before). I can't even Ctrl+C the terminal it's running in. This happens with and without the program running within a debug session

- Executing the program with gdb does not result in any usefull Error output therefore

DHeroPlasma, 2020-04-20:
### SOLUTION

I finally managed to solve the problem. The bug-label is not proper for this issue as it was an implementation-issue, which I am going to share with people who may encounter the same issue (which is rather unlikely):

The problem lied in this specific section of my code:

`boost::scoped_ptr<pluginlib::ClassLoader<planning_interface::PlannerManager>> planner_plugin_loader;

planning_interface::PlannerManagerPtr planner_instance = strategy.loadPlanner("ompl_interface/OMPLPlanner", nh_);`

Up until now, I was loading the planner in one of my class methods. It seems like there have been some copying issues with the boost::scoped_ptr, and the solution was quite simple in the end:
Just initialize your plugin_loader locally in your class-method, should you ever plan to do something similar (building an object-oriented environment for Moveit!-Applications).

This is probably also related to what @v4hn hinted at, which is the fact that a new planning model environment is used when working with orientation constraints, even though I am not quite able to  understand why the code was working using only goal constraints.

v4hn, 2020-04-20:
I'm not sure about what your fix actually was, but it sounds a bit like the typical issue of deleting a plugin loader while the plugin is still being used (a weird restriction of `pluginlib`).

Closing.

closed 2020-04-20 by v4hn