moveit/moveit #2020: OMPL planner not loaded

AndreV84, 2020-04-10:
ROS melodic
after installing moveit I find that the OMPL planner library won't load;

### Your environment
* ROS Distro: [Melodic]
* OS Version: Ubuntu 18.04
* Source build.
* If binary, which release version?
* If source, which branch? - master https://github.com/ros-planning/moveit/

### Steps to reproduce
I put the github cloned folder into catkin workspace src; then make the workspace
### Expected behaviour
when executing the default panda arm launch with roslaunch I would load movie planner and would execute movement from base state to goal state
### Actual behaviour
MoviePlanner shows that OMPL library is not loaded and when I press "plan" button nothing changes.

gavanderhoorn, 2020-04-10:
Cross-posted: [how to get OMPL library loaded for default panda arm](https://answers.ros.org/question/348982) on ROS Answers.

AndreV84, 2020-04-10:
to try another setup I tried to install ROS from apt with
` sudo apt-get install ros-melodic-desktop ros-melodic-moveit ros-melodic-panda-moveit-config  -y`
However, by the end of execution it returns:
`Unpacking ros-melodic-panda-moveit-config (0.7.3-1bionic.20200322.005309) ...
Errors were encountered while processing:
 /tmp/apt-dpkg-install-JaKQJi/217-python-catkin-pkg-modules_0.4.16-1_all.deb
 /tmp/apt-dpkg-install-JaKQJi/243-python-rospkg-modules_1.2.4-1_all.deb
 /tmp/apt-dpkg-install-JaKQJi/244-python-rosdistro-modules_0.8.0-1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

AndreV84, 2020-04-10:
on apt --fix-broken install it will return:
`Do you want to continue? [Y/n] Y
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 199254 files and directories currently installed.)
Preparing to unpack .../python-catkin-pkg-modules_0.4.16-1_all.deb ...
Unpacking python-catkin-pkg-modules (0.4.16-1) ...
Preparing to unpack .../python-rospkg-modules_1.2.4-1_all.deb ...
Unpacking python-rospkg-modules (1.2.4-1) ...
dpkg: error processing archive /var/cache/apt/archives/python-rospkg-modules_1.2.4-1_all.deb (--unpack):
 trying to overwrite '/usr/lib/python2.7/dist-packages/rospkg/__init__.py', which is also in package python-rospkg 1.1.4-1
Preparing to unpack .../python-rosdistro-modules_0.8.0-1_all.deb ...
Unpacking python-rosdistro-modules (0.8.0-1) ...
dpkg: error processing archive /var/cache/apt/archives/python-rosdistro-modules_0.8.0-1_all.deb (--unpack):
 trying to overwrite '/usr/lib/python2.7/dist-packages/rosdistro/__init__.py', which is also in package python-rosdistro 0.6.6-1
Errors were encountered while processing:
 /var/cache/apt/archives/python-rospkg-modules_1.2.4-1_all.deb
 /var/cache/apt/archives/python-rosdistro-modules_0.8.0-1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

AndreV84, 2020-04-10:
Reinstalled OS: Bionic 18.04 armhf;
Following https://wiki.ros.org/melodic/Installation/Ubuntu
`sudo apt-get install ros-melodic-desktop-full ros-melodic-moveit ros-melodic-panda-moveit-config  -y`
`echo "source /opt/ros/melodic/setup.bash" >> ~/.bashrc
`
`source ~/.bashrc
`
`sudo apt install python-rosdep python-rosinstall python-rosinstall-generator python-wstool build-essential`
then I created a folder at ~ name_ws, copied github package to src, initiated rosdep init and rosdep update
somehow works;
resolved

closed 2020-04-10 by AndreV84