moveit/moveit #2021: moveit_ros_visualization binary release

rjshim, 2020-04-10:
### Description

Overview of your issue here.
CI crash caused by binary release for the following packages not being supported.
a) ros_kinetic_moveit_ros_visualization
b) ros_melodic_moveit_ros_visualization

### Your environment
* ROS Distro: Kinetic, Melodic
* OS Version: 16.04, 18.04
* Source or Binary build? Binary
* If binary, which release version? 
* If source, which branch?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2020-04-10:
Please provide more details if you want somebody to look at this issue.
What exactly is the observed problem? Both packages are available as can be seen from these pages:
http://repositories.ros.org/status_page/ros_melodic_default.html
http://repositories.ros.org/status_page/ros_kinetic_default.html

rjshim, 2020-04-13:
Thank you for your reply. It seems like the error has been fixed.

closed 2020-04-13 by rjshim