moveit/moveit #2025: CollisionObject shapes and subframes should be defined in an object's reference frame

felixvd, 2020-04-16:
When I am placing objects in the scene, I find it extremely cumbersome to transform all of their shapes and subframes to the position I want the object to be in. It's really something that the user shouldn't have to take care of. My current workaround is to publish a temporary TF frame for the object, which is awkward, requires waiting for the frames to propagate, and isn't very stable.

I would like to replace the `header` object in the [CollisionObject message](http://docs.ros.org/melodic/api/moveit_msgs/html/msg/CollisionObject.html) to be a `PoseStamped` object called `object_pose`. All the shapes and subframes would be defined relative to that pose.

The current behavior would be easy to replicate by defining a neutral `PoseStamped` with the same `header`. With the change, it would be easier to shift the whole object, and in the future, attaching the object to a scene graph would be simple.

Does anyone see a problem with this design direction? I may start developing this soon otherwise.

rhaschke, 2020-04-16:
I can understand your frustration and I like your proposal. Actually, MoveIt's internal representation of the collision object should be changed as well, to use this pose. Currently, the "pose" of the object is extracted from the pose of its first shape, which is really awkward:
https://github.com/ros-planning/moveit/blob/4d3062164bf03ebb74c57d5eba0e3e50438481ba/moveit_core/collision_detection/src/world.cpp#L172-L174

By the way, this code doesn't check anymore that shape_poses_ isn't empty!

v4hn, 2020-04-16:
Sounds good to me.
But we probably want the message change to get in for noetic, otherwise I'm not sure this would ever be released...
So you better speed up :sunglasses: 

> By the way, this code doesn't check anymore that shape_poses_ isn't empty.

True, where's the patch? :stuck_out_tongue: 
Or do you want all three of us to file one in parallel?

rhaschke, 2020-04-16:
No time for a patch today. I was hoping for Felix...

felixvd, 2020-04-16:
Like [this](https://github.com/felixvd/moveit/tree/check-for-empty-shapes)?

I would try to get on this first thing and propose something within this month.

rhaschke, 2020-04-16:
No, more like this:
`if (it != objects_.end() && !it->second->shapes_.empty())`

v4hn, 2020-04-16:
... You know if you keep replying to everything Robert, you are not really saving time in contrast to just writing the patch.
 I just filed #2026.

felixvd, 2020-04-16:
I started working on this [here](https://github.com/felixvd/moveit/tree/add-object-origin). My first thoughts:

- Each `object` in `collision_world` currently has `shape_poses_` and `subframe_poses_`. Could each of these be split into `relative_shape_poses_` and `global_shape_poses_`? I will likely do this anyway while writing.
- The methods `addToObject` and `setSubframesOfObject` in `world.cpp` currently take global poses (defined in the `planning_frame`), so they define the `global_xxxxx_poses_`. It may make `planninng_scene.cpp` more readable if one could define the `relative_xxxxx_poses_` and let `world` calculate the global transform. Does `world` know all of the same transforms as `planning_scene`?
- Would it be better to make the new method `addToObjectRelative` or to add a parameter `use_relative_frame` to `addToObject`?
- The object's origin is not tracked in the `attached_body` structure, but it probably should be.

My second thoughts: In the interest of getting this merged quickly, it will be better avoid large changes. Strictly speaking this might not require many more changes than the first commit. The rest might be optional/for internal use.

felixvd, 2021-08-28:
Fixed by #2037

closed 2021-08-28 by felixvd