moveit/moveit #2029: Warning in CMakeLists.txt of Moveit Packages.

abhiMishra98, 2020-04-18:
catkin_make
Base path: /home/lucifer/catkin_ws
Source space: /home/lucifer/catkin_ws/src
Build space: /home/lucifer/catkin_ws/build
Devel space: /home/lucifer/catkin_ws/devel
Install space: /home/lucifer/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/lucifer/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/lucifer/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/lucifer/catkin_ws/devel;/home/lucifer/ws_moveit/devel;/opt/ros/melodic
-- This workspace overlays: /home/lucifer/catkin_ws/devel;/home/lucifer/ws_moveit/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/lucifer/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 34 packages in topological order:
-- ~~  - moveit (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'moveit' contains non standard content. Use the content of the following file instead: /home/lucifer/catkin_ws/build/catkin_generated/metapackages/moveit/CMakeLists.txt
-- ~~  - moveit_planners (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'moveit_planners' contains non standard content. Use the content of the following file instead: /home/lucifer/catkin_ws/build/catkin_generated/metapackages/moveit_planners/CMakeLists.txt
-- ~~  - moveit_plugins (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'moveit_plugins' contains non standard content. Use the content of the following file instead: /home/lucifer/catkin_ws/build/catkin_generated/metapackages/moveit_plugins/CMakeLists.txt
-- ~~  - moveit_ros (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'moveit_ros' contains non standard content. Use the content of the following file instead: /home/lucifer/catkin_ws/build/catkin_generated/metapackages/moveit_ros/CMakeLists.txt
-- ~~  - moveit_runtime (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'moveit_runtime' contains non standard content. Use the content of the following file instead: /home/lucifer/catkin_ws/build/catkin_generated/metapackages/moveit_runtime/CMakeLists.txt
-- ~~  - panda_moveit_config
-- ~~  - geometric_shapes
-- ~~  - beginner_tutorials
-- ~~  - moveit_msgs
-- ~~  - moveit_commander
-- ~~  - rviz_visual_tools
-- ~~  - moveit_core
-- ~~  - chomp_motion_planner
-- ~~  - moveit_chomp_optimizer_adapter
-- ~~  - moveit_ros_occupancy_map_monitor
-- ~~  - moveit_ros_perception
-- ~~  - moveit_ros_planning
-- ~~  - moveit_fake_controller_manager
-- ~~  - moveit_kinematics
-- ~~  - moveit_planners_ompl
-- ~~  - moveit_ros_move_group
-- ~~  - moveit_ros_manipulation
-- ~~  - moveit_ros_robot_interaction
-- ~~  - moveit_simple_controller_manager
-- ~~  - moveit_ros_control_interface
-- ~~  - moveit_visual_tools
-- ~~  - moveit_ros_warehouse
-- ~~  - moveit_ros_benchmarks
-- ~~  - moveit_ros_planning_interface
-- ~~  - moveit_jog_arm
-- ~~  - moveit_planners_chomp
-- ~~  - moveit_ros_visualization
-- ~~  - moveit_setup_assistant
-- ~~  - moveit_tutorials
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:67 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/lucifer/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/lucifer/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:2336: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

hhxzzy, 2020-06-12:
I had the same issue. How you solve this?

v4hn, 2020-06-12:
> I had the same issue. How you solve this?

Their actual error (in the end of the log) had nothing to do with warnings.
If you build MoveIt together with its external dependencies (such as OMPL or FCL) you can't build with `catkin_make`. Either you use `catkin build` (the `catkin_tools`), or `catkin_make_isolated`.

If we actually have non-standard metapackage `CMakeLists.txt` files, I would welcome a pull-request to fix it.