moveit/moveit #2032: Enable Arm Jogging in Real-Time in ROS Kinetic?

luke-kim409, 2020-04-21:
Hello,

I am wondering whether the real-time jogging (controlling the robot's end-effector with a 3D mouse in real time) can be done in ROS Kinetic as well as ROS Melodic.

I am relatively new to ROS, but based on the [tutorial on real-time jogging](https://ros-planning.github.io/moveit_tutorials/doc/arm_jogging/arm_jogging_tutorial.html), I was able to control my robot model (KUKA LBR iiwa) in Rviz and Gazebo with 3D mouse on **ROS Melodic** after some customization. I am using [iiwa_stack ](https://github.com/IFL-CAMP/iiwa_stack), which is known to support ROS Melodic, but for some unknown reason, I was not able to get my robot connected to ROS Melodic on my desktop. On my laptop with **ROS Kinetic**, however, I am able to get my robot connected and control it using Moveit.

Since I can work with my physical robot only in ROS Kinetic for now, I am wondering whether it would be technically possible to use the packages for real-time jogging (e.g. moveit_jog_arm) in **ROS Kinetic** instead of ROS Melodic. It seems some dependent packages such as "rosparam_shortcuts" or "joy_teleop" which exist in ROS Melodic do not exist in ROS Kinetic. 

Could anyone advise whether those packages required for real-time jogging can be tweaked to be used in ROS Kinetic? Any suggestion would be greatly helpful for me! Many thanks in advance!

v4hn, 2020-04-21:
At the moment, you should still be able to compile MoveIt's `master` branch on ROS kinetic.
Yes, you will have to compile a number of packages in their `melodic-devel` or `master` branches by hand.

luke-kim409, 2020-04-22:
Hello @v4hn, Thank you so much for your answer!
I am really hoping to migrate to ROS Melodic, but currently having troubles in the process due to the connection problem.

May I ask a few more questions?
So, it seems, technically, running the jog_arm module on ROS Kinetic is possible although it can be cumbersome. If technically possible, I am hoping to give it a try, but where to start is not so clear to me as a relative newbie.

Could you please elaborate a little more on these highlighted parts?
"you will have to **compile a number of packages in their melodic-devel or master branches by hand**."
"you will have to **rely on source builds**."

When I tried source build to install Moveit's master branch on ROS Kinetic by following the steps described in here: https://moveit.ros.org/install/source/, "catkin build" ended up with partial success (only 9 out of 42 packages succeeded). Probably this is what you meant by "a number of packages need to be compiled by hand". Would there be any resources from which I can learn how to compile those packages "by hand"? 

Probably this is a stupid question, but any further comment would be very helpful!

v4hn, 2020-04-22:
> When I tried source build to install Moveit's master branch on ROS Kinetic by following the steps described in here: https://moveit.ros.org/install/source/, "catkin build" ended up with partial [...]

Following the instructions, you already compile a lot of packages "by hand" (in your workspace) in the correct version ([the rosinstall)(https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall) lists them all).
Look through the error messages one by one and try to fix them. Usually, it should boil down to "MoveIt expects an API that does not exist in this module", which means "this module" is probably not installed in the correct version and you will have to add it to your workspace.

tylerjw, 2020-05-01:
Sadly, using jog arm in melodic without a source build is not possible atm.  So, even if you switched to melodic, you would still need to do a source build of moveit and it's dependencies.

luke-kim409, 2020-05-01:
Thanks for your answer! 
I found that the original jog_arm packages works quite well for Kinetic:
https://github.com/UTNuclearRoboticsPublic/jog_arm
I am closing this issue, but probably it would be helpful to leave this comment for those who want to try real-time jogging in Kinetic.

closed 2020-05-01 by luke-kim409

HYEDO-KIM, 2021-03-29:
@luke-kim409 Hello! I'm trying to jogging arm in Real-Time in ROS melodic!
It would be much easier for me to work with your help. Can you help me?
plz answer!