moveit/moveit #2034: ur5_e with moveit in webots

virtualclone, 2020-04-22:
### Description

Hello, I'm using ur5_e with moveit in webots.
I follow this [tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html), and I only can control ur5_e to execute the first trajectory (joint goal). After that, it stops. I found that the program will stop after this line
`group.go(joint_goal, wait=True)`
It seems that it's stuck in this function. How can I fix that?

By the way, I also follow the [tutorial](https://cyberbotics.com/doc/guide/ure) about ur5_e in webots
### Environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* moveit version: kinetic

felixvd, 2020-04-22:
You need to at least post the error message. This would be better on https://answers.ros.org/

felixvd, 2021-02-19:
Closing because of inactivity.

closed 2021-02-19 by felixvd