moveit/moveit #2039: Meta: Allow labeling for core contributors

simonschmeisser, 2020-04-30:
I'm not sure if this is actually possible with github but I would really like to put labels on stuff! (Who doesn't? ;) )

My main motivation currently is to put "MoveItDay" on some of the doc-issues I opened so I don't have to do it myself ...

@davetcoleman

gavanderhoorn, 2020-04-30:
The [Triage access level](https://help.github.com/en/github/setting-up-and-managing-organizations-and-teams/repository-permission-levels-for-an-organization#repository-access-for-each-permission-level) should allow this.

rhaschke, 2020-04-30:
@davetcoleman, are you the only one with access rights at the ros-planning organization level?
I just wanted to grant this right to contributors (which, I think makes absolutely sense), but I don't see the settings tab at the organization level.

felixvd, 2020-04-30:
Would be great if this worked, has been an issue for a while.

v4hn, 2020-04-30:
> I just wanted to grant this right to contributors (which, I think makes absolutely sense), but I don't see the settings tab at the organization level.

I believe the only MoveIt developers with Owner access to ros-planning (outside the general OSRF/OR crowd) are Dave and Mike Ferguson.

simonschmeisser, 2020-05-27:
@davetcoleman friendly ping before adding it to tomorrows list

davetcoleman, 2020-05-27:
"Apply labels" is now available via "Triage" access. I've just changed the "Core Contributors" access to "Triage" for all MoveIt packages.

Sorry for my delay in seeing this, Felix just emailed me the link to this issue!

closed 2020-05-27 by simonschmeisser

davetcoleman, 2020-06-02:
Documented: https://github.com/ros-planning/moveit.ros.org/pull/453