moveit/moveit #2042: Debians and circular dependency with moveit_ros_occupancy_map_monitor and moveit_ros_perception

tylerjw, 2020-05-01:
### Description

#### Context:
I discovered this while trying to debug a failed kinetic CI run for my PR here: https://github.com/ros-planning/moveit/pull/2031

`moveit_ros_perception` depends on `moveit_ros_occupancy_map_monitor` in package.xml which is all fine except `moveit_ros_occupancy_map_monitor` also depends on `moveit_ros_perception`.

The fact that CI passes at all is really strange given this circular dependency.  What it does for us is hide this bug by letting `moveit_ros_occupancy_map_monitor` link the debian released version of `moveit_ros_perception`.

One sledge hammer approach to fixing this would be to move all the code of `moveit_ros_occupancy_map_monitor` into `moveit_ros_perception`.  Are there any other ideas for how to remove this circular dependency?

### Evidence:

Here is a call stack from valgrind from my test I linked above:
```bash
==2495== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==2495==  General Protection Fault
==2495==    at 0x45C378: __gnu_cxx::__atomic_add(int volatile*, int) (atomicity.h:53)
==2495==    by 0x45C437: __gnu_cxx::__atomic_add_dispatch(int*, int) (atomicity.h:96)
==2495==    by 0x4652B4: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_add_ref_copy() (shared_ptr_base.h:134)
==2495==    by 0x461A7C: std::__shared_count<(__gnu_cxx::_Lock_policy)2>::operator=(std::__shared_count<(__gnu_cxx::_Lock_policy)2> const&) (shared_ptr_base.h:676)
==2495==    by 0xB28C392: std::__shared_ptr<bodies::ConvexMesh::MeshData, (__gnu_cxx::_Lock_policy)2>::operator=(std::__shared_ptr<bodies::ConvexMesh::MeshData, (__gnu_cxx::_Lock_policy)2> const&) (shared_ptr_base.h:867)
==2495==    by 0xB28C3BC: std::shared_ptr<bodies::ConvexMesh::MeshData>::operator=(std::shared_ptr<bodies::ConvexMesh::MeshData> const&) (shared_ptr.h:93)
==2495==    by 0xB288E67: bodies::ConvexMesh::cloneAt(Eigen::Transform<double, 3, 1, 0> const&, double, double) const (bodies.cpp:1104)
==2495==    by 0x21E586AE: point_containment_filter::ShapeMask::addShape(std::shared_ptr<shapes::Shape const> const&, double, double) (in /opt/ros/kinetic/lib/libmoveit_point_containment_filter.so.0.9.18)
==2495==    by 0x219C51F9: occupancy_map_monitor::PointCloudOctomapUpdater::excludeShape(std::shared_ptr<shapes::Shape const> const&) (in /opt/ros/kinetic/lib/libmoveit_pointcloud_octomap_updater_core.so.0.9.18)
==2495==    by 0x99F21C4: occupancy_map_monitor::OccupancyMapMonitor::excludeShape(std::shared_ptr<shapes::Shape const> const&) (occupancy_map_monitor.cpp:237)
==2495==    by 0x5364E5E: planning_scene_monitor::PlanningSceneMonitor::excludeRobotLinksFromOctree() (planning_scene_monitor.cpp:692)
==2495==    by 0x53697A5: planning_scene_monitor::PlanningSceneMonitor::startWorldGeometryMonitor(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) (planning_scene_monitor.cpp:1076)
```

tylerjw, 2020-05-01:
Is there someway we could remove packages that we are building from source from the debian install before the build in CI to guarantee this isn't happening in other places too?

It seems really broken if the dependency system will let you do this and won't even complain when you have the source of one package depending on the source of a second package that depends on an old binary of the first.

rhaschke, 2020-05-01:
I'm confused. Both in melodic and master, we have the following dependencies:
- `moveit_ros_perception` -> `moveit_ros_occupancy_map_monitor`
- `moveit_ros_planning` -> `moveit_ros_occupancy_map_monitor`
In Kinetic, `occupancy_map_monitor` is part of `moveit_ros_perception`. We only moved it into its own package on the master branch (#1533).

Could you please provide more details, of what you actually did.

rhaschke, 2020-05-01:
Maybe you just screwed your workspace. Did you build from scratch?

tylerjw, 2020-05-01:
In master:
`moveit_ros_planning` -> `moveit_ros_occupancy_map_monitor` -> `moveit_ros_planning`

The CI build for kinetic builds master in a kinetic docker image, that is how I found this.

tylerjw, 2020-05-01:
It looks like the author and creator of that PR didn't trace all the dependencies and created this circular dependency problem that let the old debian in as a dependency.

tylerjw, 2020-05-01:
In `moveit_ros_occupancy_map_monitor`:
`occupancy_map_monitor.cpp:237':
```C++
    return map_updaters_[0]->excludeShape(shape);
```

That calls `occupancy_map_monitor::PointCloudOctomapUpdater::excludeShape` which is in `pointcloud_octomap_updater.cpp:140` inside  `moveit_ros_preception`.

The catkin magic builds the linker commands to let us get away with this.  I vote we revert that change and throw it back to the author to fix this circular dependency.  Thoughts, @rhaschke ?

tylerjw, 2020-05-01:
> Maybe you just screwed your workspace. Did you build from scratch?

I am doing what the kinetic CI build is doing, building master in a kinetic docker container and running these tests.  I left instructions in my PR I linked above on how to reproduce this.

jschleicher, 2020-05-01:
> The catkin magic builds the linker commands to let us get away with this

I don't understand the problem. As far as I understand this isn't at compile time (linker) but at runtime?! Occupancy_map_monitor should run without any `_perception`, as long as there isn't any updater configured. If you want to use a `point_cloud_octomap_updater`, you of course have to install the package - but it's no longer a required dependency (pulling in approx 500MB of unrelated debians e.g. OpenCV-dev)

I'll try to use your steps to reproduce the problem locally.

tylerjw, 2020-05-01:
Linker sets us up to fail... but we don't because we (until my PR) didn't have a test that tried to execute this code.

You can see the callstack that valgrind prints in my PR I linked.

To reproduce it in valgrind in the docker container you'll have to do these additional steps:
1. modify move_group.launch used by my test (switch it to the panda_moveit_config package to make your life easier) to set valgrind as the prefix on debug
2. modify my test to launch move_group with debug
3. install valgrind
4. build in debug mode to get the symbol data (optional)
5. run my test and watch it fail

jschleicher, 2020-05-01:
@tylerjw  Thanks for digging into that issue! But I now get totally different backtraces :-(

First of all I skipped
```
TEST_BLACKLIST="moveit_ros_perception"
```
because that will probably use the system lib instead of building our own - as you described. Then valgrind tells me something within libompl while gdb complains about fcl:

[gist](https://gist.github.com/jschleicher/3dac6435ed848ecd18e6ce6612f24778)

I'm totally confused now.

jschleicher, 2020-05-01:
When [droppping](https://github.com/PilzDE/moveit/commit/d4a680824bbaa88818820e93234841927612307a)
```
TEST_BLACKLIST="moveit_ros_perception"
```

travis [runs your test successfully](TEST_BLACKLIST="moveit_ros_perception"). So we probably have to find a different way to disable the failing `mesh_filter_test` ?

tylerjw, 2020-05-01:
Ah, good catch.  I copy/pasted that line out of the travis config and then forgot about it's implications for this.  Is there a issue for the failing `mesh_filter_test` ?  What do we need to do to re-enable it?

rhaschke, 2020-05-02:
We do have an issue for the broken mesh_filter_test: #1109. Nobody cared about it for 1.5 years (and we don't have the expertise to debug Mesa OpenGL).

jschleicher, 2020-05-02:
Is there another way to disable just that single test instead of disabling the complete `moveit_ros_perception` build during testing?

rhaschke, 2020-05-02:
I didn't get the core issue yet. I will try to summarize:
Because moveit_ros_perception is skipped during the test phase, some system-installed (plujgin?) library of moveit_ros_perception is loaded instead of the correct one from the workspace (which one exactly?).
I'm still not sure, how this can happen. During build time moveit_ros_perceptiion should be still part of the game. Hence, all its exported libs should be available.

The only other option is to explicitly disable the tests in cmake. But as these tests succeed on e.g. nvidia platforms, I don't want to do this.

tylerjw, 2020-05-02:
Good catch.  Looking at [moveit_ci blacklist](https://github.com/ros-planning/moveit_ci/blob/master/travis.sh#L318) it configures that before it builds and runs the test but after the workspace is built.  I'm testing that now.

As to the mesh_filter_test is there anyway to detect Mesa GPU in the code and skip the test there?  It seems bad that we skip testing the whole perception package because of this.

rhaschke, 2020-05-02:
Looks like, catkin tools removes blacklisted packages _entirely_ from the workspace.
I switched from explicitly listing to-be-tested packages to blacklisting as part of https://github.com/ros-planning/moveit_ci/pull/45 (https://github.com/ros-planning/moveit_ci/pull/45/commits/959087ac043d0180c88a173695f2a2b9b82bd3cd).

The problem with the former approach was that meta packages (if not blacklisted as well) have pulled in all their dependencies, even if blacklisted for testing. catkin blacklisting seemed like a good alternative at that time.

tylerjw, 2020-05-04:
I dug into this a bit more with the thought that I could offer a PR to moveit_ci that changed the way blacklist works where instead we would handle blacklisting by building a whitelist of packages to run the test on but build all tests.

To test this I started over with a docker container and didn't black-list anything and tried running my test and it still fails.  @jschleicher can you remember what you did that got my test to pass?

Here is how it is crashing for me now in valgrind.  This time it isn't the issue of the wrong library at load time but instead a bunch of memory allocation errors: (gist of output)[https://gist.github.com/tylerjw/fbc69b8574e3f91967e0273b8f4785a2]

tylerjw, 2020-05-04:
To summarize where we are on this.  I think there is a way we could modify moveit_ci to fix the loading of the old debian of `_preception` libraries.

That alone won't make my test pass in kenetic :cry: as there is some other memory issue that comes up on loading before main is executed in movegroup.  Now it looks like it is either pointers being accessed/modified before they are allocated or malloc failing.  I'm looking into this now.

jschleicher, 2020-05-04:
Sorry for the wrong link above: I've interpreted https://travis-ci.org/github/PilzDE/moveit/jobs/682143605 as successful - only mesh_filter is failing, so your test was OK?!

tylerjw, 2020-05-05:
Weird.  I can't reproduce it passing locally.  I will make the small pr to moveit_ci and point my change at the pr to try it.

tylerjw, 2020-05-06:
I'm really confused at how your Travis job passed.  Here a link to one I just ran with changes to moveit_ci that I thought would solve the problem: https://travis-ci.org/github/ros-planning/moveit/jobs/683581083

As you can see move_group is crashing in this run of the test with a -6 return code.  I will investigate more.  I think if my moveit_ci change is accepted it will fix this bug though.  I will create a new issue when I have an idea what is causing the next one I found.

Debugging this is like opening an onion... As one layer is removed I discovered a new layer.  Hopefully there is only two layers.  :/

tylerjw, 2020-05-06:
Well, this is still an issue but was made irrelevant for now by the fix to the test.  The issue is caused by how --blacklist works with catkin and how we are using it in moveit_ci.  I made an attempt at fixing this, but it didn't work.  Thank you for looking into this with me.  I'll revisit this when/if I have any ideas on how to solve the blacklist problem.