moveit/moveit #2043: MotionPlanning plugin does not populate the detected objects list

Nurgak, 2020-05-02:
### Description

The motion planning plugin in Rviz has a tab named *Manipulation*, it has a nice interface that could be used to show detected objects.

[image]

I wanted to try it out and made the corresponding action server, as the *Detect* button calls an action. Upon receiving the action goal the server does its thing and published a `RecognizedObjectArray` message on the `/recognized_object_array` topic, to which Rviz is listening to.

I thought I have everything in place, but no matter what I tried the detected objects list wasn't populated with the returned items. No examples or even mentions of this feature exist on the internet. Therefore, I digged in the code and found some [very strange message handling for this feature](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame_manipulation.cpp#L170-L174).

What I understand from this is that this feature implementation is not finished as the `msg` is simply commented out and not handled at all. The [`processDetectedObjects`](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame_manipulation.cpp#L69-L110) function seems to do something, but I'm not sure what.

Could somebody help me with this? Should I go on in trying to get it working or it's an unimplemented feature right now?

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build? Binary
* If binary, which release version? 1.0.2

### Steps to reproduce
Click on the *Detect* button in the *Manipulation* tab in Rviz, properly handle it with an action server and publish to `/recognized_object_array` topic.

### Expected behaviour
The *Detected objects* list should get populated with the elements published to `/recognized_object_array`.

### Actual behaviour
Rviz hangs for about a second and nothing is populated in the *Detected objects* list or the *Support surfaces* list.

### Backtrace or Console output
The console does not produce any output from Rviz relevant to this.

v4hn, 2020-05-04:
Thank you for this issue!

As you pointed out yourself, this is horrible old code very few people ever looked at...
I'm not the author, but I also spent time with this a few years ago (and only recently fixed some structural issues with it...)

It was originally implemented for a quick demo and there is actually a video somewhere that shows how to use it with the PR2 to pick/place objects. :sunglasses: 

The missing piece of information you need to get it to work is that the action server actually has to *spawn collision objects* in the planning scene within 3 seconds of publishing the detection results. The easiest way to do that is via the `PlanningSceneInterface` in your own node.
The reason for this weird behavior is that alternatively the RViz node would have to spawn the objects in the scene and RViz is not meant to execute mandatory code in the whole workflow.

Still, the whole approach to poll for 3 seconds after a detector *action* returned (which is of course signaled by sending a message to a *random topic* instead of the action result) and
the additional related `/tables` topic is very weird and should be rewritten.
More reasonable approaches might be

- add a `MoveGroupCapability` that listens on the same topic and automatically adds objects to the scene (taking care of the one point you missed in your efforts).

- add a function to the general moveit code base that takes a recognized object array message and converts it to collision objects. If the Display makes use of it, it's still not nice in my opinion, but at least other users could just reuse the functionality... Then throw out the whole call structure and replace it by a `SimpleActionClient` that parses the result of the action, adds collision objects and populates the list.

If you are interested in contributing, your support would be much welcome! :-)

Nurgak, 2020-05-05:
Thank you @v4hn for this complete answer. I understand the situation, this makes perfect sense, some documentation/tutorials would've helped me. So it sounds like I can still do something fun with it...

I tried to publish some dummy boxes via [`add_box`](http://docs.ros.org/jade/api/moveit_commander/html/planning__scene__interface_8py_source.html#l00133) at the specified positions using `PlanningSceneInterface` as you suggested, but they did not show up in the detected objects list. They were created within the 3 second timeout, after the message to `/recognized_object_array` topic was published (with the same items). These objects did appear in the "Scene objects" list though and I could see them in RViz. Perhaps I'm supposed to give them an instance name/token somewhere? The only linking thing I could find is the `type.key` field in the `RecognizedObject` message.

Indeed, there is also topic called `/table_array` of type `object_recognition_msgs/TableArray` that RViz is listening to which I didn't even see before... I published a dummy message there too, but nothing appeared in the list. Is this list meant for the "Support Surfaces" list?

Am I missing something else?

v4hn, 2020-05-05:
> some documentation/tutorials would've helped me.

Thing is, anyone who spends enough time with it to understand it well enough, deems it the better option to rewrite it entirely instead of writing documentation and tutorials.
We actually [discussed disabling the tab altogether](https://github.com/ros-planning/moveit/pull/1611#issuecomment-597649907) recently, but didn't do it.

> So it sounds like I can still do something fun with it...

Sure, or you just rewrite the logic of the display tab, define a simpler interface and contribute it back. :-)
That sounds like fun too, doesn't it? :wink:

> I tried to publish some dummy boxes via [`add_box`](http://docs.ros.org/jade/api/moveit_commander/html/planning__scene__interface_8py_source.html#l00133) at the specified positions using [...]

The only other thing that comes to mind is that the objects have to be in the configured [region of interest](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame_manipulation.cpp#L95).
But, as you see there is simply too many conditions for this mechanism at the moment.

> These objects did appear in the "Scene objects" list though and I could see them in RViz. Perhaps I'm supposed to give them an instance name/token somewhere? The only linking thing I could find is the `type.key` field in the `RecognizedObject` message.

The `RecognizedObject` message is ignored, as I said before. There does not have to be a link to the collision objects.

> there is also topic called `/table_array` of type `object_recognition_msgs/TableArray` that RViz is listening to which I didn't even see before... Is this list meant for the "Support Surfaces" list?

Yes. The "SemanticWorld" C++ class maintains a list of support surfaces. The class name is pretty much a blatent lie. :-)

> I published a dummy message there too, but nothing appeared in the list.

This is also entwined with the "Detect Objects" trigger and might rely on a valid/reasonable table definition.

> Am I missing something else?

Lastly, there is a slim chance we actually broke this logic in the patch that followed the discussion linked above.

Nurgak, 2020-05-07:
Thank you again for such a long and detailed answer @v4hn.

I could get the following message form the console:
```
[ INFO] [1588833577.657180357]: Table callback with 1 tables
[ INFO] [1588833577.657260125]: Calling table callback
[ERROR] [1588833578.657690894]: Exception caught while processing action 'publish tables': std::bad_alloc
```

I guess the [issue starts here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame_manipulation.cpp#L203). I may look into it later.

As far as contributing goes: this feature is a nice to have thing for me, it looks relatively well suited for my quick demonstration, but investing time to fix/rehaul it is impossible.

I hope, at least, that this thread can help other people either steer away or get an idea of what needs to be done with this thing.

v4hn, 2020-05-07:
> but investing time to fix/rehaul it is impossible.

yeah, that's the usual answer... *sigh*

> [ERROR] [1588833578.657690894]: Exception caught while processing action 'publish tables': std::bad_alloc

Either you are sending invalid table messages or [`addTablesToCollisionWorld`](https://github.com/ros-planning/moveit/blob/master/moveit_ros/perception/semantic_world/src/semantic_world.cpp#L96) is bugged in generating the correct mesh/shape from it.

Could you the `master` branch from source (https://github.com/ros-planning/moveit/blob/master/moveit.rosinstall) and provide a stack trace? @peci1 improved a lot of the logic in `geometric_shapes` recently and might be able to help if the problem lies there.