moveit/moveit #2048: jog_server target_frame ignored

rfeistenauer, 2020-05-04:
I'm not sure, if it is a feature request or a bug report, since this already worked in previous versions, but got removed for some reason.

When sending a twist msg to the jog_server, its target frame id is never used. It apparently gets overridden by the config entry: `robot_link_command_frame`. Is this the intended behavior and if so, how is someone supposed to move the robot to an destination and rotate around the tcp frame, for example to screw something in, without restarting the whole node. 
Am i supposed to change the config by setting it in the parameter server all the time? 
Would that even work? If yes, is that better than using the frame_id from the message?

rfeistenauer, 2020-05-14:
Any comment if the new behavior is intended?

jschleicher, 2020-05-18:
@henningkayser @AndyZe ?

AndyZe, 2020-05-18:
Sorry I didn't see this earlier. I'm not getting tagged automatically like I should be.

> When sending a twist msg to the jog_server, its target frame id is never used. It apparently gets overridden by the config entry: robot_link_command_frame. Is this the intended behavior

Yep, that is the intended behavior. Which is not to say there's not a better way. At L177 of jog_ros_interface.cpp there's a note on why it's done that way:

>   // Copy everything but the frame name. The frame name is set by yaml file at startup.
>   // (so it isn't copied over and over)

> how is someone supposed to move the robot to an destination and rotate around the tcp frame, for example to screw something in, without restarting the whole node.

As a workaround you could send a joint command to rotate the wrist. But I see your point, it could be a little bit of an issue. Would you be able to make a  PR that:

- adds an optional ROS service in `jog_ros_interface.cpp` to provide a new command frame? The service request should be a string

- adds an optional method in `jog_cpp_interface.cpp` to provide a new command frame?

I think that would be a very welcome PR.

AndyZe, 2020-05-18:
The other workaround would be for you to transform the twist stamped command to whatever frame you want with TF2 before sending it to jog_arm. Basically, to handle it externally

v4hn, 2020-05-18:
> >   // Copy everything but the frame name. The frame name is set by yaml file at startup.
> >   // (so it isn't copied over and over)

Would it really be too much overhead to compare the frame internally when a command is received
and transform it w.r.t. the internal planning scene if it's not empty and not equal to the configured frame in the yaml file?

That would be the much nicer interface in my opinion...

AndyZe, 2020-05-18:
Yep, that sounds good too

rfeistenauer, 2020-05-19:
> Would it really be too much overhead to compare the frame internally when a command is received
and transform it w.r.t. the internal planning scene if it's not empty and not equal to the configured frame in the yaml file?

So do understand this correct, that you suggest something like this:
```
if (cmd.header.frame_id != parameters_.planning_frame)
   {
+    if (cmd.header.frame_id != "")
+    {
+      tf_moveit_to_cmd_frame_ = kinematic_state_->getGlobalLinkTransform(parameters_.planning_frame).inverse() *
+                                kinematic_state_->getGlobalLinkTransform(cmd.header.frame_id);
+    }
     Eigen::Vector3d translation_vector(cmd.twist.linear.x, cmd.twist.linear.y, cmd.twist.linear.z);
     Eigen::Vector3d angular_vector(cmd.twist.angular.x, cmd.twist.angular.y, cmd.twist.angular.z);

```