moveit/moveit #2051: jog_arm: low-pass filter on stop cmd not working

rfeistenauer, 2020-05-05:
### Description

The low-pass filter should also be applied to stop commands like introduced in #1468
This behavior seems not to be working with the current version. 

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source build
* master branch

### Steps to reproduce
Start the jog_arm with a joystick, move in any direction and release the button.
In the rqt_plot a hard stop can be observed

### Expected behaviour
soft stop

### Actual behaviour
hard stop

### Backtrace or Console output
[image]

henningkayser, 2020-05-06:
AFAIK the low-pass filters are now reset to the current joint state if paused or if commands are stale. Does your branch already include https://github.com/ros-planning/moveit/pull/2027? So with stale commands I would expect the last valid (filtered) state to be published and then switching to the actual current joint state received via `/joint_states`. Can you confirm that JogArm is publishing 0 velocities right after stopping the jog commands?
@AndyZe do you have any insights about this?

AndyZe, 2020-05-06:
It looks like this issue requests the exact opposite of the behavior that was requested in #1468. There are pros/cons to doing it either way. Maybe we should make this behavior configurable eventually. Can you make a PR to configure whether the low-pass filters get applied in a sudden stop situation?

Thanks for the plot. I'm also interested to know the answer to what Henning asked:

`Can you confirm that JogArm is publishing 0 velocities right after stopping the jog commands?`

rfeistenauer, 2020-05-07:
> It looks like this issue requests the exact opposite of the behavior that was requested in #1468.

Reading the description this indeed seems to be the case, I read the issue: https://github.com/ros-planning/moveit/issues/1456 which is closed by #1468 and expected this to be exactly what i needed. Never the less the ramp should be there if no sudden_halt is beeing triggered?

As for the conformation, I will try to confirm that lather this day

rfeistenauer, 2020-05-07:
> Does your branch already include #2027

Jep it does

> Can you confirm that JogArm is publishing 0 velocities right after stopping the jog commands
[image]

25 ms after the 0 joint_jog command the joint_group_position_controller gets the first unchanged msg with "low_pass_filter_coeff: 50" (set it that high to get clearer plot yesterday. not sure what value would be the best)

If needed i can check, if this delay does change with other values, speeds or axis.

rfeistenauer, 2020-05-07:
just realized, the msg continued repeatatly at 10.465 is the same published right before the 0 joint_jog (10.445) just the 2 messages in between are different.