moveit/moveit #2052: Planning adapter adding state to trajectory but not velocity or acceleration

MarqRazz, 2020-05-05:
### Description
I am attempting to run back to back motion on my Motoman robot arm. This includes planning for a move, executing it with the `move_group.execute(plan, wait=True)` command then after it finishes planning a second move and executing it the same way. Every once in a while I get a printout from MoveIt stating:
`ros.moveit_ros_planning: Planning adapters have added states at index positions: [ 0 ]`
 followed by the robot arm rejecting the trajectory because the first point is missing velocity data.
`Validation failed: Missing velocity data for trajectory pt 0`

Here is a screen shot of the trajectory sent to the robot arm where you can clearly see the added state that is missing the velocity and acceleration data...
[image]

Here is a printout from my terminal.
```
ros.moveit_planners_ompl.model_based_planning_context: Planner configuration 'manipulator[PRM]' will use planner 'geometric::PRM'. Additional configuration parameters will be set when the planner is constructed.
ros.moveit_planners_ompl.ompl: manipulator/manipulator[PRM]: Starting planning with 2 states already in datastructure
ros.moveit_planners_ompl.ompl: manipulator/manipulator[PRM]: Created 2 states
ros.moveit_planners_ompl.ompl: Solution found in 0.503224 seconds
ros.moveit_planners_ompl.ompl: SimpleSetup: Path simplification took 0.000002 seconds and changed from 2 to 2 states
ros.moveit_ros_planning: Planning adapters have added states at index positions: [ 0 ]
ros.moveit_ros_move_group.ExecuteTrajectoryAction: Execution request received
ros.motoman_driver: Validation failed: Missing velocity data for trajectory pt 0
```

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source build, master branch

### Steps to reproduce
I have attempted to get this error in simulation but I only see it when working with the hardware.

### Expected behaviour
When appending points to the trajectory the planning adapters should also calculate and fill in the velocity and acceleration.

### Possible fixes
I have traced one of the locations where this error occurs to [fix_start_state_bounds.cpp](https://github.com/ros-planning/moveit/blob/5553507f97f36d8c34678975b1888fb2cff02960/moveit_ros/planning/planning_request_adapter_plugins/src/fix_start_state_bounds.cpp#L197). On line 195 we update the first point in the list to have a duration from the previous point and 197 we add the new waypoint.

I have added the following to fix_start_state_bounds and it fixes the error but it is not the correct way to solve the issue. Please note that prefix_state->getVariableCount() is not the correct length for the manipulator group.
```
res.trajectory_->setWayPointDurationFromPrevious(
          0, std::min(max_dt_offset_, res.trajectory_->getAverageSegmentDuration()));
res.trajectory_->addPrefixWayPoint(prefix_state, 0.0);
std::vector<double> zero_vector(prefix_state->getVariableCount(), 0.0);
ROS_ERROR("Zero vector length: %i", prefix_state->getVariableCount());
res.trajectory_->getFirstWayPointPtr()->setJointGroupVelocities(req.group_name, zero_vector);
res.trajectory_->getFirstWayPointPtr()->setJointGroupAccelerations(req.group_name, zero_vector);
```
Where do you think I should add the velocities and accelerations when prefixing a waypoint? Do you think we should also update the velocities and accelerations for the second waypoint in the list (they are set to zero if we don't)? Any suggestion on how you would like me to calculate them? Simple linear interpolation between waypoint[0] and waypoint[2]?

I have also found that after adding the above fix there is still another adapter that is infrequently prefixing a waypoint and not adding velocity or acceleration. When I find out the location of the error I will update the issue.

gavanderhoorn, 2020-05-05:
I don't know why this is happening, but instead of ad-hoc patching planning adapters to copy timing information, I would perhaps suggest to figure out *why* those states are being added.

The order of planning adapters matters, and the IPTP (or whichever timing adapter you have configured) is typically run last. So I'm really rather curious which planning adapters are adding these points *after* the timing information is added.

gavanderhoorn, 2020-05-05:
> One thing that I find interesting is that the time parameterization did not catch or fix this issue. Here is my list of planning adapters. I expected TOTG to re-sample the trajectory and add timing.
> 
> ```
> <arg name="planning_adapters" value="default_planner_request_adapters/FixWorkspaceBounds
>               default_planner_request_adapters/FixStartStateBounds
>               default_planner_request_adapters/FixStartStateCollision
>               default_planner_request_adapters/FixStartStatePathConstraints
>               default_planner_request_adapters/ResolveConstraintFrames
>               default_planner_request_adapters/AddTimeOptimalParameterization"
>               />
> ```

IIRC, planning adapaters are run *in reverse order* of this list.

So `default_planner_request_adapters/AddTimeOptimalParameterization` is run as the first adapter here.

I'd suggest putting it at the top of the list.

MarqRazz, 2020-05-05:
@gavanderhoorn I noticed that the MoveIt Setup Assistant has recently moved time parameterization to the bottom of the list too but had not tried changing the order. All of my old configurations have it at the top. I can confirm that moving it to the top fixes one of my missing velocity errors.

The other location I am getting empty velocity trajectories is when the arm starts at the goal location.
```
header: 
  seq: 203
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "world"
joint_names: - joint_b
- joint_l
- joint_r
- joint_s
- joint_t
- joint_u
points: 
  - 
    positions: [-0.9521227478981018, -0.42967185378074646, 3.834952076431364e-05, 4.676770913647488e-05, 0.00012271845480427146, -1.012174129486084]
    velocities: []
    accelerations: []
    effort: []
    time_from_start: 
      secs: 0
      nsecs:         0
```

v4hn, 2020-05-05:
> Edit 2: ah, it seems #1439 changed it so AddTimeParameterization is at the end of the list (here).

Damn, you beat me to it :smile: 

Looking at the patch, changing the time parameterization there was a clear mistake and the current order is also broken in that `default_planner_request_adapters/FixStartStatePathConstraints` will not succeed for frames that would have to be resolved by `default_planner_request_adapters/ResolveConstraintFrames`.

gavanderhoorn, 2020-05-05:
So the change to `ompl_planning_pipeline.launch.xml` should be reverted? I'm not intimately familiar with the other planning adapters, so I'm not sure I understand which other changes are needed.

v4hn, 2020-05-05:
@MarqRazz Please have a look at #2054 for a possible fix to your one-waypoint issue.

v4hn, 2020-05-06:
Your issues should be fixed in upstream via #2053 and #2054 .
Please have a look at the first patch and update your moveit config accordingly.

I'll close this issue. Feel free to comment if your problems persist.

closed 2020-05-06 by v4hn