moveit/moveit #2055: Improve PlanningRequestAdapter code&documentation

v4hn, 2020-05-06:
In #2053, maintainers got confused over the internal workings of the `PlanningRequestAdapterChain` and the `PlanningRequestAdapters` it contains.

- Refactor the code to make use of C++11/14 features (esp. lambdas over bind) and add inline comments to improve readability

- Improve [our concepts entry](https://moveit.ros.org/documentation/concepts/) on `PlanningRequestAdapter` and possibly the [plugin documentation](https://moveit.ros.org/documentation/plugins/#planningrequestadapter) to reflect the actual semantics as described in the linked issue.

@davetcoleman: I scribbled [this picture](https://github.com/ros-planning/moveit/pull/2053#issuecomment-624319152) from memory based on another one I had seen years ago, because I could not find it anymore. I guess it might have been part of your documentation effort at WG. Do you have any idea what I'm talking about?

davetcoleman, 2020-05-06:
I've always wanted to refactor the planning request adapter to just use two simple for-loops, I've felt the over-use of virtual function handles makes the code incredibly hard to understand/maintain/debug.

From memory, I believe it currently works like this... let's say you have 3 PlanningRequestAdapters A, B, and C.

It calls things in the following order:

```
callA()
  A.precompute()
  A.callB()
    B.precompute()
    B.callC()    
      C.precompute()
      C.callPlanner()
        [OMPL etc]
      C.postcompute()
   B.postcompute()
  A.postcompute()
done...
```

I've always thought it'd be nicer to work more like:

```
for all PlanningRequestAdapters
  precompute()
callPlanner()
for all PlanningRequestAdapters
  postcompute()
```

Not sure if that answers your question at all.

v4hn, 2020-05-06:
@davetcoleman Hehe, you obviously did not read the other issue :smiley: 
I was asking whether you have a better version of the picture I linked to.

I agree with your description.
It is a rather cumbersome interface to change though and the `precompute`/`postcompute` methods are not specified in the current interface. Changing it seems a bad idea to me if the only merit is simpler code in the very backbone of MoveIt.
I would prefer refactoring the code to avoid `bind` hell and add enough comments to make it easy to understand.

rhaschke, 2020-05-06:
I agree with Michael, changing the API of PRAs is not an option.

davetcoleman, 2020-05-28:
> changing the API of PRAs is not an option.

Changing any API is an option, in my opinion. Its simply a matter of doing it between distributions, and in a documented way, and perhaps providing a backwards compatibility layer (but I don't think that is very important, actually)

v4hn, 2020-05-28:
> Changing any API is an option, in my opinion.

Of course it is, assuming it adds some requested feature or fixes some (safety-)critical issue.
Changing API just to make a currently working code base *slightly* simpler is just a pastime to annoy users though. ;)