moveit/moveit #2056: [current_state_monitor / move_group] Access past the end of an array, SEGFAULT

tylerjw, 2020-05-06:
### Description

Copied from here: https://github.com/ros-planning/moveit/pull/2031

The length of the array is 6 and the code is accessing the 6th element (one past the end of the array) of it (from current_state_monitor.cpp:450 with a ros warn added by me):

```c++
      std::vector<double> new_values(joint->getStateSpaceDimension());
      ROS_WARN_STREAM("new_values size: " << new_values.size());
      const moveit::core::LinkModel* link = joint->getChildLinkModel();
      if (link->jointOriginTransformIsIdentity())
        joint->computeVariablePositions(tf2::transformToEigen(transf), new_values.data());
```

This seems fine except `computeVariablePositions` takes a raw pointer to a double and has this code (floating_joint_model.cpp:223):
```c++
void FloatingJointModel::computeVariablePositions(const Eigen::Isometry3d& transf, double* joint_values) const
{
  joint_values[0] = transf.translation().x();
  joint_values[1] = transf.translation().y();
  joint_values[2] = transf.translation().z();
  Eigen::Quaterniond q(transf.rotation());
  joint_values[3] = q.x();
  joint_values[4] = q.y();
  joint_values[5] = q.z();
  joint_values[6] = q.w();
}
```

The logs I get before the segfault have the length of that array at 6.
```
[ WARN] [1588755613.247947347]: new_values size: 6
```

This can be reproduced on kinetic in docker using my PR by changing move_group to run with a valgrind prefix which will print this after the crash:

```
[ WARN] [1588755612.537288439]: new_values size: 6
==5657== Thread 7:
==5657== Invalid write of size 8
==5657==    at 0x9E48DBA: moveit::core::FloatingJointModel::computeVariablePositions(Eigen::Transform<double, 3, 1, 0> const&, double*) const (floating_joint_model.cpp:232)
==5657==    by 0x5405D78: planning_scene_monitor::CurrentStateMonitor::tfCallback() (current_state_monitor.cpp:455)
==5657==    by 0x540FFBA: boost::_mfi::mf0<void, planning_scene_monitor::CurrentStateMonitor>::operator()(planning_scene_monitor::CurrentStateMonitor*) const (mem_fn_template.hpp:49)
==5657==    by 0x540F22D: void boost::_bi::list1<boost::_bi::value<planning_scene_monitor::CurrentStateMonitor*> >::operator()<boost::_mfi::mf0<void, planning_scene_monitor::CurrentStateMonitor>, boost::_bi::list0>(boost::_bi::type<void>, boost::_mfi::mf0<void, planning_scene_monitor::CurrentStateMonitor>&, boost::_bi::list0&, int) (bind.hpp:253)
==5657==    by 0x540E39F: boost::_bi::bind_t<void, boost::_mfi::mf0<void, planning_scene_monitor::CurrentStateMonitor>, boost::_bi::list1<boost::_bi::value<planning_scene_monitor::CurrentStateMonitor*> > >::operator()() (bind.hpp:893)
==5657==    by 0x540CFC2: boost::detail::function::void_function_obj_invoker0<boost::_bi::bind_t<void, boost::_mfi::mf0<void, planning_scene_monitor::CurrentStateMonitor>, boost::_bi::list1<boost::_bi::value<planning_scene_monitor::CurrentStateMonitor*> > >, void>::invoke(boost::detail::function::function_buffer&) (function_template.hpp:159)
==5657==    by 0x8F40A3D: boost::signals2::detail::signal0_impl<void, boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void ()>, boost::function<void (boost::signals2::connection const&)>, boost::signals2::mutex>::operator()() (in /opt/ros/kinetic/lib/libtf2.so)
==5657==    by 0x8F35D65: tf2::BufferCore::testTransformableRequests() (in /opt/ros/kinetic/lib/libtf2.so)
==5657==    by 0x8F36604: tf2::BufferCore::setTransform(geometry_msgs::TransformStamped_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) (in /opt/ros/kinetic/lib/libtf2.so)
==5657==    by 0x5D50A8F: tf2_ros::TransformListener::subscription_callback_impl(ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&, bool) (in /opt/ros/kinetic/lib/libtf2_ros.so)
==5657==    by 0x5D57672: boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, tf2_ros::TransformListener, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<tf2_ros::TransformListener*>, boost::arg<1> > >, void, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&) (in /opt/ros/kinetic/lib/libtf2_ros.so)
==5657==    by 0x5D554DB: boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const>) (in /opt/ros/kinetic/lib/libtf2_ros.so)
```

This was extra tricky to find because the way the binary for move_group works is confusing to me.  It doesn't look like it ever passes through main as I can't get it to break on main using gdb with this bug.  Using valgrind I was able to manually follow this stack trace and add some instrumentation.

I don't understand what this vector of doubles represents, how it can get allocated to different lengths, but maybe it shouldn't be passed as a raw pointer?  That sort of defeats the niceties of the vector container.

v4hn, 2020-05-06:
For future requests: please link to the github pages of the code instead of quoting plain code.

v4hn, 2020-05-06:
> How have we not discovered this before? Maybe in most builds, it just harmlessly corrupts some memory that doesn't matter?

It probably does. Aside from that, Floating Joints are actually not used a lot.
Could you propose a pull-request to fix the reported dimension for this joint type?

tylerjw, 2020-05-06:
I can change the magic number from 6 to 7, but shouldn't I instead change the interface of the function that accepts a raw pointer and then writes 7 doubles?

v4hn, 2020-05-06:
> I can change the magic number from 6 to 7,

... as well as cross-check the code-base for uses of this function of course.

> but shouldn't I instead change the interface of the function that accepts a raw pointer and then writes 7 doubles?

No, I don't think it's a good idea to change the whole API of `RobotState` and the individual models to fix an out-of-bounds access even back in kinetic.

The current pointer-based API in these low-level classes was introduced by one of the original authors (I believe Ioan) in an effort to speed up the system.

You are encouraged to provide a better allocation-free API and adjust the internal calls, but we should only consider removing the pointer-based methods in `master`.
Personally I don't see a reason to remove these interfaces. If you know about your memory it is perfectly reasonable to use them.

tylerjw, 2020-05-06:
> If you know about your memory it is perfectly reasonable to use them.

Except our code takes a magic number from one place as the length and sets it to be the length of an array we allocate on the stack and then pass it into another method that writes past the end of that array.

This seems like a poor attempt at performance optimization where we are creating a vector (gross, dynamic memory allocation affecting response time of code at run-time), and then not passing the length of the dynamic memory we just created to a function call.  If the compiler did when optimizing on the construction of that vector the size is in a register and at most we are saving a few bytes of stack space and probably no reduced instructions.  

We are ok with creating shared_ptr everywhere and yet passing the length of an array that was dynamically allocated is really going to affect our performance?  Are we running out of stack space on the microcontroller we are running this on?

v4hn, 2020-05-06:
Fully agreed, although you are being overly polemic.
I'm happy to review patches that improve interfaces here.

Not for `kinetic-devel`, but for `master`.

Talking about `shared_ptr`, this particular hell starts at the level of handling `RobotState` and `PlanningScene` objects and is not present in the lower-level classes which might, for example, be used in sampling-based planning or optimizing IK.

tylerjw, 2020-05-06:
I'll work on finding ways to improve this while trying to avoid changing the interfaces if possible.  The change I want is for `master` and not `kinetic-devel` anyway.  I discovered this by doing a source build of master on `kinetic`.  :+1: 

I'm sorry for being overly argumentative.  I see what you mean about `RobotState` having more uses than where it is used here.

This method could take a `std::array<double, 7>*` instead of the `double*` and the compiler would do all the length checks and we'd keep all the performance of using a raw pointer... the downside is everything that uses this will now have to change their memory storage.  However, that might be a good thing since we are confused about what is stored in this array anyway.

rhaschke, 2020-05-06:
`computeVariablePositions()` is a virtual function. Thus you cannot just pass a fixed-length array, because other joint types require other array lengths.

v4hn, 2020-05-06:
> This method could take a `std::array<double, 7>*` instead of the `double*` and the compiler would do all the length checks and we'd keep all the performance of using a raw pointer...

These interfaces were never considered for C++11 benefits (or even use of references in many places).
Feel free to propose such extensions. There are limits to this though, because we abstract the joint model type and thus generalize to different state sizes.

> the downside is everything that uses this will now have to change their memory storage.

I would expect both interfaces will live alongside each other, but we could improve our internal use where necessary.

> However, that might be a good thing since we are confused about what is stored in this array anyway.

I wouldn't say we are confused about it at all.
I just didn't look at the data structures before posting (always a bad idea...).

tylerjw, 2020-05-06:
Fixed by: https://github.com/ros-planning/moveit/pull/2031

I'll look for small ways this can be refactored to improve the interface.

tylerjw, 2020-10-02:
Closing because the segfault was fixed.

closed 2020-10-02 by tylerjw