moveit/moveit #2061: Remember open tab in RViz panel

v4hn, 2020-05-07:
Since #1991 a lot of state in the Motion Planning display in rviz is saved.
The currently open tab of the display panel (Context, Planning, Scene Objects...) is not saved though.

Did you leave that out on purpose @rhaschke ?

I would prefer to end up in the Planning tab for most of my configurations, so I'm interested in saving this too.

rhaschke, 2020-05-07:
I kind of left that out on purpose, yes. I think we should just default the planning tab (instead of the first one).

v4hn, 2020-05-07:
> I kind of left that out on purpose, yes. I think we should just default the planning tab (instead of the first one).

I thought about that option too.

But that way, most users would probably never look at any of the other tabs and selecting different planning algorithms & parameterizations is in theory one of the main selling points of the framework.

Looking for opinions by other core contributors/maintainers here.

gavanderhoorn, 2020-05-07:
> But that way, most users would probably never look at any of the other tabs and selecting different planning algorithms & parameterizations is in theory one of the main selling points of the framework.

Personally I'm not sure there is a "best" option here: if users always see the first tab on startup, they may never look at the others. If we save MoveIt configuration pkgs with any tab but the first active, and this is persisted and restored, users may have trouble finding the tab described "everywhere" (ie: the first one).

felixvd, 2020-05-07:
I think the tabs are sufficiently discoverable without forcing an extra click on every startup, so I'm in favor of defaulting to the Planning tab.

There's a larger discussion to be had about the selling points of the framework, but I suspect that comparing different motion planning algorithms isn't at the forefront anymore necessarily. Prototyping, production and overall usability seem to be becoming more and more important.

rhaschke, 2020-07-20:
#2198 defines the planning tab as default.