moveit/moveit #2066: July Melodic Release

tylerjw, 2020-05-07:
I propose we cut our next release of Melodic in the first week of July.  To do this I created this issue as a place to figure out what we will include in that release and start getting those changes backported (once the current melodic-devel is synced).

For our July Melodic release, I'd like to release these two features.  @felixvd and @AndyZe can you please find the commits for these features that need backported and any pending work that you want finished before we backport:

- [ ] Subframes
- [ ] Jog Arm

Also here are some bug fixes and smaller PRs that happened recently.  To try to figure out what differs here is the output of a git log command showing the difference. 
 Lines that start with > are on upstream/master and ones that start with < are only on upstream/melodic-devel.  You'll also see there is a `reviewed-for-backport` branch, does that mark all the commits we are ready to merge onto melodic-devel, and the ones I should be concerned with are the ones that are more recent than that?

`moveit$ git log --left-right --graph --cherry-pick --oneline upstream/melodic-devel...upstream/master`

```c++
> efa7fdca8 Depend on geometric_shapes noetic-devel branch (#2059)
>   23fbea189 Merge pull request #2002 from felixvd/patch-2
|\  
| > 597dd5064 fixup! Fix TfPublisher subframe publishing
| > 60c78f29e fixup! Fix TfPublisher subframe publishing
| > dc71f6de6 Fix TfPublisher subframe publishing
> | 00f1ed933 move NOLINT instructions to intended positions (#2058)
> |   2bebf052a Merge pull request #2054 from v4hn/pr-master-totg-one-waypoint
|\ \  
| > | 12797f117 totg: add a correct time parameterization for 1-waypoint trajectories
| > | 1bfa63f0a RS: add interfaces to zero and remove dynamics
> | | 7d7bd020c fix ordering of request adapters (#2053)
|/ /  
> |   37bbccff8 Merge #2050: clang-tidy fixes
|\ \  
| > | 74bc601dc clang-format
| > | e619b6ab8 performance-*
| > | e4c6bd321 readability-identifier-naming
| > | 1d44aceec modernize-loop-convert
| > | 972b8836f modernize-redundant-void-arg, modernize-use-nullptr, modernize-use-default, modernize-use-override
| > | 7306d1df7 llvm-namespace-comment
> | | 5553507f9 Fix mesh_filter test (#2044)
> | | 2dacfe990 [ci] configure travis to report success before code-coverage finishes (#2041)
|/ /  
> | ad88e18ae [jog_arm] Keep updating joints, even while waiting for a valid command (#2027)
 /  
| > 9a9f77923 commander: python3 fix (#2030)
| > 173d6a611 collision world: check for empty shapes vector before access (#2026)
| > 4d3062164 [jog_arm] Fix param logic bug for self- and scene-collision proximity thresholds (#2022)
| > 3be547d1c [jog_arm] Split collision proximity threshold (#2008)
| > 47d3d616b add use_rviz to demo.launch in setup_assistant (#2019)
| > 9681f5064 add tests for move_group interface (#1995)
| > 18715e1c2 [jog_arm] Fix valid command flags (#2013)
| > d989220fb run shadow-fixed for main branches (#2012)
| > 488055efd Travis: re-enable ros-shadow-fixed (#2003)
| > d4ade526e Fix some clang tidy issues (#2004)
| >   034b10541 [jog_arm] Fix low-pass filter initialization (#1982)
| |\  
|/ /  
| > 2627d7eed Pause/stop JogArm threads using shared atomic bool variables
| > 85be9f8f2 Add pause/unpause flags for jog thread
| > 2e4c08d52 Verify valid joints by filtering for active joint models only
| > d168cd9ea Remove redundant joint state increments
| > ef16990f7 Wait for initial jog commands in main loop
|/  
> a6c9852a9 Travis: send emails to author/committer
> fa6b72dad temporarily skip ros-shadow-fixed (#2000)
> 5298b9ff4 Modernize Travis config (#1999)
> 379acf91a Integration test to defend subframe tutorial (#1757)
> 13bf03e93 Remove duplicate collision check in JogArm (#1986)
> 490622f02 Fix errors: catkin_lint 1.6.7 (#1987)
> 6601d299d remove unused angles.h includes (#1985)
> a672455de Remove @jonbinney as a codeowner (#1981)
> 8946d266b [jog_arm] Add a binary collision check (#1978)
> 0407e868b export  moveit_py_bindings_tools library (#1970)
> 8d497203e Merge pull request #1963 from jonasTUM/bugfix/fix-global-minimal-distance-collision-pairs
> d99295730 collision detection: add test for SINGLE distance request
[... 291 lines cut ...]
< 8e7521b7a Show attached bodies trail in RViz (#1766)
< 677096066 Provide UniquePtr macros (#1771)
< aff8ebb7f Updating deprecation method (#1748)
< 6fe7deb46 add missing dependencies to library (#1746)
< 61b244552 Update version of occupancy_map_monitor (#1693)
< 1fc63cb99 moveit.rosinstall: use MoveIt's melodic-devel branch (#1681)
< d7a149757 Adding documentation to collision detection (#1645) (#1671)
< c3f48c95b Add joint state controller config by default (#1024)
< 895ded362 Use portable string() on filesystem::path. (#1571)
< 0d0c399bb move occupancy_map_monitor into its own package (#1533)
< b9a1e8e64 1.0.2
< 91f55a580 chomp adapter: Fix spurious error message
< afa114ccd Add get_jacobian_matrix to moveit_commander (#1501)
<   1c6fc04aa Merge #1512: Partially cherry-pick #1473
|\  
| < 0a1142281 fix start-state handling
| < 7062ddc64 reduce verboseness
< 9b693402d Travis: Disable clang-tidy-fix (#1513)
< 4bf7a3d5c docker: remove CATKIN_WS, define ROS_UNDERLAY (#1451) (#1464)
< ae7309ced fix segfault when detaching attached collision object (#1438)
< 482ff84c9 Normalize quaternions when adding new or moving collision objects (#1420)
< d9c6a83e5 add arguments to launch file templates (#1397)
< 94c8d01f7 fixing the test utilities in moveit core (#1409)
< ea0a5f079 small fixes to chomp planner (#1407)
```

v4hn, 2020-05-07:
Thanks for starting the issue.

Please look at "Consider backporting" in [our guidelines](https://moveit.ros.org/documentation/contributing/pullrequests/) for an alternative listing of relevant commits and explanations on the `reviewed-for-backport` branch.

As mentioned in https://github.com/ros-planning/moveit_msgs/issues/71 , we did not break the message format in melodic yet for subframes.
I would propose to avoid this altogether and not backport the feature.
If we backport it, the new origin pose should also be included (not fully merged yet) and @felixvd currently works on some more semantic changes...
Part of the commits related to it are prior to the backport branch by now,
so you'll have to look at them again if we really want to add it.

tylerjw, 2020-05-07:
Would a September release of Melodic be a better target for Subframes then?  That would simplify the July release to be just Jog-arm and bug fixes.

AndyZe, 2020-05-07:
What I would recommend is copy/paste the whole jog_arm folder into moveit_experimental in a single commit, instead of trying to cherry-pick all the related changes.

I think that makes sense because there would be quite a few commits to cherry-pick. jog_arm doesn't exist in Melodic at all

davetcoleman, 2020-05-07:
Thanks for taking this initiative @tylerjw 

Per https://github.com/ros-planning/moveit/issues/1747#issuecomment-624673572, PickNik will lead the actual release steps this time

tylerjw, 2020-05-07:
I adjusted the above message to remove the subframes feature from this release.  We will revisit it on the next release.

For jog arm there are two options for the merge:

1. Merge each PR that has created jog_arm
2. Copy the folder in its current state into melodic-devel

@rhaschke please let us know what your preference is so we can prepare the PR for it.

For non-jog_arm related changes since the latest release, I'll make a list so we can decide which ones should be back-ported and update the description above.

v4hn, 2020-05-08:
> Merge each PR that has created jog_arm

Following this approach, there should of course be only one pull-request combining all commits in `master`.
As jog_arm is very much isolated and all related commits should *only ever* have touched the respective folders, I guess it should be a matter of `git log -S` or similar to find and pick all commits.
I would strongly prefer that.

v4hn, 2020-05-08:
> Would a September release of Melodic be a better target for Subframes then?

First point for the maintainer meeting. I believe there are different stances on whether we can still break messages in melodic.

tylerjw, 2020-05-08:
Ok, here are all the changes that have touched the jog_arm folder as of now.  Please look through these and let me know if any of these will be a problem to backport?

7306d1df7 llvm-namespace-comment
ad88e18ae [jog_arm] Keep updating joints, even while waiting for a valid command (#2027)
4d3062164 [jog_arm] Fix param logic bug for self- and scene-collision proximity thresholds (#2022)
3be547d1c [jog_arm] Split collision proximity threshold (#2008)
18715e1c2 [jog_arm] Fix valid command flags (#2013)
d4ade526e Fix some clang tidy issues (#2004)
2627d7eed Pause/stop JogArm threads using shared atomic bool variables
85be9f8f2 Add pause/unpause flags for jog thread
2e4c08d52 Verify valid joints by filtering for active joint models only
d168cd9ea Remove redundant joint state increments
ef16990f7 Wait for initial jog commands in main loop
13bf03e93 Remove duplicate collision check in JogArm (#1986)
8946d266b [jog_arm] Add a binary collision check (#1978)
6d3b89904 [jog_arm] Publish more detailed warnings (#1915)
e34a07d89 jog_arm: use wait_for_service() to fix flaky tests (#1946)
8b2a08145 Fix versioning (#1948)
5327e4197 [jog_arm] SRDF velocity and acceleration limit enforcement (#1863)
ba4b60e07 Replace namespaces robot_state and robot_model with moveit::core (#1924)
6eaa75d69 JogArm C++ API fixes (#1911)
26863598b [jog_arm] a ROS service to enable task redundancy (#1855)
7193a4ea2 Fix segfault with uninitialized JogArm thread (#1882)
15cb1a2dc Add warnings to moveit_jog_arm low pass filter (#1872)
f6308f405 Fixup: Use CMAKE_CXX_STANDARD to enforce c++14 for portability (#1607)
83728716e Fix the publishWarning() functionality
75b79108f Always provide the end effector transform, not just while robot is jogging
49892146c [jog_arm] Rework the halt msg functionality (#1868)
4f996c71a [jog_arm] Various small fixes (#1859)
1389df263 [jog_arm] Improve formatting in comments
75104b0d4 jog_arm: prevent a crash at velocity limit (#1837)
2a8559503 [jog_arm] Remove scale/joint parameter (#1838)
cd16a5033 [jog_arm] Pass planning scene monitor into cpp interface (#1849)
92032b516 Move attribution below license file, standardize with MoveIt (#1847)
3a57141b4 Reduce console output warnings (#1845)
546d6e052 Fix command frame transform computation (#1842)
06a9e4bea moveit_jog_arm: fix dependencies + catkin_lint issues
844a02128 update link transforms before calling checkCollision on robot state in jog_arm (#1825)
bc752b7c8 Add atomic bool flags for terminating JogArm threads gracefully (#1816)
e0b4f854e moveit_jog_arm: Get transforms from RobotState instead of TF (#1803)
76c16e613 jog_arm: add a C++ API (#1763)
b35e64084 Merge #1773: Fix compiler warnings
8bdf9e57b fix unused parameter warnings
4ff87ded5 Update license formatting (#1764)
0924e99de Unify jog_arm package to be C++14 (#1762)
2bef2b62a Fix jog_arm segfault (#1692)
ff8f1d850 Fix double mutex unlock (#1672)
ff71fc44b Rename jog_arm->moveit_jog_arm (#1663)

rhaschke, 2020-05-08:
1. I think we agreed - already multiple times - that we should only backport bug fixes. IMHO this excludes both, jog arm and subframes support from melodic. We should have some features only in master / noetic that will motivate people to migrate :wink:

2. I think separating the processes of selecting individual commits and actually filing PRs for backports is not very sensible: In order to decide, you need to inspect the commits anyway (for API changes). If you do so, you can also directly apply the commit to your local MoveIt, test-compile it, and finally file the PR as Michael has done recently.
Such a preselection only makes sense on an abstract level, say like we want backporting of features / bug fixes A, B, and C (as Tyler started out with).

tylerjw, 2020-05-08:
In response to #2, I should have been more clear.  I was hoping that you would look over the list and call out any you know you specifically don't want backported.

I figured we'd still have a backport PR where we would test the ones we thought were a good idea and potentially remove some we find are not from that list then too.

tylerjw, 2020-05-08:
In response to 1, I do like the idea of getting people to migrate to Noetic and think subframes would be excellent for this as it breaks the messages API.

Jog arm (to my knowledge) doesn't break any existing API in MoveIt and is a straightforward new feature.  My motivation for getting it released with MoveIt is we are supporting a client right now that is using it and having it released would be helpful because it would create the jog_arm melodic rosdistro symbols to be able to install it.  This would greatly simplify us distributing debians for jog_arm.

v4hn, 2020-05-08:
> I think we agreed - already multiple times - that we should only backport bug fixes.

I am rather convinced I did not agree to this and I'm clearly not alone.
Once noetic is out and MoveIt users actually started using it, I will of course change my opinion.

We agreed *not to break API*.

>  I was hoping that you would look over the list and call out any you know you specifically don't want to be backported.

- you missed that the project was originally added under the name of `jog_arm` and thus you missed a number of commits (as you would have noticed if you tried cherry-picking :) )

- a number of clang-tidy fixes swept through the whole code base in one commit.
  If they apply cleanly just add them, sure thing. If they don't, you can trim the commits to touch the `jog_arm` package only (and adjust the commit message accordingly).

tylerjw, 2020-05-08:
> * you missed that the project was originally added under the name of `jog_arm` and thus you missed a number of commits (as you would have noticed if you tried cherry-picking :) )

I didn't know that, thank you.  I'll see if I can figure out how to do the cherry-pick.

My comment you quoted was directed at the list in the description at the top of non jog_arm changes.

For the clang-tody fixes, that is a good point.

tylerjw, 2020-05-08:
I spent an hour or so trying to do cherry-picks and then an interactive rebase searching through the history to find the right place to insert each commit to get the jog_arm stuff to cleanly apply... and so far have failed.

@v4hn do you have any tricks for doing this with such a huge set of changes?  I'm sadly approaching the opinion that cleanly applying jog_arm might be too hard and therefore we should just target Noetic with it.  :cry: 

You said something about partially applying changes.  Do you know of any way to ask git to apply some changes to only the files in a certain directory?

tylerjw, 2020-05-11:
I created a PR for backporting changes that were clean to apply: https://github.com/ros-planning/moveit/pull/2080

As to the clang-tidy fixes, they touch so much of the code I wasn't able to get them to cleanly apply.  This mostly was because there are previous clang-tidy changes that are also not backported.  Is the normal approach here to wait for the next release for repository wide changes like these?

v4hn, 2020-05-11:
> As to the clang-tidy fixes, they touch so much of the code I wasn't able to get them to cleanly apply.
> This mostly was because there are previous clang-tidy changes that are also not backported.

The previous tidy commands also resulted in a lot of conflicts and I ultimately decided it wasn't worth the effort.
Instead, I fixed other conflicts because of the missing tidy commits by hand.

> @v4hn do you have any tricks for doing this with such a huge set of changes?
> I'm sadly approaching the opinion that cleanly applying jog_arm might be too hard and therefore we should just target Noetic with it. cry

This is really straight-forward with the right tool! But also badly documented if you don't know it. -.- https://github.com/newren/git-filter-repo works well for me:

```
git clone https://github.com/ros-planning/moveit
cd moveit
git filter-repo --path moveit_experimental/moveit_jog_arm/ --path moveit_experimental/jog_arm --refs master
git rebase melodic-devel
```

Make sure to try on fresh clones to avoid scrambling your workspace repository.

tylerjw, 2020-05-12:
I created #2083 for the jog_arm backport.  Unfortunately, it looks like it depends on this change: https://github.com/ros-planning/moveit/pull/1584 which to me looks like it introduces breaking API changes.  Am I interpreting that correctly or is there another way to make this work?  

@AndyZe would there be a way to modify jog_arm to make it work with the old internal API easily for melodic-devel?

henningkayser, 2020-05-13:
> I created #2083 for the jog_arm backport. Unfortunately, it looks like it depends on this change: #1584 which to me looks like it introduces breaking API changes. Am I interpreting that correctly or is there another way to make this work?

@tylerjw fixed it with https://github.com/ros-planning/moveit/pull/2083/commits/23947b2c180ec47dcc836dc35faceca5085c3e0e
I don't think that there would be any other issues with the backport.

v4hn, 2020-05-21:
Looking at this again, the timeline looks weird:

    June 15 - Create shadow repo for testing and post on discord
    July 1 - Run bloom

I might have misunderstood something here,
but to get into `ros-shadow-fixed`, you have to *run bloom*?
Or did you mean something else by "shadow repo"?

tylerjw, 2020-05-21:
> I might have misunderstood something here,
> but to get into `ros-shadow-fixed`, you have to _run bloom_?
> Or did you mean something else by "shadow repo"?

I think this is just my misunderstanding of our release guide.  I read that those were steps we did in that order even though I don't understand what they mean.  Please advise what it should be?

tylerjw, 2020-05-21:
@v4hn can you move the `reviewed-for-backport` as all backports for this release have been merged?  I don't think I have the permissions to do that.

v4hn, 2020-05-21:
> can you move the reviewed-for-backport

done. Doing so, I noticed d3bd523ee0ec296656cf964c1b7af190e3bfe9f9 is not ported yet and @simonschmeisser [wants to backport his RViz speedup too](https://github.com/ros-planning/moveit/pull/2049#issuecomment-632001140).
We will definitely not postpone the schedule for new backports, but I guess we will have another backports request to be merged before code freeze.

> I think this is just my misunderstanding of our release guide. I read that those were steps we did in that order even though I don't understand what they mean. Please advise what it should be?

I just noticed the timeline collides with WMD too.
[WMD2020 is scheduled on June 2](https://discourse.ros.org/t/world-moveit-day-hackathon-june-2nd/13974). I would propose to have the `melodic-devel` freeze (and discourse announcement) on June 15 and release/bloom on July 1st.
@rhaschke comments, additions, misunderstandings?

tylerjw, 2020-05-21:
@v4hn I'm ok with pushing back code freeze and backporting more stuff and giving opportunities for WMD improvements to make it in.  

I had previously assumed we'd want to let most new changes sit in master for a couple of months as a matter of policy to make it easier to have confidence in the stability of the release.  I want to push releases out every quarter though so the backporting isn't so painful and so improvements contributors submit are released in a reasonable amount of time.

My other thought was that this Melodic release wouldn't matter much to WMD and after this goes out we'd all mostly focus on Noetic and continue to only backport to Melodic things that were easy to backport (like the two PRs you referred to).  This is a good justification for getting more in this release as we might be less motivated after we start to focus on Noetic.  

I'm torn on what the right policy is here.

v4hn, 2020-05-21:
> I had previously assumed we'd want to let most new changes sit in master for a couple of months as a matter of policy to make it easier to have confidence in the stability of the release.

Makes sense as well. Having a known and patched write-past-buffer (which has been in there for a long time) in a release would make my stomach cringe though..

> My other thought was that this Melodic release wouldn't matter much to WMD

Unless someone wants to contribute something specifically to melodic (quite a likely goal for MoveIt *users*).
I don't mind not including these things in the upcoming release yet, but the branch freeze would be unfortunate...
(or we just handle it on a case-by-case basis, as we did in the past)

> and after this goes out we'd all mostly focus on Noetic

Muhahaha.

> and continue to only backport to Melodic things that were easy to backport (like the two PRs you referred to).
> This is a good justification for getting more in this release as we might be less motivated after we start to focus on Noetic.

I expect at least half a year of overlap, possibly more.
Together with your release schedule that means it is surely not the last release that includes bigger contributions from `master`.

simonschmeisser, 2020-05-21:
It's not super important for me to have the PlanningSceneDisplay speedups included in the next release as we build from source. I'll open a backport PR anyway as I need a branch to build from

tylerjw, 2020-05-21:
@simonschmeisser thank you for offering to open a backport PR for your change.  If others agree it is simple enough we can get it in this release, if not we can target it for the next.

> Having a known and patched write-past-buffer (which has been in there for a long time) in a release would make my stomach cringe though...

@v4hn yea, mine too.  I don't think to fix that will hurt anyone and we can probably include it in this one easily enough.

tylerjw, 2020-05-21:
> and after this goes out we'd all mostly focus on Noetic
> Muhahaha.

my feelings exactly... :laughing: :speak_no_evil: 

In all seriousness, the more we can get users on Noetic (if that's possible in WMD) the nicer it will be for all of us.

tylerjw, 2020-06-24:
Coming back here and iterating on this, I'm done changing jog_arm for the time being and would like to back-port the recent changes before this release.

tylerjw, 2020-07-01:
@v4hn I believe all the backports for this release have been merged now.  Do you mind moving the `reviewed-for-backport` tag?

I'll start doing the next steps after that myself.  We should open a new issue where people can start listing the changes they want to backport to the next meldoic release and when it should be ( September? ).

tylerjw, 2020-07-02:
To confirm the version for this release is 1.0.5 ?

rhaschke, 2020-07-02:
Yes, 1.0.5 will be the next Melodic release. Please file a PR before committing your first changelog.
Have a look at previous changelogs to get an idea of the targeted level of detail and the grouping of commits.

tylerjw, 2020-07-02:
:+1: 

I'm working on the changelists and a PR was my plan for that.  I'm also running the pre-release tests.  Is there anything I need to do to show that I ran them and they worked?

rhaschke, 2020-07-02:
> I'm also running the pre-release tests. Is there anything I need to do to show that I ran them and they worked?

No, I trust your word!

tylerjw, 2020-07-02:
Here is the PR for the Changelists... there were allot of changes: https://github.com/ros-planning/moveit/pull/2184

For the pre-release tests... these two are failing... :cry: 
```
moveit_ros_planning_interface/MISSING-rostest-test_move_group_interface_cpp_test.xml: 1 tests, 0 errors, 1 failures, 0 skipped
moveit_ros_planning_interface/MISSING-rostest-test_moveit_cpp_test.xml: 1 tests, 0 errors, 1 failures, 0 skipped
```

I'll look into figuring out why next.  If anyone has tricks for debugging tests failing in the pre-release process please let me know.  These are two tests that I've contributed to this time around so it is probably my fault they are failing.

rhaschke, 2020-07-02:
I will have a look into the changelogs tomorrow.
The pre-release tests run in a docker container, right? Try to save those containers, re-enter them, and manually execute the test there.

rhaschke, 2020-07-03:
I started a local build of melodic-devel as well an this is the outcome:
```
build/moveit_ros_perception/test_results/moveit_ros_perception/gtest-mesh_filter_test.xml: 48 tests, 0 errors, 2 failures, 0 skipped
build/moveit_servo/test_results/moveit_servo/rostest-test_launch_servo_drift_dimensions_service_test.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/moveit_servo/test_results/moveit_servo/rostest-test_launch_servo_halt_msg_test.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/moveit_servo/test_results/moveit_servo/rostest-test_launch_servo_msg_reception_test.xml: 1 tests, 0 errors, 1 failures, 0 skipped
build/moveit_servo/test_results/moveit_servo/rostest-test_launch_servo_vel_accel_limits_test.xml: 1 tests, 0 errors, 1 failures, 0 skipped
```
The mesh filter test fails as follows:
```xml
<testcase name="float/7" value_param="3.5" status="run" time="0.086" classname="float_test/MeshFilterTestFloat">
<failure message="/tmp/ros/src/moveit/moveit_ros/perception/mesh_filter/test/mesh_filter_test.cpp:234 The difference between filtered_depth[idx] and gt_depth[idx] is 3.599998950958252, which exceeds 1e-6, where filtered_depth[idx] evaluates to 3.599998950958252, gt_depth[idx] evaluates to 0, and 1e-6 evaluates to 9.9999999999999995e-07." type="">
<![CDATA[ /tmp/ros/src/moveit/moveit_ros/perception/mesh_filter/test/mesh_filter_test.cpp:234 The difference between filtered_depth[idx] and gt_depth[idx] is 3.599998950958252, which exceeds 1e-6, where filtered_depth[idx] evaluates to 3.599998950958252, gt_depth[idx] evaluates to 0, and 1e-6 evaluates to 9.9999999999999995e-07. ]]>
</failure>
</testcase>
```

The others seem to fail due to an pytest issue:
```
Traceback (most recent call last):
  File "/tmp/ros/src/moveit/moveit_ros/moveit_servo/test/python_tests/drift_dimensions_service/test_drift_dimensions_service.py", line 4, in <module>
    import pytest
  File "/usr/lib/python2.7/dist-packages/pytest.py", line 13, in <module>
    from _pytest.fixtures import fixture, yield_fixture
  File "/usr/lib/python2.7/dist-packages/_pytest/fixtures.py", line 842, in <module>
    class FixtureFunctionMarker(object):
  File "/usr/lib/python2.7/dist-packages/_pytest/fixtures.py", line 844, in FixtureFunctionMarker
    params = attr.ib(convert=attr.converters.optional(tuple))
TypeError: attrib() got an unexpected keyword argument 'convert'
```
I'm using the ros-testing binaries. Maybe pytest did a recent release, which is broken (but not yet synced).
If that's the reason, Travis should fail as well. I retriggered the corresponding build job:
https://travis-ci.com/github/ros-planning/moveit/jobs/356539442

rhaschke, 2020-07-03:
The pytest errors on my machine were related to an unrelated, incompatible update of the attr package on my machine:
https://stackoverflow.com/questions/58189683/typeerror-attrib-got-an-unexpected-keyword-argument-convert
Sorry for the noise.

AndyZe, 2020-07-03:
We should delete those anyway, they're redundant with C++ tests now. I can make a PR unless one of you beats me to it. (Assuming this is miveit_servo)

rhaschke, 2020-07-03:
Would be great if we could get rid off the ros_pytest dependency :wink:

AndyZe, 2020-07-03:
A PR to delete the Python tests is here:  https://github.com/ros-planning/moveit/pull/2186

v4hn, 2020-07-03:
> I believe all the backports for this release have been merged now. Do you mind moving the reviewed-for-backport tag?

I looked at it again and came up with #2190 and #2189 . Let's keep the tag where it is until both are merged (the latter one after the release). I documented the target commit in the new pull-request.

tylerjw, 2020-07-03:
@v4hn thank you for doing that.  Those both look reasonable (and good) to backport (especially the first one).  Thank you for checking on which ones were missing.

rhaschke, 2020-07-04:
Hi @tylerjw,
regarding your pre-release test and the two failing unit tests: Looks like these two tests are failing since May 2020 (#2080) already on the ROS buildfarm:
http://build.ros.org/job/Mdev__moveit__ubuntu_bionic_amd64/changes
The reason is that both of them,
`moveit_ros/planning_interface/test/move_group_interface_cpp_test.test` and 
`move_group_interface_cpp_test`
depend on `panda_moveit_config` (instead of moveit_resources). Should be easy to fix.

tylerjw, 2020-07-06:
With this PR the pre-release tests pass: https://github.com/ros-planning/moveit/pull/2197

tylerjw, 2020-07-06:
This now passes the pre-release tests.

tylerjw, 2020-07-07:
I rebased and updated the changelogs with the recent minor changes.  If @rhaschke approves, I will merge the changelogs tomorrow (July 8th) and run bloom.

tylerjw, 2020-07-14:
I received the permissions needed to bloom a release and I ran it for this.  Here is the rosdistro PR: https://github.com/ros/rosdistro/pull/25816

tylerjw, 2020-07-16:
Does anyone know how the changelogs in the ros index get populated.  The date for the last few releases are there but there are no changelogs.  Should I have put some entries in the moveit meta package?

https://index.ros.org/p/moveit/github-ros-planning-moveit/#melodic

v4hn, 2020-07-16:
It seems Isaac actually copied the changelogs together and added everything relevant to the meta package CHANGELOG as well as to each individual package. Looks like neither Robert nor you followed in that tradition.

tylerjw, 2020-07-16:
I'm sorry, should we add that to the instructions for a releases?

I am writing a summary of the release to announce it on discord soon.  In that I'll attempt to point out the amazing work that people have contributed that is now (or soon to be) available in the distribution release.  I'll post it here first to get feedback on it if anyone cares to comment.

v4hn, 2020-07-16:
> I'm sorry, should we add that to the instructions for a releases?

I guess we simply gave up on that particular tradition..
If you think it's worthwhile to pick it up again, create a pr to document it.

> I am writing a summary of the release to announce it on discord soon.

Cool!

rhaschke, 2020-07-16:
Indeed, I didn't follow this tradition (because I wasn't aware of it). The detailed changelogs are available for all individual packages though. I don't think we need to copy those. Having a summary on discourse would be great. Thanks for your work, @tylerjw.

tylerjw, 2020-07-18:
Discourse announcement draft:

# MoveIt Melodic Release (1.0.5)
Many of the awesome features and bug fixes only available to people building MoveIt on master from source are coming to an apt repo near you.

## moveit_servo (Formally moveit_jog_arm)
Real-time robot arm control.  [So cool, we even got a smile from a famous person while using it.](
https://www.youtube.com/watch?v=uwYtwQtoOh0).  [There is an tutorial for getting started with it.
](https://ros-planning.github.io/moveit_tutorials/doc/arm_jogging/arm_jogging_tutorial.html)
[image]

## moveitcpp
There is a new high level API for users who have more real-time requirements.  [Here is the tutorial for using it.](https://ros-planning.github.io/moveit_tutorials/doc/moveit_cpp/moveitcpp_tutorial.html)

## Other features
* Allow to parameterize input trajectory density of Time Optimal trajectory generation (#2185)
* Added support for hybridize/interpolate flags in ModelBasedPlanningContext via ompl_planning.yaml (#2171)
* Improve responsiveness of PlanningSceneDisplay (#2049)
* PlanningComponent: Load plan_request_params (#2033)
* Improve rviz GUI to add PlanningScene objects. Ask for scaling large meshes. (#2142)
* PlanningSceneDisplay speedup (#2049)
* Added support for PS4 joystick (#2060)
* Add use_rviz to demo.launch in setup_assistant (#2019)

## Bug fixes
* Behaviors (#2113, #2054, #2157, #2076, #2082, #2120)
* Segfaults (#2104, #2106)
* Memory leaks (#2138)
* Python 3 (#2030)

## Contributors
* Andy Zelenak
* Constantinos
* Felix von Drigalski
* Henning Kayser
* Jafar Abdi
* Jere Liukkonen
* John Stechschulte
* Mark Moll
* Michael Görner
* Patrick Beeson
* Robert Haschke
* Simon Schmeisser
* TrippleBender
* Tyler Weaver
* Yeshwanth
* Wolfgang Merkt

v4hn, 2020-07-18:
> “If you don’t like unit testing your product, most likely your customers won’t like to test it either.” – Anonymous

Not sure our current coverage justifies the quote. ;)

> There is a new high level API for users who have more real-time requirements.

I would propose "users who want convenient access to MoveIt's functionality via C++ classes".
Planning with OMPL will not magically become real-time. :D

> * Improve rviz GUI to add PlanningScene objects. Ask for scaling large meshes. (#2142)

This could be easily confused with the later features to add primitives.
That feature will only be available after the next round of reviews (& backports) though.

> ## Contributors

Sorting by first name seems weird, but nobody cares.

tylerjw, 2020-07-18:
>  “If you don’t like unit testing your product, most likely your customers won’t like to test it either.” – Anonymous
> Not sure our current coverage justifies the quote. ;)

I see how this reads now.  I was hoping for a fun goofy feeling to this announcement, but I really am bad at conveying that emotion in writing.

> Planning with OMPL will not magically become real-time. :D

True, and here is an example of me using real-time confusingly when I give others grief for doing that.

I'll revise it based on your feedback, thank you.

tylerjw, 2020-07-18:
> Sorting by first name seems weird, but nobody cares.

I fixed it, it was weird.  This happened because I went through the changelists and put the names in a text file on separate lines and then called sort on that text file and removed duplicates.  Other than doing it manually I'm not sure of an easy way to sort a list alphabetically by the last word.  There is probably some magic sed command that'd do that.

tylerjw, 2020-07-18:
> This could be easily confused with the later features to add primitives. That feature will only be available after the next round of reviews (& backports) though.

I reworded this feature, lmk if that captures what the feature is that did make it into this release?

Thank you for the feedback.  By writing this I realized how bad I am at writing marketing type cheery written announcements.  I learned from my last post (about the jog_arm rename) was a bit dry, I hope the gif helps with that.

v4hn, 2020-07-18:
> I reworded this feature, lmk if that captures what the feature is that did make it into this release?

Good enough.

> By writing this I realized how bad I am at writing marketing type cheery written announcements.

I believe you have good mentors at PickNik. ;)

> I learned from my last post (about the jog_arm rename) was a bit dry, I hope the gif helps with that.

If Dave and/or Andy approve the statement it's fine. Guess I've just seen it too often by now. :-)

henningkayser, 2020-07-18:
Maybe also reference the JogArm tutorial and add a quick one-sentence intro for those who don't know about it yet. Otherwise, I like highlighting the features and changes like this without too many words :+1:

tylerjw, 2020-07-18:
> Maybe also reference the JogArm tutorial

There is a link to the jogarm tutorial.

[2 more replies]

closed 2020-07-27 by tylerjw