moveit/moveit #2078: Use MoveItCpp as MoveGroup context

henningkayser, 2020-05-11:
I would like to let MoveGroup use MoveItCpp internally in order to reduce maintenance overhead and support some nice features like:

* Support multiple planning pipelines with MoveGroup
* Unify launch/configuration setup of MoveGroup and MoveItCpp
* Add support for MoveItCpp in MSA

v4hn, 2020-05-11:
Also makes sure we use the code as part of every standard setup, reducing redundancy and integrating the package.

> Support multiple planning pipelines with MoveGroup

This is a beautiful side-effect :)

felixvd, 2020-05-11:
Does "multiple planning pipelines" imply multiple different planners? If yes that is excellent.

What are the downsides of this change, and how large do you expect it to be?

henningkayser, 2020-05-11:
> Does "multiple planning pipelines" imply multiple different planners? If yes that is excellent.

Yes, MoveItCpp allows you to run multiple planning pipelines for each group. It's pretty straight forward to add this feature to MoveGroup, i.e. by prepending the pipeline name to the planner id.

> What are the downsides of this change, and how large do you expect it to be?

Downsides are that MoveItCpp doesn't allow too many breaking changes anymore ;)

I would probably break this up into multiple steps (still thinking about other issues/improvements). I guess a broad implementation should be doable within a day. Only support for MSA is still a bit unclear (a nice implementation would probably require unifying parts of the move_group.launch 
with a boilerplate setup for MoveItCpp)

closed 2021-03-17 by henningkayser