moveit/moveit #2081: MoveIt compile-time profiling

JafarAbdi, 2020-05-11:
A few weeks ago I did profile MoveIt's compile-time, posting it here for discussion and if anyone is interested, hopefully, we come out with ideas to decrease the compile-time

#### Tools used
- https://aras-p.info/blog/2019/01/16/time-trace-timeline-flame-chart-profiler-for-Clang/
- https://github.com/aras-p/ClangBuildAnalyzer

#### Clang-9 installation and compiling it with catkin,
- `sudo apt install clang-9`
- `catkin build --cmake-args -DCMAKE_CXX_FLAGS="-ftime-trace" -DCMAKE_CXX_COMPILER="/usr/bin/clang++-9"`
it will generate a JSON file for each target which can be viewed in chrome://tracing/ for aggregating the targets for a package [Python code to combine time trace files](https://www.snsystems.com/technology/tech-blog/clang-time-trace-feature)

#### Running ClangBuildAnalyzer for MoveIt to auto-analyzing the compile-time

```
**** Time summary:
Compilation (324 times):
  Parsing (frontend):         1843.2 s
  Codegen & opts (backend):    542.5 s

**** Files that took longest to parse (compiler frontend):
 16403 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_joints_widget.cpp.o
 14555 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_objects.cpp.o
 14401 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame.cpp.o
 13812 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_display.cpp.o
 13802 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_planning.cpp.o
 13709 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_states.cpp.o
 13592 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_scenes.cpp.o
 13248 ms: moveit_jog_arm/CMakeFiles/jog_arm_cpp_api.dir/src/jog_calcs.cpp.o
 13178 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_context.cpp.o
 13116 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_manipulation.cpp.o

**** Files that took longest to codegen (compiler backend):
 13727 ms: moveit_ros_planning_interface/move_group_interface/CMakeFiles/moveit_move_group_interface.dir/src/move_group_interface.cpp.o
 11268 ms: moveit_ros_warehouse/warehouse/CMakeFiles/moveit_warehouse.dir/src/planning_scene_storage.cpp.o
 10317 ms: moveit_ros_benchmarks/CMakeFiles/moveit_ros_benchmarks.dir/src/BenchmarkExecutor.cpp.o
 10070 ms: moveit_planners_ompl/ompl_interface/CMakeFiles/moveit_ompl_interface.dir/src/planning_context_manager.cpp.o
  9283 ms: moveit_kinematics/test/CMakeFiles/test_kinematics_plugin.dir/test_kinematics_plugin.cpp.o
  9062 ms: moveit_core/kinematics_metrics/CMakeFiles/moveit_kinematics_metrics.dir/src/kinematics_metrics.cpp.o
  8431 ms: moveit_ros_planning/trajectory_execution_manager/CMakeFiles/moveit_trajectory_execution_manager.dir/src/trajectory_execution_manager.cpp.o
  8057 ms: moveit_jog_arm/CMakeFiles/jog_arm_cpp_api.dir/src/jog_calcs.cpp.o
  7997 ms: moveit_jog_arm/CMakeFiles/jog_server.dir/src/jog_calcs.cpp.o
  7814 ms: moveit_ros_planning/planning_pipeline/CMakeFiles/moveit_planning_pipeline.dir/src/planning_pipeline.cpp.o

**** Templates that took longest to instantiate:
105532 ms: Eigen::Transform<double, 3, 1, 0>::rotation (220 times, avg 479 ms)
104054 ms: Eigen::Transform<double, 3, 1, 0>::computeRotationScaling<Eigen::Mat... (220 times, avg 472 ms)
 58688 ms: Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2>::JacobiSVD (220 times, avg 266 ms)
 58519 ms: Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2>::compute (220 times, avg 265 ms)
 56129 ms: Eigen::Transform<double, 3, 1, 0>::operator* (215 times, avg 261 ms)
 55750 ms: Eigen::internal::transform_transform_product_impl<Eigen::Transform<d... (215 times, avg 259 ms)
 18551 ms: Eigen::Block<Eigen::Matrix<double, 4, 4, 0, 4, 4>, 3, 3, false>::ope... (216 times, avg 85 ms)
 18512 ms: Eigen::BlockImpl<Eigen::Matrix<double, 4, 4, 0, 4, 4>, 3, 3, false, ... (216 times, avg 85 ms)
 18486 ms: Eigen::internal::BlockImpl_dense<Eigen::Matrix<double, 4, 4, 0, 4, 4... (216 times, avg 85 ms)
 18461 ms: Eigen::MatrixBase<Eigen::Block<Eigen::Matrix<double, 4, 4, 0, 4, 4>,... (216 times, avg 85 ms)
 18424 ms: Eigen::internal::call_assignment<Eigen::Block<Eigen::Matrix<double, ... (216 times, avg 85 ms)
 18361 ms: Eigen::internal::call_assignment<Eigen::Block<Eigen::Matrix<double, ... (216 times, avg 85 ms)
 18036 ms: Eigen::MatrixBase<Eigen::Product<Eigen::Matrix<double, 3, 3, 0, 3, 3... (220 times, avg 81 ms)
 17677 ms: Eigen::Matrix<double, 3, 3, 0, 3, 3>::Matrix<Eigen::Product<Eigen::M... (220 times, avg 80 ms)
 17644 ms: Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >::Plain... (220 times, avg 80 ms)
 17500 ms: Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >::_set_... (220 times, avg 79 ms)
 17472 ms: Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, 3, 3... (220 times, avg 79 ms)
 17322 ms: Eigen::internal::Assignment<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Ei... (220 times, avg 78 ms)
 17248 ms: Eigen::internal::generic_product_impl<Eigen::Matrix<double, 3, 3, 0,... (220 times, avg 78 ms)
 16843 ms: Eigen::Matrix<double, 3, 3, 0, 3, 3>::Matrix<Eigen::Product<Eigen::B... (216 times, avg 77 ms)
 16517 ms: Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >::_init... (216 times, avg 76 ms)
 16502 ms: Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >::_set_... (216 times, avg 76 ms)
 16473 ms: Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, 3, 3... (216 times, avg 76 ms)
 16336 ms: Eigen::internal::Assignment<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Ei... (216 times, avg 75 ms)
 16247 ms: Eigen::internal::generic_product_impl<Eigen::Block<const Eigen::Matr... (216 times, avg 75 ms)
 14838 ms: Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, 3, 3... (220 times, avg 67 ms)
[... 171 lines cut ...]
  execute_trajectory_action_capability.cpp.o execute_trajectory_action_capability.h move_group_capability.h planning_scene_monitor.h planning_scene.h robot_model.h joint_model_group.h joint_model.h  (964 ms)
  ...

148347 ms: /home/jafar/workspaces/ros/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model.h (included 225 times, avg 659 ms), included via:
  planar_joint_model.cpp.o planar_joint_model.h  (2069 ms)
  fixed_joint_model.cpp.o fixed_joint_model.h  (2038 ms)
  prismatic_joint_model.cpp.o prismatic_joint_model.h  (2006 ms)
  floating_joint_model.cpp.o floating_joint_model.h  (1984 ms)
  joint_model.cpp.o  (1877 ms)
  revolute_joint_model.cpp.o revolute_joint_model.h  (1842 ms)
  ...

119413 ms: /usr/include/boost/function.hpp (included 2856 times, avg 41 ms), included via:
  kinematics_plugin_loader.cpp.o kinematics_plugin_loader.h  (1168 ms)
  pick_place_params.cpp.o server.h  (1120 ms)
  trajectory_visualization.cpp.o string.hpp split.hpp iter_find.hpp find_iterator.hpp find_iterator.hpp  (747 ms)
  world_diff.cpp.o world_diff.h world.h  (732 ms)
  collision_matrix.cpp.o collision_matrix.h collision_common.h  (728 ms)
  world.cpp.o world.h  (723 ms)
  ...
```

#### System info:
- Ubuntu 18.04 / Melodic
- Intel® Core™ i7-8750H CPU @ 2.20GHz × 12

v4hn, 2020-05-11:
Thank you for this experiment @JafarAbdi ! It's quite cool to see the breakup of the never-ending build time.

I did not even attempt to reproduce these numbers yet, but there are a number of interesting insights in there. I am aware that all numbers here should be taken with a grain of salt, but it is still quite obvious that CI times (and also local rebuilds) tend to take quite long these days...

> 16403 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_joints_widget.cpp.o

@rhaschke Any idea why your new frame takes **that** long to parse?
I mean, sure, qt objects take forever to parse, I'm just surprised to see this one at the top.

>  13727 ms: moveit_ros_planning_interface/move_group_interface/CMakeFiles/moveit_move_group_interface.dir/src/move_group_interface.cpp.o

I have a hard time understanding why this might take so damn long...
This is mostly a bunch of simple ROS interfaces... Looking at ~30 lines of includes to major headers could explain quite a bit though...

> 105532 ms: Eigen::Transform<double, 3, 1, 0>::rotation (220 times, avg 479 ms)
> 104054 ms: Eigen::Transform<double, 3, 1, 0>::computeRotationScaling<Eigen::Mat... (220 times, avg 472 ms)
> 58688 ms: [...]

These should disappear after #1964. 

> 409502 ms: /home/jafar/workspaces/ros/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/robot_model.h (included 217 times, avg 1887 ms)

sounds like a bunch of forward declarations could actually reduce our build-time by quite a lot...
E.g., `srdf::Model` is only used via references and in `shared_ptr` objects.

henningkayser, 2020-05-13:
@JafarAbdi thanks for preparing this! I also think that there are some really interesting and unexpected results in this log.

> sounds like a bunch of forward declarations could actually reduce our build-time by quite a lot...
E.g., srdf::Model is only used via references and in shared_ptr objects.

That plus removing all unused includes from headers sounds like a great issue for WMD.
I'm also wondering if we could do something to improve compile time for all visualization plugins.

JafarAbdi, 2020-07-13:
I re-analyzed the compile-time again today

```
**** Time summary:
Compilation (343 times):
  Parsing (frontend):         2407.3 s
  Codegen & opts (backend):    732.1 s

**** Files that took longest to parse (compiler frontend):
 19042 ms: moveit_tutorials/doc/perception_pipeline/CMakeFiles/cylinder_segment.dir/src/cylinder_segment.cpp.o
 18892 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_joints_widget.cpp.o
 17793 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame.cpp.o
 17671 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_objects.cpp.o
 17628 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_display.cpp.o
 17581 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_scenes.cpp.o
 17331 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_planning.cpp.o
 17327 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_manipulation.cpp.o
 17101 ms: moveit_ros_visualization/motion_planning_rviz_plugin/CMakeFiles/moveit_motion_planning_rviz_plugin_core.dir/src/motion_planning_frame_states.cpp.o
 16746 ms: moveit_tutorials/doc/subframes/CMakeFiles/subframes_tutorial.dir/src/subframes_tutorial.cpp.o

**** Files that took longest to codegen (compiler backend):
 14182 ms: moveit_ros_planning_interface/move_group_interface/CMakeFiles/moveit_move_group_interface.dir/src/move_group_interface.cpp.o
 12420 ms: moveit_ros_warehouse/warehouse/CMakeFiles/moveit_warehouse.dir/src/planning_scene_storage.cpp.o
 11945 ms: moveit_tutorials/doc/subframes/CMakeFiles/subframes_tutorial.dir/src/subframes_tutorial.cpp.o
 11547 ms: moveit_planners_ompl/ompl_interface/CMakeFiles/moveit_ompl_interface.dir/src/planning_context_manager.cpp.o
 10535 ms: moveit_visual_tools/CMakeFiles/moveit_visual_tools.dir/src/moveit_visual_tools.cpp.o
 10514 ms: moveit_core/kinematics_metrics/CMakeFiles/moveit_kinematics_metrics.dir/src/kinematics_metrics.cpp.o
 10002 ms: moveit_ros_planning/planning_pipeline/CMakeFiles/moveit_planning_pipeline.dir/src/planning_pipeline.cpp.o
  9561 ms: moveit_kinematics/kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/src/chainiksolver_vel_mimic_svd.cpp.o
  9083 ms: moveit_jog_arm/CMakeFiles/jog_arm_cpp_api.dir/src/jog_calcs.cpp.o
  8800 ms: moveit_jog_arm/CMakeFiles/jog_server.dir/src/jog_calcs.cpp.o

**** Templates that took longest to instantiate:
134207 ms: Eigen::Transform<double, 3, 1, 0>::computeRotationScaling<Eigen::Mat... (253 times, avg 530 ms)
 76218 ms: Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2>::JacobiSVD (253 times, avg 301 ms)
 75969 ms: Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2>::compute (253 times, avg 300 ms)
 62062 ms: Eigen::Transform<double, 3, 1, 0>::operator* (236 times, avg 262 ms)
 61448 ms: Eigen::internal::transform_transform_product_impl<Eigen::Transform<d... (234 times, avg 262 ms)
 24369 ms: Eigen::MatrixBase<Eigen::Product<Eigen::Matrix<double, 3, 3, 0, 3, 3... (253 times, avg 96 ms)
 23889 ms: Eigen::Matrix<double, 3, 3, 0, 3, 3>::Matrix<Eigen::Product<Eigen::M... (253 times, avg 94 ms)
 23845 ms: Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >::Plain... (253 times, avg 94 ms)
 23650 ms: Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >::_set_... (253 times, avg 93 ms)
 23609 ms: Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, 3, 3... (253 times, avg 93 ms)
 23405 ms: Eigen::internal::Assignment<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Ei... (253 times, avg 92 ms)
 23305 ms: Eigen::internal::generic_product_impl<Eigen::Matrix<double, 3, 3, 0,... (253 times, avg 92 ms)
 22506 ms: Eigen::Matrix<double, 3, 1, 0, 3, 1>::operator=<Eigen::Product<Eigen... (227 times, avg 99 ms)
 22484 ms: Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::_set<... (227 times, avg 99 ms)
 22438 ms: Eigen::internal::call_assignment<Eigen::Matrix<double, 3, 1, 0, 3, 1... (227 times, avg 98 ms)
 22348 ms: Eigen::internal::call_assignment<Eigen::Matrix<double, 3, 1, 0, 3, 1... (227 times, avg 98 ms)
 22334 ms: Eigen::Matrix<double, 3, 1, 0, 3, 1>::Matrix<Eigen::Product<Eigen::B... (228 times, avg 97 ms)
 21471 ms: Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::_init... (228 times, avg 94 ms)
 21450 ms: Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::_set_... (228 times, avg 94 ms)
 21412 ms: Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, 3, 1... (228 times, avg 93 ms)
 21244 ms: Eigen::internal::Assignment<Eigen::Matrix<double, 3, 1, 0, 3, 1>, Ei... (228 times, avg 93 ms)
 21129 ms: Eigen::internal::generic_product_impl<Eigen::Block<const Eigen::Matr... (228 times, avg 92 ms)
 20081 ms: Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, 3, 3... (253 times, avg 79 ms)
 19957 ms: Eigen::DenseBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >::isApprox<Ei... (248 times, avg 80 ms)
[... cut ...]
  controller_edit_widget.cpp.o controller_edit_widget.h moveit_config_data.h planning_scene.h robot_model.h model.h ros.h common.h forwards.h function.hpp  (360 ms)
  initialize_demo_db.cpp.o planning_scene_storage.h moveit_message_storage.h database_connection.h message_collection.h query_results.h message_with_metadata.h metadata.h ros.h common.h forwards.h function.hpp  (358 ms)
  trajectory_constraints_storage.cpp.o trajectory_constraints_storage.h moveit_message_storage.h database_connection.h message_collection.h query_results.h message_with_metadata.h metadata.h ros.h common.h forwards.h function.hpp  (354 ms)
  ...
```