moveit/moveit #2084: Timeout when executing plan

guiolpei, 2020-05-13:
**Software and hardware setup**

Ubuntu 16.04 with ROS Kinetic.
UR5e robot arm connected using External Contol URCap.

**Code**

```
	int waitingTime = 300;
	std::cout << "Waiting time set to " << waitingTime << "\n";
	const boost::shared_ptr<tf::Transformer> tf_buffer = boost::shared_ptr<tf::Transformer>();
	ros::WallDuration wallDuration(waitingTime, 0);
	moveit::planning_interface::MoveGroupInterface robotMoveGroup(ROBOT_PLANNING_GROUP, tf_buffer, wallDuration);

	// = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = 
	std::map<std::string, double> initialJoints;
	bool initialSuccessPlan = false;
	moveit::planning_interface::MoveGroupInterface::Plan initialPlan;

	// Robot
	initialJoints = robotMoveGroup.getNamedTargetValues(INITIAL_NAMED_TARGET);

	robotMoveGroup.setJointValueTarget(initialJoints);
	initialSuccessPlan = static_cast<bool>(robotMoveGroup.plan(initialPlan));

	if(!initialSuccessPlan){
		std::cout << "[ERROR] Move robot planning to initial position failed!\n";
		return;
	}

	moveit::planning_interface::MoveItErrorCode robotInitial = robotMoveGroup.execute(initialPlan);

	switch (robotInitial.val)
	{
		case moveit::planning_interface::MoveItErrorCode::SUCCESS:
			std::cout << "Robot in initial position!\n";
			break;
		case moveit::planning_interface::MoveItErrorCode::TIMED_OUT:
			std::cout << "[ERROR] Timeout when moving robot to initial position!\n";
			break;			
		default:
			std::cout << "[ERROR] Robot initial position motion failed\n";
	}
```

**Issue**

Timeout when using execute() function with reduced UR5e speed. If speed is not set to maximum, it will timeout during execution. As shown in code, I have been trying to modify the action waiting time during the MoveGroupInterface instance creation but it does not seem to take it into consideration.

gavanderhoorn, 2020-05-13:
> As shown in code, I have been trying to modify the action waiting time during the MoveGroupInterface instance creation but it does not seem to take it into consideration.

That parameter only influences how long the class will wait for the action servers to come up. It does not influence the allowed trajectory execution time deviation.

The MoveIt docs/tutorials include some information on configuring that: [Low Level Controllers: Optional Allowed Trajectory Execution Duration Parameters](https://ros-planning.github.io/moveit_tutorials/doc/controller_configuration/controller_configuration_tutorial.html?highlight=execution#optional-allowed-trajectory-execution-duration-parameters).

I'm not aware of a way to update this at runtime though (as compared to initialisation).

Perhaps some/one of the other maintainers do.

guiolpei, 2020-05-13:
Thank you for your answer. I misunderstood the meaning of that parameter, it is clear now.

I was not aware that it could be modified from the controller definition file, this is a valid solution in this case.

Using these duration parameters it could be enough:

>    allowed_execution_duration_scaling: 1.2
>    allowed_goal_duration_margin: 0.5

The first parameter scales the expected execution duration, but what are the units of the second one?

v4hn, 2020-05-13:
On Wed, May 13, 2020 at 01:57:10AM -0700, Guillermo Oliver wrote:
> >    allowed_execution_duration_scaling: 1.2
> >    allowed_goal_duration_margin: 0.5
> 
> The first parameter scales the expected execution duration, but what are the units of the second one?

seconds. You can also disable them altogether (setting them to 0.0?) if you do not care about execution monitoring.

guiolpei, 2020-05-13:
Thank you very much, I guess this issue can be closed now.

closed 2020-05-13 by guiolpei

gavanderhoorn, 2020-05-13:
@gavanderhoorn wrote:
> I'm not aware of a way to update this at runtime though (as compared to initialisation).

just remembered there is a [dynamic_reconfigure](http://wiki.ros.org/dynamic_reconfigure) interface to those parameters, so you could use a `dynamic_reconfigure` client and update the scaling values based on the value of the speed scaling of the UR driver (which is published on a topic).

v4hn, 2020-05-13:
> so you could use a `dynamic_reconfigure` client and update the scaling values based on the value of the speed scaling of the UR driver (which is published on a topic).

Also sounds like a nice example for a `MoveGroupCapability` someone might be interested in.

gavanderhoorn, 2020-05-13:
You mean a capability which would automagically update the scaling based on what the topic publishes?

That would be nice indeed.

Sounds like a good candidate for WMD.