moveit/moveit #2088: Place interactive markers at end effector tool points

felixvd, 2020-05-14:
There is a recurring question on ROS Answers that I have trouble answering, about the position of the interactive marker on the robot. Users want to place the marker at a TCP at a tool point, like [this](https://answers.ros.org/question/350967/how-do-i-relocate-the-interactive-marker-to-the-tip-of-the-needle/).

As seen [here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/robot_interaction/src/robot_interaction.cpp#L278), currently the only interactive marker is placed at the parent link of the end effector group, which is safe, but almost never where one would really want it. (There is a fall-back behavior that places the marker at the end of the chain if there is no end-effector, but since it is required for planning with MTC, this is not generally useful as a workaround.)

@mikeferguson suggested adding a `tool_point` tag to the [SRDF](https://github.com/ros-planning/srdfdom) in [this old review](http://wiki.ros.org/srdf/review). Can anyone think of a better way to implement this? It seems like the cleanest option, but changing the format may be a little heavy, and I'm not sure what would be involved.

rhaschke, 2020-05-14:
There is a solution to this: Just define an additional, virtual link in your URDF and define **two** `end_effectors` associated with the _same_ `parent_group` in SRDF:

```
<end_effector name="tool_mount" parent_link="tool_mount" group="hand" parent_group="arm"/>
<end_effector name="tool_tip" parent_link="tool_tip" group="hand" parent_group="arm"/>
```
This will yield two interactive markers (when you select the common `parent_group`) and you can drag the arm with each of them.
As you can see from the image, both markers can have different orientations as well, not only a different tool point.
[image]

rhaschke, 2020-05-14:
I think we can close this here. @felixvd, you already gave a good reply at ROS answers.

closed 2020-05-14 by rhaschke