moveit/moveit #2090: Build error undefined reference to boost

ShrutheeshIR, 2020-05-15:
### Description

I am following the getting started page, and while building the catkin workspace, I am getting a build error that is related to Boost.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build? : Source
* If binary, which release version?
* If source, which branch? Master branch

### Steps to reproduce
I am merely following the steps given in the [Getting Started Page](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html). ( I did not do this step `git clone https://github.com/ros-planning/moveit_tutorials.git -b master` since `moveit_tutorials` was already present after I executed the `wstool` command. 

When I run `catkin build`, I get the following error.

### Expected behaviour
It should be built without error.

### Actual behaviour
/usr/bin/ld: warning: libboost_filesystem.so.1.65.1, needed by /home/olorin/Desktop/moveittuts/ws_moveit/devel/.private/moveit_core/lib/libmoveit_test_utils.so.1.0.1, may conflict with libboost_filesystem.so.1.72.0
/home/olorin/Desktop/moveittuts/ws_moveit/devel/.private/moveit_core/lib/libmoveit_test_utils.so.1.0.1: undefined reference to `boost::re_detail_107200::perl_matcher<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<boost::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::construct_init(boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::_match_flags)'
/home/olorin/Desktop/moveittuts/ws_moveit/devel/.private/moveit_core/lib/libmoveit_test_utils.so.1.0.1: undefined reference to `boost::re_detail_107200::raise_runtime_error(std::runtime_error const&)'
/home/olorin/Desktop/moveittuts/ws_moveit/devel/.private/moveit_core/lib/libmoveit_test_utils.so.1.0.1: undefined reference to `boost::re_detail_107200::get_default_error_string(boost::regex_constants::error_type)'
/home/olorin/Desktop/moveittuts/ws_moveit/devel/.private/moveit_core/lib/libmoveit_test_utils.so.1.0.1: undefined reference to `boost::re_detail_107200::verify_options(unsigned int, boost::regex_constants::_match_flags)'
/home/olorin/Desktop/moveittuts/ws_moveit/devel/.private/moveit_core/lib/libmoveit_test_utils.so.1.0.1: undefined reference to `boost::re_detail_107200::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const'
/home/olorin/Desktop/moveittuts/ws_moveit/devel/.private/moveit_core/lib/libmoveit_test_utils.so.1.0.1: undefined reference to `boost::re_detail_107200::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'
/home/olorin/Desktop/moveittuts/ws_moveit/devel/.private/moveit_core/lib/libmoveit_test_utils.so.1.0.1: undefined reference to `boost::re_detail_107200::get_mem_block()'
/home/olorin/Desktop/moveittuts/ws_moveit/devel/.private/moveit_core/lib/libmoveit_test_utils.so.1.0.1: undefined reference to `boost::re_detail_107200::put_mem_block(void*)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/olorin/Desktop/moveittuts/ws_moveit/devel/.private/moveit_core/lib/moveit_core/robot_state_benchmark] Error 1
make[1]: *** [robot_state/CMakeFiles/robot_state_benchmark.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2

rhaschke, 2020-05-15:
From the logs, one can see that you have two versions of boost installed, which are conflicting:
* boost 1.65.1 is the one shipped with Ubuntu 18.04
* but you have boost 1.72 installed as well
Please fix your local installation. Check, where the two boost versions are installed (the system one in /usr/include/boost and /usr/lib/x86_64-linux-gnu/libboost_*) and which versions are found by cmake.

closed 2020-05-15 by rhaschke

ShrutheeshIR, 2020-05-15:
Thank you :)
Could you provide a small guide on how I can solve this.
Can I remove my entire installation of boost 1.65 and leave only boost 1.72 on. Or does the package require boost 1.65 only?
I know this is beyond the scope of the project, so apologies!
Thanks

rhaschke, 2020-05-15:
I already asked specific questions to be able to provide more help.
I think it won't be possible to remove the system boost installation as many other system packages, particularly within ROS depend on it. I'm not sure, why you need boost 1.72. In the context of ROS it probably doesn't make sense to use, because then you essentially need to rebuild most/all ROS packages from source. So you should try to enforce your MoveIt workspace to be built with system boost.

gavanderhoorn, 2020-05-15:
Looks like this is also asked here: [Uninstalling a package removes all of ROS](https://answers.ros.org/question/352413).