moveit/moveit #2092: GSoC: Motion planning with general end-effector constraints in MoveIt

JeroenDM, 2020-05-17:
**mentors**: @mamoll @ommmid @felixvd 

This summer I would like to add collision-free motion planning with general end-effector constraints to MoveIt. Application examples include robot arc welding, grinding, and painting. The original discussion on discourse can be found [here](https://discourse.ros.org/t/gsoc-2020-cartesian-planning-some-trajopt/13045). I probed for interest in other issues [here](https://github.com/ros-planning/moveit/issues/1969) and [here](https://github.com/ros-planning/moveit/issues/467). The goal of this issue is to log and discuss my progress in the coming months.

## Initial project description
Many industrial robot applications specify a Cartesian end-effector path, that could be under-constrained. The MoveIt’s RobotState function, “computeCartesianPath”, fulfills only a small part of this demand. OMPL has advanced capabilities to handle constraints that are not integrated with MoveIt yet. Examples of applications that can benefit from this integration are robot welding, grinding, painting, and many more.

Concretely this would mean integrating OMPL’s Cartesian planning capabilities in the existing planning plugin. If time allows it, I would like to add support for Cartesian path constraints in TrajOpt. All deliverables should include documentation, tutorials, and tests.

## Planning
**Phase 1: Example cases and interface discussion (June)**
- Find/create example cases that use external planners to solve the planning problem and showcase what will be possible with MoveIt at the end of the project. An example could be the puzzle piece demo from [trajot_ros](https://github.com/ros-industrial-consortium/trajopt_ros) or a welding example such as the last one in [hybrid_planning_examples](https://github.com/Jmeyer1292/hybrid_planning_experiments).
- The current interface to specify path constraints in Moveit is not entirely clear, I suspect it will take some time to find a common interface to specify these planning problems.
- As a learning experience, I could refactor the current “computeCartesianPath” as a MoveIt planning plugin, as discussed [here](https://github.com/ros-planning/moveit/issues/797).

**Phase 2: Implementation (July)**
- Start implementing the features that we agreed upon in the first month. This will most likely mean working in the code of the current OMPL interface. I don’t know the exact approach yet, as explained below in the comments from the kick-off meeting.
- @v4hn also pointed out that [the broken pose_model_state_space](https://github.com/ros-planning/moveit/tree/master/moveit_planners/ompl/ompl_interface/src/parameterization/work_space) should be replaced. (I did not look into the details of this yet.)

**Phase 3: Documentation and TrajOpt Cartesian constraints (August)**
- Finish the implementation and tests.
- I would really like to spend some time writing a good tutorial on how to use the new features. 
- Add the missing cost terms for the Cartesian constraint in TrajOpt, as discussed [here](https://github.com/ros-planning/moveit/issues/1467).

In addition, it would be nice to have some benchmarks comparing, for example, OMPL with TrajOpt.

## Interface
A large unresolved point is how to model the constraints and describe them in MoveIt. Planners such as [Descartes](https://github.com/ros-industrial-consortium/descartes) use discrete waypoints along an end-effector path. There are several options to integrate this type of constraints:
- [moveit_msgs/MotionPlanRequest.msg](http://docs.ros.org/melodic/api/moveit_msgs/html/msg/MotionPlanRequest.html) the `trajectory_constraints` field, possible renamed to waypoints as discussed [here](https://github.com/ros-planning/moveit/issues/804).
- [moveit_msgs/MotionSequenceRequest](http://docs.ros.org/melodic/api/moveit_msgs/html/msg/MotionSequenceRequest.html).
- [MoveGroupCartesianPathService](https://github.com/ros-planning/moveit/blob/master/moveit_ros/move_group/src/default_capabilities/cartesian_path_service_capability.h) as done for Descartes [here](https://github.com/PickNikRobotics/descartes_capability).

On the other hand, the [constrained planners](https://ompl.kavrakilab.org/constrainedPlanning.html) in OMPL require a generic constraint function F(q) = 0 (and ideally the Jacobian). For simple cases, this could be put into the `path_constraints` of a motion planning request. For more general cases I have no (good) idea yet of how to do this. Possible, the concept of [Task Space Regions](https://dl.acm.org/doi/10.1177/0278364910396389) can be helpful here.

JeroenDM, 2020-05-17:
### Comments during the kick-off meeting
@mamoll mentioned an alternative to full integration of OMPL’s ConstraintStateSpace is to implement something similar in MoveIt by extending the [JointModelStateSpace](https://github.com/ros-planning/moveit/blob/master/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/parameterization/joint_space/joint_model_state_space.h) and using the existing interface to plan within this state space.

This would mean reimplementing the techniques to handle Constraints currently available in OMPL, ProjectedStateSpace, AtlasStateSpace, and TangentBundleStateSpace. Only the first one seems feasible to me within the given timeline. An advantage of reimplementing is that it would be easy to add explicit sampling of the constraints as done in Descartes.

@ommmid explained that supporting Cartesian path constraints in TrajOpt should not be too difficult. But currently, the collision cost in TrajOpt is not finished yet. This would be beyond the scope of my project.

v4hn, 2020-05-17:
> by extending the JointModelStateSpace and using the existing interface to plan within this state space.

@mamoll sure knows what he's talking about and has his reasons to propose this.
Having spent several weeks of developement time understanding internals of ModelBasedPlanningContext and its derivatives in past years,
I would greatly appreciate *not* adding more complexity there if it can be avoided.
The `PoseModelStateSpace` was an attempt for eef planning implemented in such a wrapper and it's a horrible broken mess...

felixvd, 2020-05-18:
So, is that an argument against the proposed implementation...? I'm not familiar with that part of the code.

v4hn, 2020-05-18:
> So, is that an argument against the proposed implementation...? I'm not familiar with that part of the code.

Rewriting the ompl plugin ModelBasedStateSpace entirely would definitely be an option,
but I think it does not make sense to build on that code even more structure.

Nor am I familiar with the OMPL codebase for handling manifolds. Maybe it's even worse. :D
Otherwise it would probably be cleaner *and* include more functionality to wrap OMPLs support here.

JeroenDM, 2020-05-24:
>> by extending the JointModelStateSpace and using the existing interface to plan within this state space.

> @mamoll sure knows what he's talking about and has his reasons to propose this.

I think it was more mentioned as an alternative approach because it had been done like this in the past in a closed source project. I don't think we discussed in great detail why this would be a better approach.

> Otherwise it would probably be cleaner *and* include more functionality to wrap OMPLs support here.

I certainly agree that it would be better if we could reuse as much code as possible that already exists in OMPL.

JeroenDM, 2020-05-24:
I did not look into the details of the MoveIt code yet but went through the [constrained planning](http://ompl.kavrakilab.org/constrainedPlanning.html) documentation on ompl's website. In this process I revamped some old code I had lying around that implements an SE3 state space that inherits from [RealVectorStateSpace](http://ompl.kavrakilab.org/classompl_1_1base_1_1RealVectorStateSpace.html) and therefore can be used by the constrained planning algorithms. Although I do not expect to do something along these lines here since I will be doing planning for robot arms parameterized in joint space.

I'm also happy to see `Determinism` and `Fully featured Cartesian Planner` on the MoveIt [roadmap](https://moveit.ros.org/documentation/contributing/roadmap/). Deterministic state samplers where [added](https://github.com/ompl/ompl/pull/695) to ompl recently.

Unfortunately, I had quite a lot of other work this week that prevented me from a more elaborate preparation of my GSoC project.

JeroenDM, 2020-06-01:
I went through all of the [ompl interface code](https://github.com/ros-planning/moveit/tree/master/moveit_planners/ompl) and created a diagram to get a general idea of what's in it: [moveit_ompl_interface_diagram.pdf](https://github.com/ros-planning/moveit/files/4710545/moveit_ompl_interface_diagram.pdf).

Some general questions I asked myself when reading the code:
- Why are two classes (`OMPLPlannerManager` and `ConstraintApproximationStateSampler`) defined in a source file instead of the header file? (@ommmid maybe you know this, as it is also the case in the TrajOpt plugin?)
- The sbpl planner inherits from `planning_interface::Planner` instead of `planning_interface::PlannerManager`. Is this an artifact from an older version of the planning interface?
- Why not inherit from ompl's `RealVectorStateSpace` instead of `StateSpace`? This would allow for some reuse of sampling methods implemented in OMPL.
- `JointModelStateSpace` does not do much compared to it's parent class `ModelBasedStateSpace`. It is probably there to differentiate it from `PoseModelStateSpace`, which does specialize.
- It will be helpful if I figure out in more detail which parts of the code are actually used at the moment.

I'll dive into it on World MoveIt day tomorrow to find answers and even better questions.

ommmid, 2020-06-01:
In the same context, I had looked at the OMPL inteface in moveit too and [here](https://github.com/ros-planning/moveit/files/3230071/moveit_ompl_plugin.pdf) is the graph of it just as a reference and giving more information.

About why OMPLPLannerManager does not have any header:
CHOMPPlannerManager class also is declared and defined in the source file. OMPL interface did the same thing. I just followed the structure for trajopt as well. I am not sure why that is but maybe because it does not need to be shared with any other files. Thus, it does not need to have a header file, making it easier for the compiler?

My understanding is:
OMPL/CHOMP/TrajOptPlannerManager are mostly only used to override the virtual functions and that is it, no new method is created in these derived classs. So why bother to create any header file for them? they are not going to be shared with anything else. Including the base calss which is PlannerManager (from planning_interface.h) should be enough.
while if you look at ModelBasedPlanningContext, it has bunch of other methods that are new, specific to this derived class and not overriding anything. So we need to share its header if we want to used this new methods.

JeroenDM, 2020-06-01:
@ommmid Thanks for the graph. It's helpful, as was the tutorial you wrote on MoveIt plugins. I went through it today and proposed some minor improvements in a [pull request](https://github.com/ros-planning/moveit_tutorials/pull/492).

> no new method is created in these derived classs. So why bother to create any header file for them?

That makes sense. I was used to reading header files as documentation, hence my confusion :)

My plan of action now is to start with `refactor the current “computeCartesianPath” as a MoveIt planning plugin` as mentioned in the top post. This should be a good opportunity to propose an interface and learn the details of planning plugins.

JeroenDM, 2020-06-02:
I uploaded an example of the type of problem I would like to start with [here](https://www.youtube.com/watch?v=Ps1eyC-8-Qg&feature=youtu.be).
(I will first have to clean up the code I used to generate this video, as this is currently part of my bulky attempt at benchmarking complex planning problems in [this repository](https://github.com/JeroenDM/benchmark_ros_ws)).

I will first implement an example that does Cartesian planning using the `reference_trajectories` field in a `MotionPlanRequest`, but does not yet use OMPL for Cartesian planning.

As far as I know there is currently no planner that supports a Cartesian trajectory in `reference_trajectories`. There is an example for a joint trajectory in the [TrajOpt example code](https://github.com/ros-planning/moveit_tutorials/blob/9acbbe0c88f2967d97296f1bec9707632cca4f7e/doc/trajopt_planner/src/trajopt_example.cpp#L159). A would use some very basic proof of concept, or maybe Descartes, for this first version.

ommmid, 2020-06-02:
> I uploaded an example of the type of problem I would like to start with [here](https://www.youtube.com/watch?v=Ps1eyC-8-Qg&feature=youtu.be).

Looks like a good problem for Cartesian planning.

> As far as I know there is currently no planner that supports a Cartesian trajectory in `reference_trajectories`. There is an example for a joint trajectory in the [TrajOpt example [...]

`reference_trajectories` was added quite recently for optimization based planners. So yes, other planners have not used this. In TrajOpt it is used as an initial guess trajectory.

mamoll, 2020-06-02:
@JeroenDM I like the welding problem in the YouTube video you linked to. If it is easy to test out whether TrajOpt can solve this problem (while ignoring collisions for now), then give that a try.

For OMPL it's not that easy to incorporate a reference trajectory. The internal data structure that each planner in OMPL is different, so they each need to do something different with the additional info provided by a reference trajectory. One approach that I have seen used is to treat an initial trajectory as a sequence of _n_ states. This is fed to a special sampler that first generates the _n_ states and then falls back on the default random sampling. This might help speed up OMPL, but I would not worry about this initially. Just getting constrained _global_ planning working without this bootstrapping optimization is hard enough.

> `reference_trajectories` was added quite recently for optimization based planners. So yes, other planners have not used this. In TrajOpt it is used as an initial guess trajectory.

@ommmid wasn't the idea that another planning pipeline could seed a trajectory optimization method with a _feasible_ initial guess?

ommmid, 2020-06-02:
> @ommmid wasn't the idea that another planning pipeline could seed a trajectory optimization method with a _feasible_ initial guess?

I am not aware of that but it seems a better idea for feasible initial guess.

JeroenDM, 2020-06-02:
@mamoll
> If it is easy to test out whether TrajOpt can solve this problem

As soon as the TrajOpt plugin integrates collision checking it should be easy to do, (by adding the Cartesian cost terms). I could also use the tesseract version of TrajOpt to do in now of course.

> For OMPL it's not that easy to incorporate a reference trajectory.

That's an intersting idea I did not think of (oddly enough). I planned to use the reference trajectory as nominal values to center the constraints around. It is not trivial to convert these constraints to global constraints (F(x) = 0) for OMPL, but I had some ideas.

Anyway, I will continue with my prove of concept tomorrow  and provide a more detailed proposal.

JeroenDM, 2020-06-03:
Given the welding example posted earlier, I decided to leave out the free space motion in between the welds. This allows me to focus on the constrained planning part.
When constrained planning works properly, the MoveIt Task Constructor can be used to combine the different sub problems.

As a starting point I tried solving the welding case using [descartes_capabilities](https://github.com/PickNikRobotics/descartes_capability), but I there are some still some issues, especially when adding the work object (not shown below).
I will clean up and push my code to a public repository tomorrow before I continue working on it.

[image]

JeroenDM, 2020-06-04:
I created a [repository](https://github.com/JeroenDM/moveit_constrained_planning_examples) with some examples. I experimented a bit with the different formats to specify the constrained planning problems and wrote down my conclusion in [doc/interface.md](https://github.com/JeroenDM/moveit_constrained_planning_examples/blob/master/doc/interface.md). The main difficulty, how to represent the constraints for ompl, is discussed in [the second section](https://github.com/JeroenDM/moveit_constrained_planning_examples/blob/master/doc/interface.md#discrete-points-vs-continuous-constrained-region).

Now I plan to dive into the details of how the ompl interface currently represents and samples the constraints and identify the specific parts I have to change/extend.

I did not create a planning plugin for the basic Cartesian planner, as alluded to earlier. I think I understand the basic plugin interface enough for now. And since the code is already [taken out](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/include/moveit/robot_state/cartesian_interpolator.h) of the  robot state class there is maybe not so much value in adding this.

JeroenDM, 2020-06-08:
I noticed I forgot to post my work of the end of last week here.

I started working on a minimal example using OMPL's [ProjectedStateSpace](https://ompl.kavrakilab.org/classompl_1_1base_1_1ProjectedStateSpace.html) to do constrained planning in [this repository](https://github.com/JeroenDM/moveit_constrained_planning_plugin).

(I keep a daily log of my work [here](https://docs.google.com/document/d/1PxItLhBOdmTKrARTYpf57xNYrKbn2MC0qXPK_ZEOpMc/edit?usp=sharing) as mentioned in the first comment.)

JeroenDM, 2020-06-08:
Finally, the visualization is working so I could make a video of the demo mentioned above :)
[https://youtu.be/KvgvV8T2zvs](https://youtu.be/KvgvV8T2zvs)

The x-position equality constraint uses the Jacobian from MoveIt, the bounds on the z-position not yet.

JeroenDM, 2020-06-12:
I wrote down what I'm trying out in code more formally, so it is easier to check if it makes sense.
[constraints_model.pdf](https://github.com/ros-planning/moveit/files/4769286/constraints_model.pdf)

I'm having trouble getting the Orientation constraints to work. I plan to create a simpler example with a 6 dof robot for debugging. (I was using the panda robot now.)

The implementation of TrajOpt by ROS-Industrial seemed to have [similar issues](https://github.com/ros-industrial-consortium/trajopt_ros/blob/77af966c7255b94199416c03fb7878110d5162fd/trajopt/src/kinematic_terms.cpp#L199). They used finite differences for a specific parts of the Jacobian to solve the issue, but for angle-axis error instead of roll pitch yaw angles.

[I compared](https://github.com/JeroenDM/moveit_constrained_planning_plugin/blob/d3a76f6343d1749cd948c56650438b2d1493d5ab/planning_plugin_examples/src/try_jacobian_projection.cpp#L31) the analytical Jacobian with a finite differences one for RPY angles, and it seems to work. But I'm not sure if I'm using it correctly when using it for the Jacobian of the constraints [here](https://github.com/JeroenDM/moveit_constrained_planning_plugin/blob/d3a76f6343d1749cd948c56650438b2d1493d5ab/constrained_planning_plugin/src/constraint.cpp#L97). Could be that I'm doing something wrong related to things being expressed in different reference frames.

JeroenDM, 2020-06-15:
### Meeting notes 15/06/2020
@felixvd @mamoll @ommmid 

**Orientation constraints**
In the previous comment, I illustrated an issue I encountered with orientation constraints. These constraints are specified using roll, pitch, and yaw angles, as implemented in MoveIt's [constraint samplers](https://github.com/ros-planning/moveit/blob/272eaa01064df0b0184e2ad8766751ea27ef6c92/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp#L614). (Also see [this new issue](https://github.com/ros-planning/moveit_msgs/issues/82).) This is intuitive for the user, but probably causes convergence problems for the constraint projection methods. Using unit quaternions should give better results, but it is more difficult for users to specify orientation constraints. I think I will continue working as follows:

1. Start with a fixed end-effector pose as orientation constraints. This can be specified and implemented using quaternions.
2. Try using angle-axis tolerance specification as used by [TrajOpt](https://github.com/ros-industrial-consortium/trajopt_ros/blob/a393abb99b221c97603ae40fcbd0f172f94a7fcc/trajopt/include/trajopt/utils.hpp#L161).

A more esoteric solution Mark proposed is to formulate constraints as a region on the 4D unit sphere (which is the space quaternions live in). A 3D Ellipsoid could represent something useful maybe? (Or a [Squircle](https://en.wikipedia.org/wiki/Squircle)?). Something in this direction already exists in OMPL in [SO3StateSampler::sampleUniformNear](https://github.com/ompl/ompl/blob/e27ebff31f3533c57b2be4f033ca022a7967fba7/src/ompl/base/spaces/src/SO3StateSpace.cpp#L104). (Written by @mamoll ?)

**Jacobians**
It was not entirely clear how I used the Jacobian [here](https://github.com/JeroenDM/moveit_constrained_planning_plugin/blob/d3a76f6343d1749cd948c56650438b2d1493d5ab/planning_plugin_examples/include/planning_plugin_examples/example_robot.h#L72) as explained in [constrained_model.pdf](https://github.com/ros-planning/moveit/files/4769286/constraints_model.pdf). Maybe to clarify this a bit it is useful to mention how the Jacobian will be used by the constrained planner inside OMPL. This is explained on page 15 of the paper [Sampling-Based Methods for Motion Planning with Constraints](http://www.kavrakilab.org/publications/kingston2018sampling-based-methods-for-motion-planning.pdf)

**General remarks**

- Omid: Add instructions to run the example code I write. **TODO**
- Felix: In example videos, visualize start state, goal state, and path constraints. **TODO**
- Everyone: Reach out during the week if I’m stuck on a problem.
- I will take some vacation time next week Monday and Tuesday, if the weather is nice at least :)

I hope I did not miss anything mentioned in the meeting.

JeroenDM, 2020-06-16:
I created another [very simple planning case](https://github.com/JeroenDM/moveit_constrained_planning_plugin/blob/363061ea374e6b0ef6e588eea478b387626c5897/planning_plugin_examples/scripts/kuka_kr5_example.py) for debugging, using a 6 DOF Kuka KR5 robot. This case also has issues with orientation constraints.

In the [video](https://youtu.be/Kgqu3vFWX8A) I experimented with [rviz_tools_py](https://github.com/DavidB-CMU/rviz_tools_py) to visualize the planning problem before showing the solution. (Without having to edit video's.) It doesn't work that well, but maybe already better than text in the video description :) @felixvd

felixvd, 2020-06-17:
Interesting package! Definitely looks better, but if you're already displaying it manually, try publishing the start and goal state as a [RobotState](http://docs.ros.org/melodic/api/moveit_msgs/html/msg/RobotState.html) (maybe colored green and gold to keep the same color scheme as the MotionPlanning plugin) and setting the alpha of the trajectory to 1.0. And naturally leave out the pose that isn't part of the planning problem, that's confusing. Should look real good.

JeroenDM, 2020-06-17:
> try publishing the start and goal state as a RobotState (maybe colored green and gold to keep the same color scheme as the MotionPlanning plugin) ...

That was my initial idea, but the python interface for visualizing stuff is Rviz is not ideal yet, and it was getting late :) I'll keep this in mind. It would be great if [moveit_visual_tools](https://github.com/ros-planning/moveit_visual_tools) is ported to Python, or does this already exist and did I miss it?

mamoll, 2020-06-17:
I wouldn't call my approach to sampling nearby orientations esoteric. I prefer "correct" :-). If you want uniform sampling of orientations that are _at most_ some distance _d_ away, that's the way to do it. If you want orientations that are _exactly_ distance _d_ away, then skip the `pow(...)` term in the code you linked to. You could create a polyhedral approximation by just taking the convex hull of a few quaternions that are distance _d_ away. This is analogous to how friction cones in 3D are often approximated by a polyhedron.

felixvd, 2020-06-18:
@mamoll I didn't have a look at the details for this, but how would you constrain the rotation around a certain axis in quaternion space? Can the [descartes](https://github.com/PickNikRobotics/descartes_capability) use case (one axis almost unconstrained) be described? It's a very common constraint.

@JeroenDM Seconding that a Python version of `moveit_visual_tools` would be very useful, I just assumed there wasn't one.

mamoll, 2020-06-18:
@felixvd Let the desired rotation axis be omega and the range of rotations be [theta_low, theta_high]. Then sampling any theta in that range, combined with omega will give you a valid rotation in axis-angle representation. Converting that to a unit quaternion is straightforward. 

Normally, quaternion distance between two unit quaternions q0 and q1 is defined acos(q0 . q1), with "." being the vector inner product. By changing that to acos(q0^T x D x q1), where D is a symmetric positive definite matrix, you create an anisotropic distance measure. For a given q0, all the q1's that are within some distance threshold using this measure will form an ellipsoid. With an appropriately selected D, you should be able to model constrained motions around 1 axis or 2 axes (both would be degenerate ellipsoids).

JeroenDM, 2020-06-18:
@mamoll 
> I wouldn't call my approach to sampling nearby orientations esoteric. I prefer "correct" :-).

:) Absolutely! I don't know why I wrote esoteric... It is the correct approach for uniform *sampling*, and a valid alternative to *represent* a constraint region. My current implementation never directly *samples* the constrained region.

@felixvd 
> Can the descartes use case (one axis almost unconstrained) be described?

For tolerance on rotation around a single axis, you can sample this angle uniformly. But with tolerance around more that one axis, the current implementation of descrates produces biased samples. The important question is then, does this bias influence the planner's performance. For constraints similar to the ones I want to describe, [Berenson](https://dl.acm.org/doi/10.1177/0278364910396389) suggests it does not:

> We observe that while our method ensures a uniform sampling in the bounds of Bw, it could produce a biased sampling in the subspace of constrained spatial displacements SE( 3) that Bw parameterizes. However, this bias has not had a significant impact on the runtime or success-rate of our algorithms.

It would be very interesing to validate this, by comparing both approaches.

mamoll, 2020-06-18:
Sampling orientations that satisfy orientation constraints and projecting orientations onto the nearest orientations that satisfy orientation constraints are two different different things. You also need the latter when trying to "fix up" configurations along a path that violate the orientation constraint, right?

mamoll, 2020-06-18:
If there's a simple implementation in Descartes that produces biased samples, then I'm fine if you use that for now. We can look it at it later whether that bias really matters.

JeroenDM, 2020-06-19:
> You also need the latter when trying to "fix up" configurations along a path that violate the orientation constraint, right?

@mamoll Should I be doing this? As far as I understand, the states from OMPL's solution should **not** violate the constraints as mentioned [here](http://ompl.kavrakilab.org/constrainedPlanning.html).
> **Any motion plan generated using this augmented state space will satisfy the constraint,** as the primitive operations used by motions planners (e.g., ompl::base::StateSpace::interpolate) automatically generate constraint satisfying states. 

Or did I misinterpret your question?

ommmid, 2020-06-22:
@JeroenDM maybe it is good to investigate more in details how [`interpolate()`](https://github.com/JeroenDM/elion/blob/59bfc3f4ac9a5499c17ac1a5bde5332c867b461c/elion_planner/src/planner.cpp#L77) generates states between the start and end without violating constraints and explain it here.

Just to throw and idea:
What if you get the solution of only-position-constraint problem and examine its path for the orientation. If the orientations along the path are not too off compared to the orientation constraint, then break down the path into segments and put the orientation constraint at the end of each segment and then solve and use [`interpolate()`](https://github.com/ompl/ompl/blob/8f21d6519878456a7664732a78fde814f2e2f3ee/src/ompl/geometric/src/PathGeometric.cpp#L307).

JeroenDM, 2020-06-25:
> @JeroenDM maybe it is good to investigate more in details how interpolate() generates states between the start and end without violating constraints and explain it here.

I looked into it a little bit, but it does not look trivial. Therefore it seemed sensible to first check if the interpolated solution does violate the constraints. I [added a simple check](https://github.com/JeroenDM/elion/commit/ff30737d6ce9872fd01f93361296d7a913ad588f) for the solution path and it does not violate the constraints. So, if this check is correct, I don't have to worry about invalid paths for now.

@ommmid I do not entirely understand the idea. Is the approach you explained intended to handle problems with both position *and* orientation constraints?

> then break down the path into segments and put the orientation constraint at the end of each segment 

The current implementation has no concept of path segments or adding constraints *at the end of a segment*. The `path_constraints` are applied to the complete path. In other words, the constraints are used to model a new state space (`ProjectedStateSpace`) in which the standard OMPL planners can do there thing, without having to know how the constraints work.

JeroenDM, 2020-06-25:
I started working on combining position and orientation constraints. OMPL's [ConstraintIntersection](https://ompl.kavrakilab.org/classompl_1_1base_1_1ConstraintIntersection.html) class could be a nice way to achieve this. It would also allow support for an arbitrary number of positions or orientation constraints. (In addition, I could split up position constraints for each variable, instead of using infinite bounds on unconstraint variables.)

I created a simple working example outside MoveIt [here](https://github.com/JeroenDM/ompl_constrained_planning_in_se3/blob/2597ca5ad8e8148827fd9417740b73590c21f57f/constraint_intersection_on_se3.cpp#L90). Although, for melodic I will have to use an older version of OMPL, related to [these changes](https://github.com/ompl/ompl/commit/d4088bbcc124eef5c2e34c1ef6dc79d68200fa77).

JeroenDM, 2020-06-25:
After struggling with the `ConstraintIntersection` class a bit, I tried another simpler approach (but less flexible) to combine position and orientation constraints. I created a [welding example](https://github.com/JeroenDM/elion/blob/welding-example/elion_examples/src/welding_example.cpp) script and [hardcoded](https://github.com/JeroenDM/elion/blob/00cd0fbcb456939b6080aa94477ca19eb20b1bf6/elion_planner/src/constraint.cpp#L212) the constraints, where I ignore constraints on the rotation around the tool's z-axis.

When I make the constraint region to small, it is too slow or does not work yet. I will investigate this and explain it in more detail tomorrow. [Here](https://youtu.be/2MmdGjysqUE) is already a quick preview.

For this type of problem with a low dimensional constrained manifold (6 degrees of freedom - 5 constraints = 1 dimension), it could be that this projection based planning is not ideal. I expect that using analytical inverse kinematics in the [Constraint::project](https://ompl.kavrakilab.org/classompl_1_1base_1_1Constraint.html#a84203729cf979b398e8e59f92d136ed3) method could greatly improve performance for these low dimensional manifolds. But this is not trivial to implement and can only be applied to specific robots.

JeroenDM, 2020-06-26:
I tried adding collision checking in a minimal way because most of the examples can be solved with simple Cartesian interpolation if there are no collision objects. I created a new branch [add-collision-checking](https://github.com/JeroenDM/elion/tree/add-collision-checking) and the implementation is mostly copied from the existing interface in MoveIt. 

I discovered that I have absolutely no idea how the `planning_scene`, `planning_scene_interface`, and `planning_scene_monitor` work to add collision objects. I still have a lot to learn.

Also, the results of a task with orientation constraints and a single collision object are not ideal yet :) I probably should add some workspace bounds.
[image]

felixvd, 2020-06-26:
Which reminds me: Do we have an easy way in MoveIt to constrain only part of the robot (e.g. the end effector) to a bounded area? I found [this thread](https://answers.ros.org/question/270389/fix-workspace-bounds/), but it doesn't sound promising.

v4hn, 2020-06-26:
On Fri, Jun 26, 2020 at 07:46:21AM -0700, Felix von Drigalski wrote:
> Which reminds me: Do we have an easy way in MoveIt to constrain only part of the robot (e.g. the end effector) to a bounded area? I found [this thread](https://answers.ros.org/question/270389/fix-workspace-bounds/), but it doesn't sound promising.

The workspace concept does not even apply to the stationary robot at all... Something we might want to change at some point..

You can add path constraints with position regions for individual links.
I'm not sure whether the planners will actually handle it correctly though.

JeroenDM, 2020-06-26:
> You can add path constraints with position regions for individual links.
I'm not sure whether the planners will actually handle it correctly though.

Mmm, interesting, I don't know yet how the current OMPL interface handles constraints on arbitrary links. I would expect that it relies on the [constraint_samplers](https://github.com/ros-planning/moveit/tree/master/moveit_core/constraint_samplers) in `moveit_core`. I should look into this in more detail.
In my current implementation, it is not to difficult to allow constraints on arbitrary links, I think.

JeroenDM, 2020-06-26:
I created [another example](https://github.com/JeroenDM/elion/blob/03815c6c213ef4f6dcc8ee03f1c0c548f89e44e6/elion_examples/src/kuka_example_with_collision.cpp) to check planning with position constraints and collision avoidance. This one is slow. Compiled in release mode it takes 10 to 100 seconds.

Any suggestions for non-industrial examples are welcome. A more diverse set of examples should hopefully lead to more flexible code. Maybe something with a humanoid robot?

[image]

v4hn, 2020-06-26:
> I would expect that it relies on the [constraint_samplers](https://github.com/ros-planning/moveit/tree/master/moveit_core/constraint_samplers)

For goal constraints I would expect so. But if I remember correctly non-orientation path constraints will not change the sampling procedure.
I would expect simple rejection sampling for pure position path constraints on arbitrary links.

That is to say, I expect better from your approach. :-)

mamoll, 2020-06-26:
That Kuka example looks pretty cool. I suspect planning is slow because you don't have an analytical Jacobian. However, for end-effector pose constraints you can get the constraint Jacobian terms from the RobotState via [RobotState::getJacobian()](http://docs.ros.org/melodic/api/moveit_core/html/classmoveit_1_1core_1_1RobotState.html#a3242f96101baaed7ce83881c67d182bc). Using that inside PositionConstraint::jacobian and RPYConstraints::jacobian should speed things up a lot.

JeroenDM, 2020-06-28:
@mamoll 

For `PositionConstraints`, I use an analytical Jacobian at the moment, although there is some level of indirection and Eigen constructor calls that could make it slow.

1. The `RobotState::getJacobian()` is called in `BaseConstraint::geometricJacobian`.
2. The first three rows are extracted in the position constraints specific `PositionConstraint::calcErrorJacobian`.
3. Then the Jacobian is converted to take the bounds into account in the generic method `BaseConstraint::jacobian`.

It is the last one that I override for `AngleAxisConstraints` with the default numerical one in `ompl::base::Constraint::jacobian`.

Other performance notes:
- If I don't use the analytical Jacobian for position constraints in the Kuka problem with collision objects, it consistently times out at 100 seconds. ( And I did not wait longer.)
- Without the collision objects, the problem Kuka problem gets solved in under a second.

JeroenDM, 2020-06-28:
> For goal constraints I would expect so. But if I remember correctly non-orientation path constraints will not change the sampling procedure.
I would expect simple rejection sampling for pure position path constraints on arbitrary links.

It would be very interesting to implement direct sampling of (simple) path constraints, but still modeling the state using the joint values. This would be a new (and hopefully better) version of the current [pose_model_state_space](https://github.com/ros-planning/moveit/blob/b8b57846650f7662a44106e0b7728e0c9ca36888/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/parameterization/work_space/pose_model_state_space.h). The latter stores the state as an SE3StateSpace. I suspect it could be faster than OMPL's projection based constrained planning approach for problems with both position and orientation constraints.

@mamoll @felixvd @ommmid This could be the topic of the second phase of GSoC.

JeroenDM, 2020-06-28:
I tried adding support for arbitrary orientation of the box that specifies the constraint region for position constraints. This kind of works, but not ideally. I quickly wrote down the formula used in a pdf.
[position_constraints.pdf](https://github.com/ros-planning/moveit/files/4842840/position_constraints.pdf)

JeroenDM, 2020-06-29:
## Meeting notes 29/06/2020

**General remarks**

- Use whatever latest version of MoveIt and OMPL I need to solve a problem quickly.
- Visualize constraints by coloring the robot depending on whether it satisfies the constraints in RVIZ (when moving it around with an interactive marker.)
- To have a better idea of what is working, create an example with a parameter that can be changed to create slight variations. Some of these variations work, others not. (Reproducible!)
- Don’t focus on multiple constraint regions for now. This could be done with the MoveIt Task Constructor in the future.
- Don’t add lazy goal sampling for now. (Where the planner selects a joint configuration for a given goal region in task space.) A fixed joint configuration goal is fine for now.
- Don't be sad if the planner is slow.

**Plan for this week**

- Figure out why OMPL’s sanity checks for the Jacobian do not work. Explain the sanity checks I did myself for the Jacobian.
- Organize the examples (in a table?). Start from a very simple example and build it up gradually. The table can have empty slots for aspirational examples.

JeroenDM, 2020-07-01:
I created a [script](https://github.com/JeroenDM/elion/blob/5709ef13d4c58568f8448b1feb22fbf135b680f3/elion_examples/scripts/publish_collision_scene.py) to make it easy to add different collision scenes to test different planning cases. For example, to load the collision geometry from `panda_1.urdf`:

```bash
rosrun elion_examples publish_collision_scene.py panda_1
```

I have some trouble understanding how to use this planning scene when planning, so I asked a [question](https://answers.ros.org/question/356183/how-to-use-the-planning-scene-monitor-in-c/) on ROS answers.

All this is in the context of [creating an overview](https://docs.google.com/spreadsheets/d/1stQlRhBhPN7oWiNFP_NP_y2UuipUJ-9rYIfIOLDluFg/edit?usp=sharing) of the examples I tried and posted here in a very unorganized way.

JeroenDM, 2020-07-06:
## Meeting notes 06/05/2020
Thank you for going through my work and providing feedback :) @mamoll @ommmid @felixvd 

Done, last week I:

- [Found out](https://github.com/JeroenDM/elion/blob/7424f4741e74503508071b6f1c41e3f7a8bb88d8/elion_planner/test/test_constraints.cpp#L241) why OMPL sanity check failed.
- [Organized](https://docs.google.com/spreadsheets/d/1stQlRhBhPN7oWiNFP_NP_y2UuipUJ-9rYIfIOLDluFg/edit?usp=sharing) the examples, they run from JSON files now.
- Fixed many smaller errors in analytical jacobian RPY and angle axis constraints.
- Reached enlightenment regarding [parts of the PlanningSceneMonitor](https://github.com/JeroenDM/moveit_planning_scene_example).

Some todo alternatives for next week:

- Keep diving deeper into the constraint formulation for a while. RPY or Quaternion constraints.
- Write a planning scene monitor tutorial.
- Pose goal sampling -> move towards usable planner.
- Fix long-standing orientation bug in the current interface.

Feedback and comments during the meeting:

- Keep in mind that I have to check whether a solution even exists for a planning example, so I don’t spend time debugging examples that could never be solved anyway.
- It would be valuable to provide an easy interface or good tutorial for z-rotation free orientation constraints. (Glass almost upright, but rotation around symmetry axis is allowed.) This is to difficult to specify for many users.
- Maybe try [AtlasStateSpace](https://ompl.kavrakilab.org/classompl_1_1base_1_1AtlasStateSpace.html) and [TangentBundleStateSpace](https://ompl.kavrakilab.org/classompl_1_1base_1_1TangentBundleStateSpace.html). These could perform better for some of the planning examples. I only use [ProjectedStateSpace](https://ompl.kavrakilab.org/classompl_1_1base_1_1ProjectedStateSpace.html) at the moment. (There could be some theoretical issues with how I formulate the constraints now. E.g. are the bounds included or not. I can discuss this in detail with @mamoll  if needed. Certainly interesting but maybe not the focus of this project.)
- Use different planners, everything is done with RRTConnect for now. This is also a way to test my implementation. In theory it should not make a difference in the rest of the code which planner I use.
- Some of the less ideal paths in the examples could be a direct consequence of the fact that RRTConnect does not optimize anything, it finds a feasible path. Try RRTStar? (But do not spend time on specialized cost functions, use the default one.)
- Rviz constraint visualization 2.0: use an interactive marker to move the robot inside the constraints and block moving outside of them.

Conclusion:

- Create a plan for integration into MoveIt (to be presented/discussed on the next MoveIt maintainer meeting).
- Convert some of my examples in tutorials. Good examples and tutorials are important!

JeroenDM, 2020-07-08:
Until now, all examples used `RRTConnect`. I added [some code](https://github.com/JeroenDM/elion/commit/5e048593b1f23cc465dd3e36b1dbe02802a6f10b) so you can try other OMPL planners by setting the `"planner_id"` in the JSON file. I tried some other planners and this seems two work as expected.

OMPL also offers different ConstraintStateSpace implementations. Until now I used `ProjectedStateSpace`.  `AtlasStateSpace` and `TangentBundleStateSpace` do not work for the current constraint formulation, throwing the exception:
```
Error:   ompl::base::AtlasStateSpace::newChart(): Failed because manifold looks degenerate here.
```

[Rewriting position and orientation constraints](https://github.com/JeroenDM/elion/commit/3b981fc8d3ac007aeef40bdfae6e3ced9813b83a) as equality constraints allow us to use them. This gives interesting results for the difficult example `kuka_collision_2.json`. Using `PRMstar` in combination with `AtlasStateSpace` results in a reasonable solution within 10 seconds. With the `ProjectedStateSpace`, it takes 200 seconds to find a far-from-optimal solution.

[image]

For orientation constraints, I get the following warning while planning:
```
Warning: ompl::base::AtlasStateSpace::sampleUniform(): Took too long; returning center of a random chart.
```

I'm not sure what is making it slow, but I'll leave it at this and continue working on the MoveIt integration for now. The current demo code probably shows that it is possible and useful to do constrained planning with OMPL. Now I can start making it useable :)

JeroenDM, 2020-07-10:
@v4hn Do you know why the [ModelBasedStateSpace](https://github.com/ros-planning/moveit/blob/d6666a9aee734acd1891a43657cf162c9ca4e9b7/moveit_planners/ompl/ompl_interface/src/parameterization/model_based_state_space.cpp) inherits from `ompl::base::StateSpace`, and not from [RealVectorStateSpace](https://github.com/ompl/ompl/blob/master/src/ompl/base/spaces/src/RealVectorStateSpace.cpp). It seems this could avoid quite some code duplication. (Both originally written by @isucan .) I can't figure out why this would not work.

I'm not sure this would help me to integrate constrained planning, but it would at least make the current interface slightly easier to understand.

mamoll, 2020-07-10:
@JeroenDM, the  ModelBasedStateSpace::StateType is quite different from the RealVectorStateSpace::StateType. The constrained planning code in OMPL converts things internally to Eigen vectors, so maybe what you need is some helper method in the StateSpace base class that returns a "view" of a state as an Eigen::Map. For state spaces with states composed of a contiguous array of doubles (or with a fixed stride), that is easy to do. For everything else, you can throw an exception. @zkingston, any thoughts you want to add?

v4hn, 2020-07-10:
That being said, (1) I believe MoveIt's defaults do not use the Eigen::Map approach yet, even if continuous joints sets are used. This could really speed things up according to some tests by Dave. And, (2), if changing state space can simplify the ompl wrapper, I'm all for it. The two might be exclusive though.

mamoll, 2020-07-10:
_In general_ robot states cannot be mapped with Eigen::Map, but ModelBasedStateSpace::StateType contains a double*, just like RealVectorStateSpace::StateType. I am a little confused by @v4hn's comment. I think you are talking about something else? Avoiding allocs & copies are generally good ideas, but I am talking specifically here about providing the constrained motion planning framework within OMPL with the right hooks to be able to work with both RealVectorStateSpace and ModelBasedStateSpace.

JeroenDM, 2020-07-11:
@mamoll 
>  I am talking specifically here about providing the constrained motion planning framework within OMPL with the right hooks to be able to work with both RealVectorStateSpace and ModelBasedStateSpace.

Although this is what I am trying to do in the "long" run, I asked the question also in the context of @v4hn's comment:
>  if changing state space can simplify the ompl wrapper

---

But **in** the context of the constrained motion planning with OMPL, I was thinking about the idea of extending `ModelBasedStateSpace` to be a valid `ompl::base::ConstrainedStateSpace`. This would make the integration much easier, but I do not think it's possible, because you would end up with some weird multiple inheritance scenario.

In any case, I think simplifying the current interface would greatly improve my chance of success. This would require some tests to make sure I don't break anything. This, in turn, would require me to have a good idea of the requirements of the current interface. That's the main part where I'm lacking. Especially when it comes to multiple planning groups and [subspaces](https://github.com/ros-planning/moveit/blob/091bdfb765e46ba399e6983f49a81c502b0bcb65/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp#L353).

I'm digitalizing all my notes from going to the current interface and summarizing what I tried so far. (With the next maintainer meeting in mind.)

JeroenDM, 2020-07-13:
[Some notes](https://github.com/JeroenDM/elion/wiki/Current-ompl-interface-in-MoveIt) from going through the code of the current interface.

zkingston, 2020-07-13:
@mamoll I don't think there is anything technically incorrect with having ModelBasedStateSpace inherit from RealVectorStateSpace. The constrained planning code really just relies on the double array within the state so it can use Eigen::Map, as we don't have a fancier way in OMPL of specifying that there is an underlying contiguous array for the state's real values. IIRC some other code might also play nicer if ModelBasedStateSpace is a RealVectorStateSpace, like BIT* or some of the KPIECE projections. See my ancient pull request (https://github.com/ros-planning/moveit/pull/897), I did this exact change to simplify interface stuff.

@JeroenDM 
Extending ModelBasedStateSpace to ConstrainedStateSpace is also not a bad idea - you would just have to implement your own constrained integration code which could easily be cribbed from the ProjectedStateSpace.

JeroenDM, 2020-07-15:
I kind of made it work in the current interface by ignoring all the code related to goal sampling and temporarily assuming a single joint goal constraint which is extracted manually in [ModelBasedPlanningContext::configure](https://github.com/JeroenDM/moveit/blob/18198b73f568c1c0fa3816168681fa3d82081d51/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L152).

I think the issue in the code related to goal sampling is the following. In many places a generic `ompl::base::State` gets cast to a `ModelBasedStateSpace::StateType` to access the joint values:
```C++
state->as<StateType>()->values
```

When the state that goes in has the type `ConstrainedStateSpace::StateType`, the casting goes wrong and bad things happen.

> @JeroenDM
Extending ModelBasedStateSpace to ConstrainedStateSpace is also not a bad idea - you would just have to implement your own constrained integration code which could easily be cribbed from the ProjectedStateSpace.

This would solve the problem mentioned above, but I don't like the idea of copy-pasting code from OMPL if I can find a way to use it directly. (I assume this is what you mean by "cribbed", I've never heard that word in this context :) @zkingston )

JeroenDM, 2020-07-16:
I came up with another option to solve the casting problem. We can add a third type of state space that inherits from `ModelBasedStateSpace`, let's call it `ConstrainedPlanningStateSpace`. (It is *not* a constrained state space, but a state space to be used when doing constrained planning.)
Now we can override `copyToRobotState` and implement it as:
```C++
void ompl_interface::ConstrainedPlanningStateSpace::copyToRobotState(moveit::core::RobotState& rstate,
                                                                     const ompl::base::State* state) const
{
  const Eigen::Map<Eigen::VectorXd>& x = *state->as<ompl::base::ConstrainedStateSpace::StateType>();
  rstate.setJointGroupPositions(spec_.joint_model_group_, x);
  rstate.update();
}
```
instead of the default implementation in `ModelBasedStateSpace`:
```C++
void ompl_interface::ModelBasedStateSpace::copyToRobotState(moveit::core::RobotState& rstate,
                                                            const ompl::base::State* state) const
{
  rstate.setJointGroupPositions(spec_.joint_model_group_, state->as<StateType>()->values);
  rstate.update();
}
```

This means I can revert my changes in [ModelBasedPlanningContext::convertPath](https://github.com/JeroenDM/moveit/blob/400ea7596a34b5b331030746da891d690f85fd20/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L480) to use `spec_.state_space_->copyToRobotState` again. However, I'm struggling a bit to do the same for `copyToOMPLState` and `copyJointToOMPLState`.

What do you think of this idea? @v4hn @mamoll

JeroenDM, 2020-07-16:
The approach explained above seemed to work well! I can now use the existing goal sampling functionality, although I have yet to test pose goals. I'm coming up with a plan to organize the pull requests to introduce the changes. I hope to submit them tomorrow or Monday so we can start discussing actual code :)

JeroenDM, 2020-07-22:
The action is moving away from this thread at the moment, but here is a quick update:

### OMPL constrained planning integration
I split up the integration process is several smaller pieces. My plan is to continue the discussion specific to this integration in the main pull requests that explains this process: #2219. It still needs a lot of testing / benchmarking / examples to clearly show what the advantages and disadvantages of the new planning technique will be.

### Tracking down difficult segfaults
I encountered some bugs that only showed up in release mode. I started to use [valgrind](https://valgrind.org/) like this:
```bash
valgrind --leak-check=full --track-origins=yes --xml=yes --xml-file="/home/jeroen/valout.xml"  $TEST_EXECUTABLE
```
To make it work, I also **build OMPL from source** now (version 1.5.0), because of [this issue](https://bitbucket.org/ompl/ompl/issues/543/valgrind-overflow-error-due-to-boost-long).
I ended up using **address sanitizer** from clang more that valgrind, building the tests with:
```bash
catkin build moveit_planners_ompl --no-deps --make-args tests --cmake-args -CMAKE_CXX_COMPILER=/usr/lib/ccache/clang++ -DCMAKE_CXX_FLAGS=-fsanitize=address
```
and running them:
```bash
ASAN_SYMBOLIZER_PATH=/usr/lib/llvm-6.0/bin/llvm-symbolizer $TEST_EXECUTABLE
```
Where the prefixed variable is needed to get meaningful file descriptions in the output. A small inconvenience, once all of the above is enabled and used, I cannot get it disabled anymore without cleaning and rebuilding the workspace (which can take +30 minutes, since I'm also building OMPL now).

### Simplifying the current interface
I keep finding things I can **remove / rewrite / simplify** in the current interface, now that I'm getting familiar with it. (Although I'm strategically avoiding the `PoseModelStateSpace` for now.) One of the more interesting rewrites that took me a while to understand was in [void ompl_interface::ModelBasedStateSpace::copyJointToOMPLState](https://github.com/JeroenDM/moveit/commit/8e7a9443e8cd7e71602862a80f014f96aab0dc0e)

## Conclusion
I'm still having fun :) but I'm possibly doing too many things at once since I'm working on ~10 branches simultaneously...

JeroenDM, 2020-07-23:
In the last meeting, we discussed that I should create more examples that show why adding the new capability would be a good idea. @mamoll @felixvd @ommmid 

One of the original goals was the solve the joint space jumps in plans created by the default planners using the `PoseModel`. But after a little experimenting with simple examples, I'm not sure that the new planner will be a better alternative than the current `enforce_joint_state_space` hack. The best solution for this specific problem is still fixing the `PoseModelStateSpace`, I think.

There are of course many other exciting possibilities for specific types of constraints, mainly with small constrained regions where the `enforce_joint_state_space` hack (rejection sampling) underperforms. But I need more time to investigate this with good examples. I only have the [welding example](https://github.com/ros-planning/moveit/issues/2092#issuecomment-652502533) and another example with equality orientation constraints shown below.

(I wrote the above intermediate conclusion in the context of the upcomming [MoveIt manipulation meeting](https://discourse.ros.org/t/moveit-manipulation-wg-meeting-july-23rd/15304). I've created [some slides](https://docs.google.com/presentation/d/1AH00yOQRF33PloTbtKEZVBj7sL6jwIcnKYIHlnvyMTc/edit?usp=sharing) with short videos to support the discussion.)

[image]

[18 more replies]