moveit/moveit #2093: Ninja for builds (as a tactic to reduce build times)

tylerjw, 2020-05-18:
The ninja build system (https://ninja-build.org/manual.html) is a language like make that cmake can generate for.  `catkin_tools` does not support his build system, however, `catkin-make` does (with the `--use-ninja` flag).  Locally I got about a 4min speedup that was invariant to the state of my build-cache by building using catkin_make with the ninja flag as opposed to building moveit with catkin_tools or catkin_make without the flag.

`colcon` does support ninja through the argument `--cmake-args " -GNinja"`.  If you try this with `catkin_tools` it doesn't work as catkin is not pure cmake and doesn't account for any other generator than make.

For moveit2 we can easily just use ninja.  If we want to reduce the build times of moveit we could consider contributing ninja support to catkin or convert moveit1 to colcon.  I'm not sure how much work either of those options are.

henningkayser, 2020-05-18:
Colcon supposedly supports (isolated) catkin workspaces https://colcon.readthedocs.io/en/released/user/quick-start.html#build-ros-1-packages. I've never tried this but for CI this might actually be an option.

v4hn, 2020-05-18:
> I've never tried this but for CI this might actually be an option.

If possible, CI should build with the same configuration we deploy and advise users to use.

tylerjw, 2020-05-18:
It looks like there is a problem with how ninja output is processed with colcon that hasn't been addressed yet, so sadly we might not be able to use it with ROS2 ci systems yet either: https://github.com/colcon/colcon-cmake/issues/67