moveit/moveit #2095: Stop Closing of Gripper if contact with object is made.

ShrutheeshIR, 2020-05-19:
**Is your feature request related to a problem? Please describe.**
I am not sure if this issue belongs to the feature requests template, if it does not, apologies for the same.  
I am simulating a robotic arm in Gazebo using Moveit! I am able to control my planning groups and get my arm and gripper near the object in Python. My issue is with closing the gripper on the object. Since ROS does not have knowledge of the object, it proceeds to close the gripper fully, and thus, in my simulation, grasping is not successful, and the object flies away, due to the gripper constantly closing. Is there a good way to approach this, to stop the gripper closing on contact? I did look up the tutorials on Moveit for PickPlace and Grasps, however, they hardcode the collision objects. Currently, I do not such flexibility. How would I go about this issue?

**Describe the solution you'd like**
One solution I could think of, is to publish Contact messages for the gripper jaws through Gazebo (Using the contact plugin, I am able to do this), and once a contact has been made, instruct moveit to stop execution. Is this a good way to approach it? If it is, how can I asynchronously tell the moveit commander to stop the grasp for the jaw?

**Describe alternatives you've considered**
Another solution I could think of, was to have some physical constraints in the gripper jaw encoded through Gazebo such that moveit closing will fail and stop automatically at some point. But I'm not sure if such a feature does exist. If so, how do I do it?

**Additional context**
Simply put, how do people approach this issue, or am I doing something entirely wrong?

v4hn, 2020-05-19:
> Is there a good way to approach this, to stop the gripper closing on contact?

Usually you do not want to stop the gripper, because it should still exert a reasonable grasping force.
This can be achieved with a fitting controller parameterization or a maximum force for the closing motion.
Still, you likely have to use a custom FJT/GripperCommand controller to report success if the gripper is "stuck", having grasped something successfully.

Achieving reasonable grasp control is quite difficult actually, but often be replaced by "dump" solutions.

ShrutheeshIR, 2020-05-19:
Hi @v4hn Thanks for your input!
Could you roughly direct me on how to set the maximum force for closing motion?

felixvd, 2020-05-19:
Duplicate of https://answers.ros.org/question/352566/moveit-stop-gripper-close-on-contact-with-object/

v4hn, 2020-05-19:
> Could you roughly direct me on how to set the maximum force for closing motion?

Sorry, but that depends entirely on your controller.
If you are talking about the MoveIt side though, sending the grasping trajectory,
the GripperCommand action client interprets the effort of the last waypoint as the grasping force:
https://github.com/ros-planning/moveit/blob/master/moveit_plugins/moveit_simple_controller_manager/include/moveit_simple_controller_manager/gripper_controller_handle.h#L133

More generally, if you specify the effort for your waypoints, your controller could consider this information.
Or you just generally restrict the maximum output force on the controller side.

ShrutheeshIR, 2020-05-19:
Hi @v4hn alright, I suppose I have to play around with my controller forces, and dig deeper into it. Thanks for guiding me!
I'll close the issue

closed 2020-05-19 by ShrutheeshIR