moveit/moveit #2099: PlanningScene could maintain a variable number of robots

v4hn, 2020-05-21:
Taken from #2097:

> [...] I think most people would prefer if the collision environment was simpler and the collision environment not split into robot and world.

I don't see a need to give up on the notion of a "robot" in the scene as long as it is transparent w.r.t. transforms. But it would make some sense to generalize the PlanningScene class to maintain a set of robots, supporting zero as well as multiple robots. :sunglasses:

Comments? Ideas? Initial implementations?

felixvd, 2020-05-21:
I think there is a discussion to be had about what a `move_group` and a planning group should contain, how to deal with multiple robot arms moving simultaneously and/or independently (a real requirement) and manipulators on mobile robots (an equally real use case). I remember that those use cases came up at the workshop in Macau last year, too, and I dimly remember @mikeferguson wanting to write up the results of our group, but that might be wishful thinking on my end.

Apart from that, I am not sure how to implement it concretely, but I would be interested to know how other motion planning software like OpenRAVE approaches it, and if @Levi-Armstrong has any thoughts from Tesseract.

henningkayser, 2020-06-02:
In general, I would really like to have built-in support for this instead of having to work around this with multiple joint model groups. My concern is just that maybe it's not really worth it to represent all robots in a single PlanningScene class if the different robots can't be added dynamically. I would rather look into supporting syncing/merging multiple planning scenes dynamically so that multiple robots can "share" their knowledge about their environment. This could be implemented as a peer-to-peer setup (multiple PSMs are synced using registered update topics) or there could be a master PSM acting as a ground truth (PSMs of individual robot setups would sync to this one). For both approaches, there would have to be a method to register the robot transforms in reference to each other, possibly by using a common global scene with fixed world frame.