moveit/moveit #2107: Some tests take too long in debug mode

peci1, 2020-05-25:
As observed in #1964 by me and (reportedly) elsewhere by @v4hn, some tests take very long when run in debug mode.

I'm creating this issue to track the long tests. Any effort to make the tests debug-friendly will be welcome.

I've built moveit in debug mode and ran all the tests using `make run_tests` in the build directory.

This is the list of all tests that took more than 10 seconds on my computer. The list was built running:

    find test_results -name '*.xml' -exec grep -H "testsuite " {} \; | grep moveit | sed 's/^\([^:]*\):.*name="\([^"]*\)".*time="\([^"]*\)".*$/\1,\2,\3/' | sort -t, -k3 -gr | grep '[0-9]\{2,\}\.[0-9]*$'

Results file | Test Suite | Test time (sec)
-- | -- | --
test_results/moveit_ros_robot_interaction/gtest-locked_robot_state_test.xml | LockedRobotState | 3813,58
test_results/moveit_core/gtest-test_distance_field.xml | TestSignedPropagationDistanceField | 423,5
test_results/moveit_core/gtest-test_collision_distance_field.xml | DistanceFieldCollisionDetectionTester | 313,944
test_results/moveit_ros_planning_interface/rostest-test_robot_state_update.xml | testrobot_state_update | 153,6266
test_results/moveit_ros_planning_interface/rosunit-robot_state_update.xml | test | 149,607
test_results/moveit_planners_chomp/rostest-test_chomp_moveit.xml | testchomp_test | 95,4951
test_results/moveit_planners_chomp/rosunit-chomp_test.xml | CHOMPMoveitTest | 80,478
test_results/moveit_core/gtest-test_time_optimal_trajectory_generation.xml | time_optimal_trajectory_generation | 47,418
test_results/moveit_ros_planning_interface/rostest-test_move_group_interface_cpp_test.xml | testmove_group_interface_cpp_test | 41,7406
test_results/moveit_ros_planning_interface/rosunit-move_group_interface_cpp_test.xml | MoveGroupTestFixture | 40,758
test_results/moveit_ros_planning_interface/rostest-test_cleanup.xml | testcleanup_tests | 19,9096
test_results/moveit_ros_planning_interface/rostest-test_subframes_test.xml | testsubframes_test | 11,363
test_results/moveit_commander/rostest-test_python_moveit_commander_ns.xml | testpython_moveit_commander_ns | 10,0509

peci1, 2020-05-25:
Also,

    test_results/moveit_planners_chomp/rostest-test_chomp_moveit.xml

times out in debug mode.

For comparison, this is what I measured when running in release mode:

Results file | Test Suite | Test time (sec)
-- | -- | --
test_results/moveit_ros_planning_interface/rostest-test_robot_state_update.xml | testrobot_state_update | 154.2273
test_results/moveit_ros_planning_interface/rosunit-robot_state_update.xml | test | 151.3830
test_results/moveit_ros_planning_interface/rostest-test_move_group_interface_cpp_test.xml | testmove_group_interface_cpp_test | 33.8443
test_results/moveit_ros_planning_interface/rosunit-move_group_interface_cpp_test.xml | MoveGroupTestFixture | 32.943
test_results/moveit_ros_planning_interface/rostest-test_cleanup.xml | testcleanup_tests | 17.8916
test_results/moveit_ros_robot_interaction/gtest-locked_robot_state_test.xml | LockedRobotState | 11.584
test_results/moveit_ros_planning_interface/rostest-test_subframes_test.xml | testsubframes_test | 9.8739
test_results/moveit_ros_planning_interface/rostest-test_python_move_group.xml | testpython_move_group | 9.1544
test_results/moveit_ros_planning_interface/rostest-test_python_move_group_ns.xml | testpython_move_group_ns | 9.3556
test_results/moveit_planners_chomp/rostest-test_chomp_moveit.xml | testchomp_test | 9.9640

v4hn, 2020-05-25:
Thank you for investigating. I don't think this was reported anywhere else, I just said I noticed it locally.

By "debug mode", we refer to a catkin workspace configured with
```
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Debug
```

Results file | Test Suite | Test time (sec)
-- | -- | --
test_results/moveit_ros_robot_interaction/gtest-locked_robot_state_test.xml | LockedRobotState | 3813,58
test_results/moveit_core/gtest-test_distance_field.xml | TestSignedPropagationDistanceField | 423,5
test_results/moveit_core/gtest-test_collision_distance_field.xml | DistanceFieldCollisionDetectionTester | 313,944
... | ... | ...
test_results/moveit_ros_planning_interface/rostest-test_robot_state_update.xml | testrobot_state_update | 154.2273
test_results/moveit_ros_planning_interface/rosunit-robot_state_update.xml | test | 151.3830

The runtime of all of these tests is unacceptably high from my perspective.
3800s for fuzzy-testing thread-safe access to `RobotState` (granted, only in debug mode) is ludicrous.

As a rough guideline, I would think it a reasonable goal to have each test pass in under a minute in release and in under two minutes in Debug mode.