moveit/moveit #2108: Collision objects not align in Rviz

EulerPiGauss, 2020-05-25:
Hi, I'm using moveit+rviz to visualize a robot workspace. The planning scene is composed of an Abb manipulator and a few boxes as collision objects. Two boxes' centers(upper face) are measured in CAD software

[image]

For clarity, only robot base is visualized in this picture. As shown above, relative to robot's base link, bigger box(conveyor_belt) has center point (x=1025.379mm, y=1042.188mm) and smaller box(inner_bucket) has center point (x=584.102mm, y=1604.807mm). And the relative pose is a pure rotation by z axis with 51.892degree(quaternion [x, y, z, w]=[ 0, 0, 0.4375239, 0.8992068 ])

I use these measurements to apply collision objects in moveit. 

```yaml
collision_scene: # every shape should have "type", "name", "reference", "size", "position", "quaternion", "mesh_file" #  for primitive shapes like box, mesh_file is ""
  conveyor_belt:
    type: "box"
    name: "conveyor_belt"
    reference: "base_link"
    size: [0.9, 7.40, 0.8]
    position: [-1.025379, -1.042188, 0.0]
    quaternion: [0.8992068, 0.0, 0.0, 0.4375239] # w, x, y, z
    mesh_file: ""
  inner_bucket:
    type: "box"
    name: "inner_bucket"
    reference: "base_link"
    size: [0.9, 0.9, 0.58]
    position: [-0.584102, -1.604807, 0.69]
    quaternion: [0.8992068, 0.0, 0.0, 0.4375239]
    mesh_file: ""
```
and use getObjectsPoses API to get objects pose(shown in below screenshot), which is same as config.
[image]

However, in rviz, the conveyor_belt and inner_bucket are not aligned
[image]

Could you  give me some help for this strange case?

rhaschke, 2020-05-25:
This is a question to be put on https://answers.ros.org.
Please try to place your collision objects at positions first that are easier to verify. If you can still reproduce the issue, please provide a minimal working example that illustrates the issue.

EulerPiGauss, 2020-05-25:
Thank you for your advice!

v4hn, 2020-05-25:
closing for now. If there is an actual issue in the framework, please comment again.

closed 2020-05-25 by v4hn