moveit/moveit #2109: How to use moveit on ROS Noetic?

jahad9819jjj, 2020-05-25:
I am ROS beginner from ROS Melodic era. I frequently used Moveit for a ROS Project and I experienced to upgrade version in ROS this first time. 
Of course, I know what Moveit cannot be used right after an upgrade. 

MoveIt is a very useful tool and I want to continue to use it, so I want to contribute to its development so that it can be used in Noetic. What can I do to help you?

It may not be the right question to ask here. If so, please close this. Thank you!

rhaschke, 2020-05-25:
The release is planned for end of June: https://moveit.ros.org/#release-versions
You are welcome to [build MoveIt from source](https://moveit.ros.org/install/source/) and report any issues you observe - optimally with appropriate PRs to fix them.

closed 2020-05-25 by rhaschke

hariharan382, 2020-09-02:
but still not yet released .but i found some dependacies in Arch linux can anyone say how to use it in noetic
https://aur.archlinux.org/packages/ros-noetic-moveit/