moveit/moveit #2110: TOPPRA time parameterization

v4hn, 2020-05-25:
I guess there are numerous other path parameterization algorithms out there,
but even the (quite nice) [TOTG](https://github.com/ros-planning/moveit/blob/master/moveit_core/trajectory_processing/include/moveit/trajectory_processing/time_optimal_trajectory_generation.h) algorithm we recently added and which improves tremendously over previous versions [is 8 years old by now](http://www.golems.org/projects/traj.html) and it would be interesting to see how more current results, as for example [TOPPRA](https://github.com/hungpham2511/toppra) (published 2018), behave in comparison.

The library seems well maintained and offers C++ bindings we could just forward from a PlanningRequestAdapter.

felixvd, 2020-05-26:
How about adding the WMD label?

v4hn, 2020-05-26:
I'm not sure whether it's *that* simple to do, but sure, someone might attempt it :+1:

gavanderhoorn, 2020-05-26:
[larics/topp_ros](https://github.com/larics/topp_ros) provides a service based interface to TOPP-RA (the Python version, not the C++ one).

v4hn, 2020-05-27:
> [larics/topp_ros](https://github.com/larics/topp_ros) provides a service based interface to TOPP-RA (the Python version, not the C++ one).

Could make sense for looking at parameterization results.
But of course we would be interested in a `PlanningRequestAdapter` and a C++ class abstraction in `moveit_core/trajectory_processing`.
Just mentioning this to avoid confused contributors. :-)

davetcoleman, 2020-06-18:
I think this is a great idea!

I had worked on porting TOPP to MoveIt a few years ago, but the version I ended up with was slow, perhaps due to a lot of strings being used. I probably wouldn't bother working off of my efforts:
https://github.com/davetcoleman/moveit_topp

felixvd, 2020-10-21:
In #160 @rbbg mentioned that they forked that and worked on it [here](https://github.com/rbbg/moveit_topp). Quoting from there:

> I haven't done any extensive testing but TOPP usually takes <200ms on a 7dof manipulator, but I've seen it take as long as 600ms.