moveit/moveit #2111: [QUESTION] Is it possible to make path planning for floating/planar robots?

alexisgaziello, 2020-05-27:
**Is your feature request related to a problem? Please describe.**
I am trying to set up moveit for path planning for a floating robot. I have a robot that consists of a simple link (let's say it's a sphere) and a virtual floating joint that goes from the reference frame to the base link.

When I try to use the MoveIt C++ interface, I am unable to do anything:
- getCurrentState doesn't work because there aren't any joints published in joint_state topic
- setPose doesn't work probably for similar reasons.
- most of functionality doesn't work.
The only thing I have managed to make "sort of work" is the function setJointValueTarget(), where I can send a vector of 7 doubles that correspond to the virtual joint positions (x,y,z...w), but if there are no articulated joints, the library is unable to make any planning.

**Describe the solution you'd like**
Make the current C++ interface work with only virtual non-fixed joints.

**Describe alternatives you've considered**
Maybe I am not setting up MoveIt correctly, but is the only alternative writing my own implementation of OMPL library for path planning?

v4hn, 2020-05-27:
MoveIt has been used for quadrotor planning in the past, but it's not the normal use case.
I would not be surprised to hear about issues with many APIs.

Feel free to file issues if you looked at the code and are sure the interface you are trying to use is broken. Minimal examples are always useful.

- Some code might (unnecessarily) require receiving messages on `/joint_states`. To be fair you're looking at an extreme corner case because almost everything you would call a robot has at least a single 1dof joint.

- I would expect `MoveGroupInterface::setPoseTarget` to work with a single floating joint.
  If it fails, it is surely not related to the `/joint_states` topic.

- "most of the functionality" is no concrete issue.

alexisgaziello, 2020-05-27:
Helo @v4hn 

Thank you for your response. I am going to try to develop a little more.

So this is what I have come up to:

1. It is true that a robot with only 1 virtual link is an extreme case. It may not even apply to my situation. The error occured when I downloaded a model of a quadcopter I found on internet. I wanted to make things as simple as possible to retrace errors to the origin. So I found this [model](http://wiki.ros.org/hector_quadrotor) only composed by a quadrotor and one joint that is fixed (that goes to a sonar, not important). After I created a package through the assistant where I named the planning group as "robot_group". [Here](https://drive.google.com/drive/folders/1nUznu1c5IU9EnLvJ2o41WHeEmEcPLBQb?usp=sharing) you should be able to download the package. (It will require Hector quadrotor [dependencies](https://drive.google.com/open?id=15VzmzjZR52A7ZCJT-f6pv0LP5sX3AbLn)).

_Note_: I changed in the tutorial code 2 strings, the PLANNING_GROUP from "panda_arm" to "robot_group" and the base link name, from "panda_link0" to "base_link". Edited [folder](https://drive.google.com/open?id=1kW7ruSeGt5BheE7IxlKZ-7UDXNNXoSyT) from the package moveit_tutorial.

When I launch the tutorial I get the following error:

```
[ INFO] [1590610442.220651547]: Loading robot model 'quadrotor'...
[ WARN] [1590610442.284916856]: No kinematics plugins defined. Fill and load kinematics.yaml!
[ INFO] [1590610443.492575568]: Ready to take commands for planning group robot_group.
[ INFO] [1590610444.496926142]: Didn't received robot state (joint angles) with recent timestamp within 1 seconds.
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1590610444.497000899]: Failed to fetch current robot state
[move_group_interface_tutorial-1] process has died [pid 32124, exit code -11, cmd /home/alexis/catkin_ws/devel/lib/moveit_tutorials/move_group_interface_tutorial __name:=move_group_interface_tutorial __log:=/home/alexis/.ros/log/8b739934-a056-11ea-b919-7cd1c3e3d80b/move_group_interface_tutorial-1.log].
log file: /home/alexis/.ros/log/8b739934-a056-11ea-b919-7cd1c3e3d80b/move_group_interface_tutorial-1*.log
```

When the pointer tries to get instantiated 

`  const robot_state::JointModelGroup* joint_model_group =
      move_group.getCurrentState()->getJointModelGroup(PLANNING_GROUP);`

There is an exception. Maybe this is intented, but the joint model group in this case should be the virtual floating joints right?

2. Let's try a case not that particular, with a robot with some "normal" joints, but let's configure it as passive and include in the planning group, as before, only the virtual joint and the base link.

This is the output I get when I run the tutorial:

```
[ WARN] [1590610288.189776151]: Could not identify parent group for end-effector 'robot_endEffector'
[ WARN] [1590610288.261047308]: No kinematics plugins defined. Fill and load kinematics.yaml!
[ INFO] [1590610289.353968727]: Ready to take commands for planning group robot_group.
[ INFO] [1590610289.623224608]: RemoteControl Ready.
[ WARN] [1590610290.130873308]: Topic '/rviz_visual_tools' unable to connect to any subscribers within 0.5 sec. It is possible initially published visual messages will be lost.
[ INFO] [1590610290.131214930]: Reference frame: /world_ned
[ INFO] [1590610290.131294317]: End effector link: 

Waiting to continue: Press 'next' in the RvizVisualToolsGui window to start the demo... continuing
[ERROR] [1590610320.845194632]: No end-effector to set the pose for
[ WARN] [1590610320.940889398]: Fail: ABORTED: Invalid goal constraints
[ INFO] [1590610320.941248919]: Visualizing plan 1 (pose goal) FAILED
[ INFO] [1590610320.941360370]: Visualizing plan 1 as trajectory line
```

The issue seems to be related with the end effector. The problem is that calling the function setEndEffectorLink("base_link") doesn't do anything. Creating an end_effector group through the setup assistant doesn't work either ways.

simonschmeisser, 2020-06-03:
I'd be surprised if there is nothing closer to your goal than MoveIt in the whole ROS ecosystem. MoveIt is mostly about stationary robots with many 1DoF joints (ie robotic arms). Maybe you should ask on https://answers.ros.org about how to best do planning for quadrocopters? Simple in terms of MoveIt would be an arm with 6 joints.

v4hn, 2020-06-05:
There are actually quadrotor videos highlighted in MoveIt's spotlight videos.
So if we don't consider this to be part of the domain, I would question why these videos are in there. :-)

alexisgaziello, 2020-06-06:
Hello @simonschmeisser , thank you for your interest. I agree with you, I found it kind of weird since it looks like a basic function that many robots would use. I have tried asking in ros answers but nobody responded, that's why I came here.

@v4hn What do you understand by MoveIt _Spotlight_ videos? Sorry, I'm not native English speaker and I am not sure to understand the word in this context, and google didn't get me any results.

I have searched YouTube and I found the following videos with MoveIt! and drones:

- https://www.youtube.com/watch?v=VlBQLbmc03g
- https://www.youtube.com/watch?v=lxyiQw5BzRw

The problem is that these videos use the “plan” button of the interface of MoveIt!, which behind the scenes just calls setJointValueTarget(), which also works for me. It's unknown if the rest of the functionality (for ex cartesianPath or setPose functions) of the MoveIt C++ Interface works for them, since they don't use it.

I also found a [video](https://www.youtube.com/watch?v=V4-GEtwcyJ0&t=203s) on setting up the drone, which didn't help since it's how I did the setup.

v4hn, 2020-06-08:
> What do you understand by MoveIt Spotlight videos?

I was referring to the two MoveIt Montage videos. However, looking at them, there are not quadrotors shown at all and I'm wondering which videos I had in mind...

Either way, I would consider your use-case should/is supported by MoveIt and your reports point out bugs with it.
I would encourage bug fixes (or additional features if never supported before) for interfaces that currently do not support floating joints.

JeroenDM, 2020-07-22:
> robots with many 1DoF joints

Going through the OMPL interface, there are many examples of code that [hints](https://github.com/ros-planning/moveit/blob/89a6752133177557fd4419d9f784d0f9c737c5c2/moveit_planners/ompl/ompl_interface/src/parameterization/model_based_state_space.cpp#L344) at support for > 1 DoF joints. Although I have idea yet to what extend this support continues throughout MoveIt.

For a free-floating joint, I would expect planning in OMPL needs to happen with a `PoseModelStateSpace` (which has an SE3 state space embedded). But without path constraints, this is not what happens, [I think](https://github.com/JeroenDM/moveit/pull/2).

I'm tremendously interested in any information on this topic :)

alexisgaziello, 2020-09-12:
Hello @JeroenDM, just saw the message.

The way I finally made path planning for floating robots (6DoF) was with the joint model group instead of the position of the end effector.

The joint model group has the property that the first joints are the virtual joints of the robot, which correspond to the 6DoF (3 of position, 4 of rotation).

The code looks something like:

  ```
void Display::plan(geometry_msgs::Pose & start, geometry_msgs::Pose & goal)
{

// Raw pointers are frequently used to refer to the planning group for improved performance.
  const robot_state::JointModelGroup* joint_model_group =
      move_group->getCurrentState()->getJointModelGroup(robot_group);

  // START POSITION
  robot_state::RobotState start_state(*move_group->getCurrentState());
  std::vector<double> joint_group_start(7);

  joint_group_start[0] = start.position.x;
  joint_group_start[1] = start.position.y;
  joint_group_start[2] = start.position.z;
  joint_group_start[3] = start.orientation.x;
  joint_group_start[4] = start.orientation.y;
  joint_group_start[5] = start.orientation.z;
  joint_group_start[6] = start.orientation.w;

  start_state.setJointGroupPositions(robot_group, joint_group_start);
  move_group->setStartState(start_state);

  // GOAL POSITION
  robot_state::RobotStatePtr current_state = move_group->getCurrentState();
  std::vector<double> joint_group_goal;
  current_state->copyJointGroupPositions(joint_model_group, joint_group_goal);

  joint_group_goal[0] = goal.position.x;
  joint_group_goal[1] = goal.position.y;
  joint_group_goal[2] = goal.position.z;
  joint_group_goal[3] = goal.orientation.x;
  joint_group_goal[4] = goal.orientation.y;
  joint_group_goal[5] = goal.orientation.z;
  joint_group_goal[6] = goal.orientation.w;

  move_group->setJointValueTarget(joint_group_goal);

  // Now, we call the planner to compute the plan and visualize it.
  // Note that we are just planning, not asking move_group
  // to actually move the robot.
  moveit::planning_interface::MoveGroupInterface::Plan my_plan;

  bool success;
  success = (move_group->plan(my_plan) == moveit::planning_interface::MoveItErrorCode::SUCCESS);

   // Save my_plan variable with you preferred method.
}
```

To create a cartesian path, just add the plan to and object of type trajectory:

    // Initialization of trajectory
    plan_ = moveit_msgs::DisplayTrajectory();
    plan_.trajectory_start = moveit_msgs::RobotState(my_plan.start_state_);
 
    // Adding points
    plan_.trajectory.push_back(my_plan.trajectory_);

I am unsure if this solution will work with every 6DoF robot, but it did for me.

Gaoyuan-Liu, 2021-03-01:
Hey @alexisgaziello,
I'm trying to do the same thing as you, and facing the same problem.
I tried the solution in your last post, but I still get the error:"Failed to fetch current robot state", I guess it's because the command `robot_state::RobotStatePtr current_state = move_group->getCurrentState();` is till there. 

Did you do other changes in your setup process? Do you mean you defined 6 joints between the /world frame and /base_link frame?

Thanks! Any information would be appreciated.