moveit/moveit #2112: Isometry check might be too strict

AndyZe, 2020-05-27:
Commit d15682b added a bunch of asserts like this:

ASSERT_ISOMETRY(pose)

Which is causing errors like this for my unit tests:

> 
The given transform is not an isometry! Its linear part involves non-unit scaling. The scaling matrix diagonal differs from [1 1 1] by [3.16556e-07 3.16556e-07 3.16808e-07] but the required precision is 1e-12.

Maybe we could reduce the tolerance to something like 1e-5?

AndyZe, 2020-05-27:
The tolerance is defined here:

`https://github.com/ros-planning/geometric_shapes/blob/melodic-devel/include/geometric_shapes/check_isometry.h`

rhaschke, 2020-05-27:
Have you involved float precision somewhere? Do you multiply matrices over and over?
By the way, the asserts should only apply to debug builds. But I agree increasing the tolerance threshold.

AndyZe, 2020-05-27:
Thanks for the suggestions - not that I know of. There's a Python camera calibration script I'm a little suspicious of. It does this:

`dq.inverse().homogeneous_matrix()[:3,:3]`  (Possibly not the right type of matrix)

Anyway, made a PR for the tolerance here:

https://github.com/ros-planning/geometric_shapes/pull/145

AndyZe, 2020-05-28:
> Do you multiply matrices over and over?

I would think MoveIt is doing that internally, right? When calculating distal link transforms.

AndyZe, 2020-05-28:
If this issue is also affecting you, and you came here looking for the solution, please leave a comment. I would like to know how many other users this is affecting.

v4hn, 2020-05-28:
> > Do you multiply matrices over and over?
> 
> I would think MoveIt is doing that internally, right? When calculating distal link transforms.

I believe Robert was talking about multiplying them over time (at least a few hundred times),
not across across the robot structure (which is limited by the longest kinematic chain).

If you can show that MoveIt accumulates such high noise internally, we do indeed need to relax
the requirement and/or need to renormalize under some circumstances.

jliukkonen, 2020-06-01:
I ran into this same issue today when trying to load scene geometry from a `.scene` file.

AndyZe, 2020-06-01:
A thought about relaxing this --

A robot's reach is usually about 1m. Requiring a "precision" of 1e-12 puts us at the picometer level. You measure atom radii in picometers. If we bumped this up, even to 1e-6, that would still be way more precise than robots ever actually need to be.

JStech, 2020-06-02:
What if instead of just dying the transforms were fixed? I'm thinking of something like renormalizing a quaternion, although I admit I don't know what the equivalent operation would be on an isometry. (Just dividing through by the scale factor, I guess?)

Maybe then a large deviation would be fatal (say, >1e-3), but something smaller would be fixed with a warning.

v4hn, 2020-06-02:
> A robot's reach is usually about 1m.

There is users out there who tend to disagree, operating big industry arms. :)

> Requiring a "precision" of 1e-12 puts us at the picometer level. You measure atom radii in picometers.
> If we bumped this up, even to 1e-6, that would still be way more accurate than robots ever actually need to be.

From my perspective, this is not about noise or eef precision, but about correctness of the isometry invariant.
If you can argue why isometries with less precision (and thus bounded non-orthonormal basis vectors) will not break anything aside from introducing bounded eef position errors, I'm perfectly happy with relaxing the precision to float.

> What if instead of just dying the transforms were fixed?

This check is only triggered in `Debug` builds. If you use `Release` (as is the default), the checks will never be computed (and thus can't fix anything).  Before the relevant patch, MoveIt implicitly normalized everywhere, at a high computational cost.

gavanderhoorn, 2020-06-02:
> > A robot's reach is usually about 1m.
> There is users out there who tend to disagree, operating big industry arms. :)

indeed.

Personal favourite: 16m linear track + 2.7m 6dof robot + 0.8m multi-dof eef.

> Before the relevant patch, MoveIt implicitly normalized everywhere, at a high computational cost.

And potentially changing the behaviour as well.

v4hn, 2020-06-02:
> Personal favourite: 16m linear track + 2.7m 6dof robot + 0.8m multi-dof eef.

Damn, now I'm jealous.. Any chance of me visiting to play around with it for a while? :)

gavanderhoorn, 2020-06-02:
Not really I'm afraid.

Also: once you've seen it move twice, it sort of loses it's appeal :)

AndyZe, 2020-06-03:
OK, a badass 20m robot -> 20 micrometers of inaccuracy due to scaling if we set the tolerance at 1e-6. Still negligible. I might start to get concerned if the robot had 100m of reach. If we set the tolerance at 1e-7, you could have a kilometer-long robot. Got any examples of that?  ;)

v4hn, 2020-06-03:
> If we set the tolerance at 1e-7, you could have a kilometer-long robot. Got any examples of that?  ;)

Well, be prepared for the future... No seriously. I fully understand your argument about scale and it is a valid one.
I am simply concerned about the isometry invariant, which is violated quite a bit with less strict boundaries and you ignored that argument.
I do see how this is hard to refute, but maybe we can ask about it in the Eigen developer community?

If you are unsure about the accuracy of your transforms before passing them to MoveIt, you can simply paste it into an `Affine3d` and call `rotation()`, which will implicitly normalize the resulting `3x3` matrix.
We could even provide a helper for that in `moveit_core`.

JStech, 2020-06-04:
A mathematician would argue that if you're worried about correctness the threshold should be 0.

v4hn, 2020-06-05:
> A mathematician would argue that if you're worried about correctness the threshold should be 0.

Any mathematician worth their title either frowns at `double` altogether or could qualify the required threshold for the operations performed.

v4hn, 2020-06-11:
I lurked around in the `#eigen` IRC channel today. Discussing with one of the present users, it probably depends on the exact application and code a lot.

I would still prefer to just add short helper functions to normalize the isometry for python and c++ which you can use if you stumble into problems. Is that really not acceptable for your use cases?

Alternatively it's probably a bigger point for the next maintainer meeting.

AndyZe, 2020-06-15:
it's not an immediate problem for us anymore because I manually normalized the quaternions in the scene file. I would still prefer simply to widen the tolerance. But I guess we can keep this issue open and wait for others to weigh in if it affects them

rhaschke, 2020-06-16:
Fixing the offending quaternions is the better solution, for sure.

JStech, 2020-09-22:
> Have you involved float precision somewhere?

We ran into this problem again, and floats were the culprit. Double precision is overkill for most applications, why is it required?